esp8266/Arduino #8817: IPAddress::operator= needs 32-bit aligned pointer

TD-er, 2023-01-18:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[Current master / 2.7.4]
- Development Env:		[Platformio]
- Operating System:		[Windows|Ubuntu]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

```
LoadStoreAlignmentCause: Load or store to an unaligned address
  epc1=0x402a1358 in IPAddress::operator=(unsigned char const*) at ??:?
en deze stacktrace:
0x4029280d in String::operator=(char const*) at ??:?
0x401012bb in umm_free_core at umm_malloc.cpp:?
0x401015a2 in malloc at ??:?
0x402d1866 in wifi_get_opmode at ??:?
0x402a2568 in spiffs_cache_page_get_by_fd at ??:?
0x402a2568 in spiffs_cache_page_get_by_fd at ??:?
0x402a2568 in spiffs_cache_page_get_by_fd at ??:?
0x402a2568 in spiffs_cache_page_get_by_fd at ??:?
0x4024821f in setConnectionSpeed() at ??:?
0x40248dc2 in prepareWiFi() at ??:?
0x40241468 in WiFiEventData_t::markWiFiBegin() at ??:?
0x40248f29 in AttemptWiFiConnect() at ??:?
0x40205e24 in ESP8266WiFiSTAClass::status() at ??:?
0x40248084 in wifiAPmodeActivelyUsed() at ??:?
0x4024909c in WiFiConnectRelaxed() at ??:?
0x40245ce9 in NetworkConnectRelaxed() at ??:?
0x4024bf23 in ESPEasy_setup() at ??:?
0x40215084 in setup at ??:?
0x40293ddb in loop_wrapper() at core_esp8266_main.cpp:?
0x40100469 in cont_wrapper at ??:?
```

See the assignment operator and other functions where the `const uint8_t*` is cast to an `const uint32_t*`

https://github.com/esp8266/Arduino/blob/f06710eb6e1559b8a0035d02c0c0fc0c860b821f/cores/esp8266/IPAddress.cpp#L118-L122

https://github.com/esp8266/Arduino/blob/f06710eb6e1559b8a0035d02c0c0fc0c860b821f/cores/esp8266/IPAddress.cpp#L130-L132

The call is simply like this:

```c++
  const IPAddress ip     = Settings.IP;
  const IPAddress gw     = Settings.Gateway;
  const IPAddress subnet = Settings.Subnet;
  const IPAddress dns    = Settings.DNS;
```

Where all these are part of a larger struct, which begins like this:

```c++
  unsigned long PID = 0;
  int           Version = 0;
  int16_t       Build = 0;
  uint8_t       IP[4] = {0};
  uint8_t       Gateway[4] = {0};
  uint8_t       Subnet[4] = {0};
  uint8_t       DNS[4] = {0};
```

As can be seen, these are indeed not 32-bit aligned.

I think the build toolchain may recently have been updated, as this struct and the code that's now failing has not been changed for at least 4 years.

The same code in `IPAddress` has not been changed for a while as the current master branch has the same code for these functions as the 2.7.4 we're using.

mcspr, 2023-01-18:
s/Needs aligned/We need memcpy into u32 and then copy instead of reinterpret cast/?
If you compare with 2.7.4, things definitely have changed b/c of GCC 10.3

TD-er, 2023-01-18:
There is already a proper constructor which does copy the bytes one-by-one, so why should there be a reinterpret_cast or memcpy?

By the way, I did a few more tests to let the compiler find all occurences of this `operator=` and found a few more.
I did make the assignment operator private.

But strangely enough the exact lines as above with `const IPAddress ip = Settings.IP;` etc. were not found.
This one was: `WiFiEventData.dns0_cache = dns;` which was one line below in my code. (`dns0_cache ` is of type `IPAddress`)

So maybe the compiler already may strip this code in the optimizer step?

This is the complete function:
https://github.com/letscontrolit/ESPEasy/blob/0c0175ebed4938aecc21d60ea595af3f88fe9544/src/src/ESPEasyCore/ESPEasyWifi.cpp#L1448-L1468

d-a-v, 2023-01-18:
> `v4() = *reinterpret_cast<const uint32_t*>(address); `

Can you try 

```cpp
memcpy_P(v4(), address, 4);
```

TD-er, 2023-01-18:
OK.. also tried with `IPAddress& operator=(uint32_t address);` set to private to see what this may show up and then noticed some very strange thing in `int ESP8266WiFiGenericClass::hostByName(const char* aHostname, IPAddress& aResult, uint32_t timeout_ms)`

`aResult = static_cast<uint32_t>(INADDR_NONE);` where `INADDR_NONE` is already of type `IPAddress`, so casting to uint32_t is not needed.
And what makes it even more strange is that `INADDR_NONE` is 255.255.255.255 on ESP8266, but 0.0.0.0 on ESP32.

mcspr, 2023-01-18:
Let's just figure out assignments & ctors first. I'll prepare a PR.

TD-er, 2023-01-18:
> > `v4() = *reinterpret_cast<const uint32_t*>(address); `
> 
> Can you try
> 
> ```c++
> memcpy_P(&v4(), address, 4);
> ```

This is a fix.
Just tested it and with this code it works just fine. reverted it and crashes very reproducible.

Just strange that the lines with the single used `const IPAddress ip = Settings.IP;` didn't fail to compile with the assignment operator set to private.

PR #8818 (2023-01-20): Properly handle u8 pointers when assigning and comparing
* Simplify ctors and operator=, use a common code paths instead of special handling here and there
* Fix #8817 u8<->u32 assignment with pointers, copy before using u8 data
* Do not use raw\_address() internally

raw6() is one thing left over, we sometimes have `u8 scope` at the end so it is not techically valid code to reinterpret like that. alignment works out, though
tbh I'd prefer we return a real value instead of pointer for our internal usage, and leave ptr access for Arduino compatibility (meaning, std::array<uint8_t, 4> and std::a [... cut]
closed 2023-01-20 by mcspr