esp8266/Arduino #8830: Frequent Exception 0 Crashes with v3.1.0 and v3.1.1, but none with Core <v3.0.2

ken830, 2023-01-26:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:				ESP8266EX
- Core Version:			3.1.1., 3.1.0
- Development Env:		Arduino IDE, Platformio
- Operating System:		Windows

### Settings in IDE

- Module:  			Wemos D1 mini r2
- Flash Mode:			[qio|dio|other]?
- Flash Size: 			4MB/1MB
- lwip Variant:   		v2 Lower Memory
- Reset Method:		[ck|nodemcu]?
- Flash Frequency:		[40Mhz]?
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   		921600

### Problem Description

I (and several others) have an [AirGradient Pro kit](https://www.airgradient.com/open-airgradient/kits/), which runs on a Wemos D1 Mini. I am seeing repeated Exception 0 crashes every few minutes when compiled with Core v3.1.0 and v3.1.1, but no Exceptions at all (tested over 24 hours) with v3.0.2 and below (I tested down to v3.0.0). This behavior is consistent with the latest Arduino IDE 2.0.3, Arduino IDE 1.8.19, and VSCode+PlatformIO with platform-espressif8266 v4.1.0 (which is updated to Arduino Core v3.1).

Because it crashes every few minutes, I captured a lot of exceptions, but I had no way of decoding in Arduino IDE 2.x and IDE 1.8.x didn't work with the Arduino Core v3.1.x because of Issue #8811, but I finally worked-around the issue to and got v3.1.x core to work with the older Arduino IDE v1.8.x  so that I could finally run the ESP Exception Decoder. I think this is what is needed?

It crashes so often, I have decoded dozens of exceptions and each one is always the same:

`circular_queue ::available() const at c:\users\ken\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.1.0-gcc10.3-e5f9fec\xtensa-lx106-elf\include\c++\10.3.0\bits/atomic_base.h line 420`

I'm not a very experienced, but I have spent hours isolating it to the Core v3.1.x update and getting the decoder working. Apologies if I'm not reporting this correctly or if there is some debug procedure I should do that is not obvious to me. Thanks.

### Debug Messages

```
Exception 0: Illegal instruction
PC: 0x40219d04: circular_queue ::available() const at c:\users\ken\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.1.0-gcc10.3-e5f9fec\xtensa-lx106-elf\include\c++\10.3.0\bits/atomic_base.h line 420
EXCVADDR: 0x00000000

Decoding stack results
0x40100648: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_main.cpp line 238
0x40100c04: interrupt_handler(void*, void*) at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_wiring_digital.cpp line 167
0x40100b40: interrupt_handler(void*, void*) at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_wiring_digital.cpp line 138
0x40100c04: interrupt_handler(void*, void*) at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_wiring_digital.cpp line 167
0x40100b40: interrupt_handler(void*, void*) at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_wiring_digital.cpp line 138
0x4010062b: ets_intr_unlock() at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_main.cpp line 226
0x40228c85: sys_timeout_abs at core/timeouts.c line 189
0x401000ab: app_entry_redefinable() at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_main.cpp line 386
0x40100648: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_main.cpp line 238
0x40100648: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_main.cpp line 238
0x40223559: glue2esp_linkoutput at glue-esp/lwip-esp.c line 301
0x40223787: new_linkoutput at glue-lwip/lwip-git.c line 272
0x40100648: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_main.cpp line 238
0x4022cb94: ip4_output_if_opt_src at core/ipv4/ip4.c line 1764
0x40223559: glue2esp_linkoutput at glue-esp/lwip-esp.c line 301
0x40223787: new_linkoutput at glue-lwip/lwip-git.c line 272
0x40223be2: ethernet_output at netif/ethernet.c line 312
0x4010107e: umm_free_core(umm_heap_context_t*, void*) at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\umm_malloc\umm_malloc.cpp line 642
0x40100c04: interrupt_handler(void*, void*) at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_wiring_digital.cpp line 167
0x40100648: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_main.cpp line 238
0x40100b40: interrupt_handler(void*, void*) at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_wiring_digital.cpp line 138
0x402176d3: Twi::WAIT_CLOCK_STRETCH() at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266/PolledTimeout.h line 130
[... 211 lines cut ...]
0x4020349c: u8g2_ll_hvline_vertical_top_lsb at C:\Users\Ken\Documents\Arduino\libraries\U8g2\src\clib\u8g2_ll_hvline.c line 97
0x40201d7e: u8g2_font_decode_len at C:\Users\Ken\Documents\Arduino\libraries\U8g2\src\clib\u8g2_font.c line 459
0x40100648: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_main.cpp line 238
0x40100c04: interrupt_handler(void*, void*) at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_wiring_digital.cpp line 167
0x40217734: Twi::WAIT_CLOCK_STRETCH() at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_si2c.cpp line 167
0x4021790d: Twi::write_bit(bool) at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_si2c.cpp line 325
0x40217b48: Twi::writeTo(unsigned char, unsigned char*, unsigned int, unsigned char) at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_si2c.cpp line 368
0x4021787d: Twi::write_stop() at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_si2c.cpp line 304
0x40217bb1: Twi::writeTo(unsigned char, unsigned char*, unsigned int, unsigned char) at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_si2c.cpp line 393
0x40212662: TwoWire::write(unsigned char const*, unsigned int) at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\libraries\Wire\Wire.cpp line 209
0x40212834: TwoWire::endTransmission(unsigned char) at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\libraries\Wire\Wire.cpp line 172
0x40212d09: u8x8_byte_arduino_hw_i2c(u8x8_t*, uint8_t, uint8_t, void*) at C:\Users\Ken\Documents\Arduino\libraries\U8g2\src\U8x8lib.cpp line 1354
0x40202619: u8x8_cad_ssd13xx_fast_i2c at C:\Users\Ken\Documents\Arduino\libraries\U8g2\src\clib\u8x8_cad.c line 586
0x4021790d: Twi::write_bit(bool) at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_si2c.cpp line 325
0x4021787d: Twi::write_stop() at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_si2c.cpp line 304
0x40217b01: Twi::readFrom(unsigned char, unsigned char*, unsigned int, unsigned char) at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_si2c.cpp line 433
[... cut ...]
0x40216e2e: esp_try_delay(unsigned int, unsigned int, unsigned int) at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_main.cpp line 182
0x40217d90: __delay(unsigned long) at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266/coredecls.h line 69
0x402037a7: updateTVOC() at C:\Users\Ken\Documents\Arduino\sketch_jan24a/sketch_jan24a.ino line 184
0x40204a84: loop() at C:\Users\Ken\Documents\Arduino\sketch_jan24a/sketch_jan24a.ino line 149
```

mcspr, 2023-01-26:
> Exception 0: Illegal instruction
PC: 0x40219d04: circular_queue ::available() const at c:\users\ken\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.1.0-gcc10.3-e5f9fec\xtensa-lx106-elf\include\c++\10.3.0\bits/atomic_base.h line 420
EXCVADDR: 0x00000000
> ...
> 0x40100110: SoftwareSerial::preciseDelay() at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\libraries\SoftwareSerial\src/SoftwareSerial.h line 246

*Likely* something to do with SoftwareSerial; 3.0.2 ships with 6.12.7, 3.1.x is 7.0.0
https://github.com/plerup/espsoftwareserial/search?q=circular_queue

And, unless I am misreading the EXCVADDR, looks like `this` of that queue is pointing at nil instead of an object (for whatever reason)

mcspr, 2023-01-26:
For the above, see https://docs.platformio.org/en/latest/projectconf/build_configurations.html
We need `-g` in `build_flags` or `build_type = debug` (which does that for us) for decoder to work

ken830, 2023-01-26:
> And, unless I am misreading the EXCVADDR, looks like `this` of that queue is pointing at nil instead of an object (for whatever reason)

This observation of yours really struck me and I remember seeing warnings when compiling the sketch from AirGradient. And I _just_  noticed that I don't see those warnings when compiling the sketch recently with Core v3.0.2. 

#### Core v3.1.1
```
C:\Users\Ken\Documents\Arduino\libraries\AirGradient_Air_Quality_Sensor\AirGradient.cpp: In member function 'TMP_RH AirGradient::returnError(TMP_RH_ErrorCode)':
C:\Users\Ken\Documents\Arduino\libraries\AirGradient_Air_Quality_Sensor\AirGradient.cpp:548:14: warning: converting to non-pointer type 'float' from NULL [-Wconversion-null]
  548 |   result.t = NULL;
      |              ^~~~
C:\Users\Ken\Documents\Arduino\libraries\AirGradient_Air_Quality_Sensor\AirGradient.cpp:549:15: warning: converting to non-pointer type 'int' from NULL [-Wconversion-null]
  549 |   result.rh = NULL;
      |               ^~~~
. Variables and constants in RAM (global, static), used 33372 / 80192 bytes (41%)
║   SEGMENT  BYTES    DESCRIPTION
╠══ DATA     1516     initialized variables
╠══ RODATA   3832     constants       
╚══ BSS      28024    zeroed variables
. Instruction RAM (IRAM_ATTR, ICACHE_RAM_ATTR), used 62695 / 65536 bytes (95%)
║   SEGMENT  BYTES    DESCRIPTION
╠══ ICACHE   32768    reserved space for flash instruction cache
╚══ IRAM     29927    code in IRAM    
. Code in flash (default, ICACHE_FLASH_ATTR), used 355008 / 1048576 bytes (33%)
║   SEGMENT  BYTES    DESCRIPTION
╚══ IROM     355008   code in flash   
esptool.py v3.0
Serial port COM4
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 84:f3:eb:24:4b:d0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 394432 bytes to 280039...
Writing at 0x00000000... (5 %)
Writing at 0x00004000... (11 %)
Writing at 0x00008000... (16 %)
Writing at 0x0000c000... (22 %)
Writing at 0x00010000... (27 %)
Writing at 0x00014000... (33 %)
Writing at 0x00018000... (38 %)
Writing at 0x0001c000... (44 %)
Writing at 0x00020000... (50 %)
Writing at 0x00024000... (55 %)
Writing at 0x00028000... (61 %)
Writing at 0x0002c000... (66 %)
Writing at 0x00030000... (72 %)
Writing at 0x00034000... (77 %)
Writing at 0x00038000... (83 %)
Writing at 0x0003c000... (88 %)
Writing at 0x00040000... (94 %)
Writing at 0x00044000... (100 %)
Wrote 394432 bytes (280039 compressed) at 0x00000000 in 6.4 seconds (effective 495.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

#### Core v3.0.2
```
Executable segment sizes:
ICACHE : 32768           - flash instruction cache 
IROM   : 353468          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 29585   / 32768 - code in IRAM          (IRAM_ATTR, ISRs...) 
DATA   : 1516  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 3840  ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 27904 )         - zeroed variables      (global, static) in RAM/HEAP 
Sketch uses 388409 bytes (37%) of program storage space. Maximum is 1044464 bytes.
Global variables use 33260 bytes (40%) of dynamic memory, leaving 48660 bytes for local variables. Maximum is 81920 bytes.
esptool.py v3.0
Serial port COM4
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 84:f3:eb:24:4b:d0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 392560 bytes to 277859...
Writing at 0x00000000... (5 %)
Writing at 0x00004000... (11 %)
Writing at 0x00008000... (17 %)
Writing at 0x0000c000... (23 %)
Writing at 0x00010000... (29 %)
Writing at 0x00014000... (35 %)
Writing at 0x00018000... (41 %)
Writing at 0x0001c000... (47 %)
Writing at 0x00020000... (52 %)
Writing at 0x00024000... (58 %)
Writing at 0x00028000... (64 %)
Writing at 0x0002c000... (70 %)
Writing at 0x00030000... (76 %)
Writing at 0x00034000... (82 %)
Writing at 0x00038000... (88 %)
Writing at 0x0003c000... (94 %)
Writing at 0x00040000... (100 %)
Wrote 392560 bytes (277859 compressed) at 0x00000000 in 6.3 seconds (effective 497.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

####Here is the code snippet from AirGradient.cpp. This is part of the overall set of code that reads the temperature (TMP) and relative humidity (RH) from the sensor. I'm guessing this piece defines the values that are returned if the sensor reading results in an error.
```
546   TMP_RH AirGradient::returnError(TMP_RH_ErrorCode error) {
547     TMP_RH result;
548     result.t = NULL;
549     result.rh = NULL;
550   
551     result.t_char[0] = 'N';
552     result.t_char[1] = 'U';
553     result.t_char[2] = 'L';
554     result.t_char[3] = 'L';
555   
556     result.rh_char[0] = 'N';
557     result.rh_char[1] = 'U';
558     result.rh_char[2] = 'L';
559     result.rh_char[3] = 'L';
560   
561     result.error = error;
562     return result;
}
```

I'm a HW engineer, so forgive me if I'm a bit dense here. I don't know why the two core versions would treat this code differently. And I can't really understand the problem. Is a float or int not able to take on the `NULL` value? Why is it converting it?

And I'm not so sure this is related to the SoftwareSerial piece at all because the temperature/humidity sensor is on the i2c interface. This might be nothing.

I'm working through the parts of code that use SoftwareSerial now and it's related to the PM2.5 and CO2 sensors.

mcspr, 2023-01-26:
Warning is unrelated to the error, you just need to use literal `0` instead of `NULL`
(which is a pointer type, not a number! sometimes this is allowed - https://en.cppreference.com/w/cpp/language/implicit_conversion - sometimes this does bad things and should be avoided)

Arduino IDE 1.x and 2.x disable compiler warnings by default. Since #8495, we work around that, but looks like at least one of warnings was missed.

It is totally possible that the issue is not in SoftwareSerial, but we just happen to break its operation with something directly or indirectly related to how it is used in the code :shrug:

ken830, 2023-01-26:
Thank you, @mcspr! I probably understood NULL was a pointer at some point during my un-grad studies over 20 years ago.

Because you mentioned:

> For the above, see https://docs.platformio.org/en/latest/projectconf/build_configurations.html We need `-g` in `build_flags` or `build_type = debug` (which does that for us) for decoder to work

I decided to see if the PlatformIO decoder would give me any additional clues and so I set `build_type = debug` and waited for an exception. When I didn't see an exception in over 15 minutes, I removed the `build_type` option (reverting back to `build_type = release` as a default, I assume), re-compiled, and it crashed several times within 15 minutes. That was my sanity-check that the exact same source code crashes frequently normally. I then immediately re-instated `build_type = debug` , re-compiled, and let it run over-night while I slept and now I find it **hasn't crashed at all in over 7 hours**, which gives me confidence to say that this debug build option "fixes" the issue the same way that moving down to Core v3.0.2 to work around the [#8811 ](https://github.com/esp8266/Arduino/issues/8811) issue also inadvertently "fixed" the issue the last time.

Any clue as to what direction I need to head down to figure out how building with `build_type = debug` affects the behavior of the code? I would imagine it should minimally affect the code behavior, otherwise, it wouldn't be very useful in debugging purposes. And so with that premise, I'm hopeful this clue can really narrow-in on possible root causes.

mhightower83, 2023-01-26:
Exception 0 is often ICACHE code that is not cached at the time of access and the SPI bus is busy. The processor reads zero which is also defined as an illegal instruction.

Can be caused by a missing IRAM_ATTR on the ISR or functions that it calls.
Also, inline code that fails to inline and lands as a function in ICACHE/flash can cause this problem.

`0x40219d04` `atomic_base.h` line 420 is in ICACHE address space. I assume it is in an ICACHE function that is called by an ISR. If I am looking at the correct copy of `atomic_base.h` line 420, to me the macro `_GLIBCXX_ALWAYS_INLINE` implies that it should have been inlined.

ken830, 2023-01-27:
I know that the issue first appeared when moving to Core v3.1.x. It is also mysteriously-resolved when compiling with `build_type = debug`. My board (with espressif8266 platform v 4.1.0 which includes the Core v3.1) is running crash-free for almost 12 hours now whereas it would crash within 5 minutes before.

I can understand that it could crash if an ISR is missing the IRAM_ATTR attribute and an interrupt comes in causing the processor to access it from Flash when the SPI bus happens to be busy, but I don't see how that behavior could be different between the Core versions or the build_type flag?

The inline bit is interesting too but again, it looks like it is inline. Your comment about "fails to inline" -- Is that a thing? Is that just a suggestion to the compiler and it's not guaranteed? Could that be something the new Core version or build flag could affect?

At the time I captured those exceptions, I also captured the relevant section of atomic_base.h:

```
419      _GLIBCXX_ALWAYS_INLINE __int_type
420      load(memory_order __m = memory_order_seq_cst) const noexcept
421      {
422	memory_order __b = __m & __memory_order_mask;
423	__glibcxx_assert(__b != memory_order_release);
424	__glibcxx_assert(__b != memory_order_acq_rel);
```
Because I just looked at Arduino and ESP (or any microcontroller) for the first time just a few days ago, have almost no software background, and none of the AirGradient code is mine, I've asked AirGradient for help and was told they are contacting their technical person to see if he can help.

mhightower83, 2023-01-27:
>  Your comment about "fails to inline" 

As I understand it `inline` is a suggestion to the compiler. It is not obligated to inline the code; however, with added attributes like `__attribute__((__always_inline__))` the inline should be forced. Regarding the `_GLIBCXX_ALWAYS_INLINE` macro, it appears to use this attribute, unless it is redefined by a build option. And yet the crashing code is not in IRAM. 

> It is also mysteriously-resolved when compiling with build_type = debug.

My only thought on this and the older version of the core working is code size changes and which code lands and stays in the cache changes. Some problems can be made worse by changing the cache size from 32K to 16K.

ken830, 2023-01-27:
Looking at [https://github.com/esp8266/esp8266-wiki/wiki/Memory-Map](url), I see 0x4020_0000 is the beginning of SPI Flash mapped area. The D1 Mini has 4MB SPI Flash, so if my mental math serves me, that would be up to 0x403F_FFFF (22 bits). So, I see the exception occurs within that area. Is that indicative of anything? If the instructions of a function is indeed in IRAM, would the PC be pointing to there (0x4010_0000) instead? Or is this why you're saying that the crashing code is not in IRAM (the PC is pointing to Flash)?

What about the case of a cached instruction? Would the PC in the exception dump still be pointing to the non-cached location?

> My only thought on this and the older version of the core working is code size changes and which code lands and stays in the cache changes. Some problems can be made worse by changing the cache size from 32K to 16K.

But if that's the case, I would expect it to appear to be very temperamental. Like I could get code that works great with one version and not another, then grow the code a bit and then have it work horribly with the previously-good version and great with the previously-bad version, simply because of where things happen to land. But I'm not seeing that. With the several versions of code I've tested, it's always bad with the newer core versions (without the debug flag) -- to a varying degrees (some code crashes much more frequently).

And when we are talking about where functions land, are we talking about at compile/build-time or at runtime? I only have a crude understanding of processor architecture (and certainly nothing modern), so sorry if dumb question, but instruction caching sounds like it should be a runtime thing, right? And the inlining and IRAM would be determined at build-time?

mhightower83, 2023-01-27:
I believe the cacheable address space is limited to 1MB for the ESP8266
Yes, the fact the crashing location is in flash/ICACHE. The executable flash address range starts at 0x4020000. An illegal instruction opcode is three bytes of zeros. When the SPI bus is busy and a cache miss cannot be loaded from flash, zeros are read for instructions. I think these hint toward being called from an Interrupts context when the SPI bus was busy.

> where functions land,

I was poorly referring to how the contents of the cache are always changing. The way the compiler and linker organize the code could change with small code changes. The timing of operations could alter the current cache contents between interrupts.
That said, it is hard to see how the debug build would be so solid. Then again if more calls were made to a piece of cacheable code that is also needed for an ISR that would help it work; however, it is hard to see why debug code would be making more calls to `circular_queue::available()`. I am probably not helping here.

I cannot make out from the dump who is calling `circular_queue::available()`. I think this is what they call a leaf function it doesn't call any other functions so the caller's return address may be held in a register and never written to the stack.

I have used this macro in leaf functions to force the compiler to save the return address on the stack. It contains no assembly, it lies to the compiler. Instructing it that we are going to clobber register "a0" so that it will save it.
```cpp
#define DEBUG_ESP_BACKTRACELOG_LEAF_FUNCTION(...) __asm__ __volatile__("" ::: "a0", "memory")
```
You would place the macro call at the beginning of the function to be sure register "a0" is saved as soon as possible.

And the build define  `-fno-optimize-sibling-calls` can help push more call details to the stack.

mcspr, 2023-01-27:
Just as a reminder, we also could compare built .elf files

`SoftwareSerial::rxBitISR()` and `::rxBitSyncISR()` are interrupt handlers that explicitly call `available()`
https://github.com/plerup/espsoftwareserial/blob/56d19bf5fe9b38cdd438e06e1396b4a341d86274/src/SoftwareSerial.cpp#L622-L658
https://github.com/plerup/espsoftwareserial/blob/56d19bf5fe9b38cdd438e06e1396b4a341d86274/src/SoftwareSerial.cpp#L251-L255

And... nothing else, apparently. `::read()` and `::capacity()` indirectly use it, but not from ISR context.

mcspr, 2023-01-27:
Aaand we also may run into a different code when using PlatformIO debug; it also replaces optimization mode with `-Og`, my mistake thinking these are the same as `-g`

```python
# build_type = debug
  'CCFLAGS': [ '-mlongcalls',
               '-mtext-section-literals',
               '-falign-functions=4',
               '-U__STRICT_ANSI__',
               '-ffunction-sections',
               '-fdata-sections',
               '-Wall',
               '-Werror=return-type',
               '-free',
               '-fipa-pta',
               '-Og',
               '-g2',
               '-ggdb2'],
```

```python
# `build_flags = -g`
  'CCFLAGS': [ '-g',
               '-Os',
               '-mlongcalls',
               '-mtext-section-literals',
               '-falign-functions=4',
               '-U__STRICT_ANSI__',
               '-ffunction-sections',
               '-fdata-sections',
               '-Wall',
               '-Werror=return-type',
               '-free',
               '-fipa-pta'],
```

ken830, 2023-01-27:
Thanks @mhightower83 & @mcspr ! I might need a bit of time to fully digest what you two are telling me.

> An illegal instruction opcode is three bytes of zeros. When the SPI bus is busy and a cache miss cannot be loaded from flash, zeros are read for instructions. I think these hint toward being called from an Interrupts context when the SPI bus was busy.

One question out of curiosity: Are three bytes of zeros the ONLY way to hit an Exception 0? Wouldn't any/all invalid opcode do the same?

> I have used this macro in leaf functions to force the compiler to save the return address on the stack. It contains no assembly, it lies to the compiler. Instructing it that we are going to clobber register "a0" so that it will save it.
> 
> ```c++
> #define DEBUG_ESP_BACKTRACELOG_LEAF_FUNCTION(...) __asm__ __volatile__("" ::: "a0", "memory")
> ```
> 
> You would place the macro call at the beginning of the function to be sure register "a0" is saved as soon as possible.
> 
> And the build define `-fno-optimize-sibling-calls` can help push more call details to the stack.

I understand the idea behind this to make the stack more informative when it dumps. I may have to try this, but I will probably need some hand-holding when the time comes. But I don't know the full mechanics of how this works. This looks like a parameterized macro substitution, but what source file does this go in? Somewhere in the SoftwareSerial code? I don't see `DEBUG_ESP_BACKTRACELOG_LEAF_FUNCTION` in any of the source?? I don't really need to understand though. At least maybe not now.

> Just as a reminder, we also could compare built .elf files

Does this mean that comparing the ELF files would be able to tell us what is calling `circular_queue::available()` at the time of the exception?
 
> `SoftwareSerial::rxBitISR()` and `::rxBitSyncISR()` are interrupt handlers that explicitly call `available()` https://github.com/plerup/espsoftwareserial/blob/56d19bf5fe9b38cdd438e06e1396b4a341d86274/src/SoftwareSerial.cpp#L622-L658 https://github.com/plerup/espsoftwareserial/blob/56d19bf5fe9b38cdd438e06e1396b4a341d86274/src/SoftwareSerial.cpp#L251-L255
> 
> And... nothing else, apparently. queue's `::pop_n()` and `::capacity()` indirectly use it, but not from ISR context.

So, I think this means that if the working theory that an ISR is calling `circular_queue::available()` occasionally causing the exception, then it would have to be one of these two (`SoftwareSerial::rxBitISR()` and `::rxBitSyncISR()`) ISRs.

> _edit:_ Speaking of debug mode and build flags - IDE uses `-g` by default

What do you mean by "IDE"? The Arduino IDE? And on PlatformIO, I've only tried `build_type = debug`. Is `-g` flag identical?
EDIT: Nevermind that last bit. I just saw your updated post with the the -g and build_type differences.

Update on my end: Shortly after my last post here, I decided to try to used Core v3.1.1 and simply replace the SoftwareSerial v7.0.0 with 6.12.7. I did this in my Arduino IDE's library, recompiled, and the board is now running for nearly 5 hours with no exceptions. If this proves to be stable, then I will revert to 7.0.0 to confirm that it crashes. Hopefully, it does. Then, I can narrow down the exact release that triggers the exceptions. Now I'm wondering if I can get the two sets of code as close as possible to each other, would that make your ELF comparison idea much easier to pinpoint?

mcspr, 2023-01-27:
> Does this mean that comparing the ELF files would be able to tell us what is calling circular_queue::available() at the time of the exception?

If the problem is with related to inlining / not inlining, then yes. `-g` would simplify things by a lot by including debug info, too

> Can be caused by a missing IRAM_ATTR on the ISR or functions that it calls.

No IRAM attribute
https://github.com/plerup/espsoftwareserial/blob/bd2f6ce6a78d0c31b6978bcffb3c8379a5e4e2e4/src/circular_queue/circular_queue.h#L111
Pretty sure, even if it is attributed with IRAM, GCC still has a bug where template and attributes don't mix.

So, suppose circular\_queue::available() is not inlined, should we just match it here and force IRAM section?
https://github.com/esp8266/Arduino/blob/e25f9e95783d6de59d6b3b7b4ebc029428c9b539/tools/sdk/ld/eagle.app.v6.common.ld.h#L143-L145

mhightower83, 2023-01-27:
> One question out of curiosity: Are three bytes of zeros the ONLY way to hit an Exception 0? Wouldn't any/all invalid opcode do the same?

I would like to think so. 

I'll say yes based on page 84 of "Xtensa® Instruction Set Architecture (ISA) Reference Manual"
> "ILL or illegal instruction - Illegal instruction executed - The opcode ILL is guaranteed to always be an illegal"
"Attempt to execute an illegal instruction or a legal instruction under illegal conditions"

Also, there appears to be a narrow (two-byte) version of the `ILL.N` instruction as well.

> DEBUG_ESP_BACKTRACELOG_LEAF_FUNCTION

Sorry for the confusion, that is the form I use it. It makes it easy to turn off and on for debug builds. 
Alternatively, you could do something like this in `circular_queue::available()`
```cpp
size_t available() const
{
    __asm__ __volatile__("" ::: "a0", "memory");
    int avail = static_cast<int>(m_inPos.load() - m_outPos.load());
    if (avail < 0) avail += m_bufSize;
    return avail;
}
```
On entry to the function, register `a0` holds the caller's return address. 
`__asm__ __volatile__(...)`  is used to inject inline assembly instructions into `C/C++` code; however, in this case, the instructions list is empty, `""`. And, `"a0"` is in the register clobber list field. This tells the compiler that the listed assembly instructions will destroy the contexts of register `a0`. The compiler pays no attention to the instructions we provide (or not providing) and takes action to protect the contents of the register we claim to be using/clobbering. `"memory"` tells the pesky optimizer, it should not move code before or after the line. At least that is my understanding, sometimes I miss or forget some of the nuances.

ken830, 2023-01-28:
I see you're adding `circular_queue::available()` to IRAM in that commit. That's trying to force the linker to put it in IRAM? Well, in my testing below I saved the ELF files so how would we go about to compare/check if in-lining is at play here?

My testing using Core v3.1.1 with SoftwareSerial v6.12.7 completed 14h of uptime. I then reverted back to SoftwareSerial v7.0.0 and confirmed multiple Exception 0 crashes within 5 minutes. Finally, re-compiled with SoftwareSerial v6.17.1 (version immediately prior to current release) and confirmed running fine for over 4 hours. And finally, for my own sanity, I downloaded v7.0.0 source directly from the SoftwareSerial repository, compared to what I had pulled out to ensure they are the same (they were) and then re-built once again with v7.0.0 to re-verify that it indeed crashes every few minutes with the same Exception.

Core v3.1.1 + SoftwareSerial v7.0.0 => Exception 0
Core v3.1.1 + SoftwareSerial v6.17.1 => Working
Core v3.1.1 + SoftwareSerial V6.12.7 => Working

Core v3.0.2 + SoftwareSerial v7.0.0 => Exception 0
Core v3.0.2 + SoftwareSerial v6.17.1 => Working
Core v3.0.2 + SoftwareSerial V6.12.7 => Working

Core v3.1.1 => Exception 0
Core v3.1.1 + `build_type = debug` => Working

Core v3.1.1 => Exception 0
Core v3.0.2 => Working
Core v3.0.0 => Working

I zipped up the entire build folder from Arduino IDE's Temp folder. The ELF and other stuff are all in there.

[arduino_build_822815_Core3.1.1_SoftwareSerial7.0.0_EXCEPTION0.zip](https://github.com/esp8266/Arduino/files/10524298/arduino_build_822815_Core3.1.1_SoftwareSerial7.0.0_EXCEPTION0.zip)
[arduino_build_750952_Core3.1.1_SoftwareSerial6.17.1_WORKING.zip](https://github.com/esp8266/Arduino/files/10524299/arduino_build_750952_Core3.1.1_SoftwareSerial6.17.1_WORKING.zip)
[arduino_build_401175_Core3.0.2_SoftwareSerial7.0.0_EXCEPTION0.zip](https://github.com/esp8266/Arduino/files/10524301/arduino_build_401175_Core3.0.2_SoftwareSerial7.0.0_EXCEPTION0.zip)
[arduino_build_270738_Core3.0.2_SoftwareSerial6.17.1_WORKING.zip](https://github.com/esp8266/Arduino/files/10524300/arduino_build_270738_Core3.0.2_SoftwareSerial6.17.1_WORKING.zip)

> I would like to think so.
>
> I'll say yes based on page 84 of "Xtensa® Instruction Set Architecture (ISA) Reference Manual"

Yes, but I read this and wasn't sure:

**Table 4–59. Instruction Exceptions under the Exception Option**
| Condition | Description | Required Option & Vector|
| ------------- | ------------- | ------------- | 
|Illegal instruction | Attempt to execute an illegal instruction or a legal instruction under illegal conditions |  Exception Option General vector|

What could "a legal instruction under illegal conditions" be? Does this suggest that Exception 0 _could_ be encountered somehow besides ILL or ILL.N?

> Sorry for the confusion, that is the form I use it. It makes it easy to turn off and on for debug builds. Alternatively, you could do something like this in `circular_queue::available()`
> 
> ```c++
> size_t available() const
> {
>     __asm__ __volatile__("" ::: "a0", "memory");
>     int avail = static_cast<int>(m_inPos.load() - m_outPos.load());
>     if (avail < 0) avail += m_bufSize;
>     return avail;
> }
> ```
> 
> On entry to the function, register `a0` holds the caller's return address. `__asm__ __volatile__(...)` is used to inject inline assembly instructions into `C/C++` code; however, in this case, the instructions list is empty, `""`. And, `"a0"` is in the register clobber list field. This tells the compiler that the listed assembly instructions will destroy the contexts of register `a0`. The compiler pays no attention to the instructions we provide (or not providing) and takes action to protect the contents of the register we claim to be using/clobbering. (For now "protect" gets the register saved on the stack. A future compiler revision could take a different route.) `"memory"` tells the pesky optimizer, it should not move code before or after the line. At least that is my understanding, sometimes I miss or forget some of the nuances.

Ahh... This is great. I re-installed core v3.1.1, replaced the code with yours, re-compiled, and ran. As usual, it crashes with exception 0 every few minutes the same way as before. Here're some stack dumps. One thing that I don't understand, but I guess it makes sense: The PC is now decoded to circular_queue.h line 119 (before, it was atomic_base.h line 420.

Here's the snippet from circular_queue.h:
```
108    /*!
109        @brief	Get a snapshot number of elements that can be retrieved by pop.
110    */
111	/*
112    size_t available() const
113    {
114        int avail = static_cast<int>(m_inPos.load() - m_outPos.load());
115        if (avail < 0) avail += m_bufSize;
116        return avail;
117    }
118	*/
119	size_t available() const
120	{
[... 174 lines cut ...]
0x40201d7e: u8g2_font_decode_len at C:\Users\Ken\Documents\Arduino\libraries\U8g2\src\clib\u8g2_font.c line 459
0x40202374: u8g2_draw_l90_r0 at C:\Users\Ken\Documents\Arduino\libraries\U8g2\src\clib\u8g2_setup.c line 312
0x40217ac4: Twi::readFrom(unsigned char, unsigned char*, unsigned int, unsigned char) at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_si2c.cpp line 414
0x40100648: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_main.cpp line 238
[... cut ...]
0x40217dd0: __delay(unsigned long) at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266/coredecls.h line 69
0x402037a7: updateTVOC() at C:\Users\Ken\Documents\Arduino\sketch_jan24a/sketch_jan24a.ino line 184
0x40204ac4: loop() at C:\Users\Ken\Documents\Arduino\sketch_jan24a/sketch_jan24a.ino line 149
```

I also zipped the build folder of this one as well. ELF file inside, if that would be helpful.
[arduino_build_270738_Core3.1.1_MACRO_BACKTRACE.zip](https://github.com/esp8266/Arduino/files/10526076/arduino_build_270738_Core3.1.1_MACRO_BACKTRACE.zip)

mhightower83, 2023-01-28:
Great verification. 
Sadly, I don't see our expected caller to `circular_queue::available()`
> And the build define -fno-optimize-sibling-calls can help push more call details to the stack.

This is also needed. Was it included in this build? This optimization hinders seeing the call chain in the stack dump. When a function is going to return after a call, the compiler replaces the call with a jump instruction. After a crash, there are no tracks in the stack leading into the crash.

rochi7, 2023-01-28:
I made a similar observation on my weather station.
Since updating to core 3.1.1 I've been getting exceptions at irregular intervals
- ResetReason=Exception
- ResetInfo=Fatal exception:0 flag:2 (Exception) epc1:0x4021559c epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
- SDK=2.2.2-dev(38a443e)
- Core=3.1.1

in the MAP File i found the address **4021559c** related to circular_queue object

[image]

ken830, 2023-01-28:
> This is also needed. Was it included in this build? This optimization hinders seeing the call chain in the stack dump. When a function is going to return after a call, the compiler replaces the call [...]

Oh, no, I totally overlooked that part! I couldn't find any place to add build options in ArduinoIDE, so I'm looking at doing this in PlatformIO now. Is it under Build  Options ==> `build_flags` that I should define this option?

ken830, 2023-01-28:
> Oh, no, I totally overlooked that part! I couldn't find any place to add build options in ArduinoIDE, so I'm looking at doing this in PlatformIO now. Is it under Build Options ==> `build_flags` that I should define this option?

Hmm.. Now that I think about it, I can't do this in PlatformIO because I can get the Exception Decoder to work there unless I use the `build_type = debug` flag. But if I do a debug build, then it won't crash. I'll need to figure out a way to do this in Arduino IDE 1.8.

mhightower83, 2023-01-28:
> .. I use the build_type = debug flag. 

Right, I forgot about that.
For the Arduino IDE, I would have suggested using \<SketchName\>`.ino.globals.h` 
```cpp
/*@create-file:build.opt@
-fno-optimize-sibling-calls
*/
```
However, for this to work on Windows, you will need to the updated [./tools/mkbuildoptglobals.py](https://github.com/esp8266/Arduino/blob/master/tools/mkbuildoptglobals.py). This should appear later in 3.1.2.

[More on global build options](https://arduino-esp8266.readthedocs.io/en/latest/faq/a06-global-build-options.html#how-to-specify-global-build-defines-and-options)

ken830, 2023-01-28:
> Right, I forgot about that. For the Arduino IDE, I would have suggested creating a file <SketchName>`.ino.globals.h` with the build options:
> 
> ```c++
> /*@create-file:build.opt@
> -fno-optimize-sibling-calls
> */
> ```
> 
> However, for this to work today on Windows, you will need to the updated [./tools/mkbuildoptglobals.py](https://github.com/esp8266/Arduino/blob/master/tools/mkbuildoptglobals.py). This should appear later in the 3.1.2 update.
> 
> [More on global build options](https://arduino-esp8266.readthedocs.io/en/latest/faq/a06-global-build-options.html#how-to-specify-global-build-defines-and-options)

I'm so glad for this since I was halfway down a rabbit hole reading about having to be cautious with build flags because there is no dedicated spot for the users to put in flags and using existing options risks overwriting flags defined by platform developers. This was no problem at all since I already replaced `mkbuildoptglobals.py` a few days ago to get v3.1.1 to compile in Arduino IDE v1.8 so that I can get to the Exception Decoder with crashing code.

I created the <SketchName>`.ino.globals.h` file in the same location as the sketch source as you've instructed.

I turned on compiler verbosity to make sure the flag got incorporated. I think it did:

```
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Ken\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Ken\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Ken\Documents\Arduino\libraries -fqbn=esp8266:esp8266:d1_mini:xtal=80,vt=flash,exception=disabled,stacksmash=disabled,ssl=all,mmu=3232,non32xfer=fast,eesz=4M2M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600 -vid-pid=1A86_7523 -ide-version=10819 -build-path C:\Users\Ken\AppData\Local\Temp\arduino_build_127341 -warnings=all -build-cache C:\Users\Ken\AppData\Local\Temp\arduino_cache_961296 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.xtensa-lx106-elf-gcc-3.1.0-gcc10.3-e5f9fec.path=C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.mklittlefs.path=C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.mklittlefs-3.1.0-gcc10.3-e5f9fec.path=C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.python3.path=C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.mkspiffs.path=C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.mkspiffs-3.1.0-gcc10.3-e5f9fec.path=C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\3.1.0-gcc10.3-e5f9fec -verbose C:\Users\Ken\Documents\Arduino\sketch_jan24a\sketch_jan24a.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Ken\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Ken\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Ken\Documents\Arduino\libraries -fqbn=esp8266:esp8266:d1_mini:xtal=80,vt=flash,exception=disabled,stacksmash=disabled,ssl=all,mmu=3232,non32xfer=fast,eesz=4M2M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600 -vid-pid=1A86_7523 -ide-version=10819 -build-path C:\Users\Ken\AppData\Local\Temp\arduino_build_127341 -warnings=all -build-cache C:\Users\Ken\AppData\Local\Temp\arduino_cache_961296 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.xtensa-lx106-elf-gcc-3.1.0-gcc10.3-e5f9fec.path=C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.mklittlefs.path=C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.mklittlefs-3.1.0-gcc10.3-e5f9fec.path=C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.python3.path=C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.mkspiffs.path=C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.mkspiffs-3.1.0-gcc10.3-e5f9fec.path=C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\3.1.0-gcc10.3-e5f9fec -verbose C:\Users\Ken\Documents\Arduino\sketch_jan24a\sketch_jan24a.ino
[... cut ...]
0x40216f6a: esp_try_delay(unsigned int, unsigned int, unsigned int) at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_main.cpp line 182
0x40217dc0: __delay(unsigned long) at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266/coredecls.h line 69
0x40203877: updateTVOC() at C:\Users\Ken\Documents\Arduino\sketch_jan24a/sketch_jan24a.ino line 183
0x40204b7b: loop() at C:\Users\Ken\Documents\Arduino\sketch_jan24a/sketch_jan24a.ino line 148
0x40216e50: loop_wrapper() at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_main.cpp line 258
```

mhightower83, 2023-01-29:
Great! And the ISR caller `SoftwareSerial::rxBitISR` that @mcspr suggested is now showing in the trace. The line number isn't exactly where we would expect; however, "in the vicinity" is often what I see.
```
0x401004f1: SoftwareSerial::rxBitISR(SoftwareSerial*) at C:\Users\Ken\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\libraries\SoftwareSerial\src\SoftwareSerial.cpp line 624
```

jjsuwa-sys3175, 2023-01-29:
@mhightower83 
> And the build define `-fno-optimize-sibling-calls` can help push more call details to the stack.

As an implementer of sibling call optimization in xtensa, I'd like to step in off-topic.
I think it should be specified in the documentation (if it exists) as a note when tracing stack dumps.

mhightower83, 2023-01-29:
@jjsuwa-sys3175 I think that is a good suggestion. I too have been thinking it should be added to the docs. I'll take this as a nudge to get it done.

ken830, 2023-01-29:
> Great! And the ISR caller `SoftwareSerial::rxBitISR` that @mcspr suggested is now showing in the trace. The line number isn't exactly where we would expect; however, "in the vicinity" is often what [...]

Just in case, this is from the exact file on my system:

```
622 void IRAM_ATTR SoftwareSerial::rxBitISR(SoftwareSerial* self) {
623    const bool level = *self->m_rxReg & self->m_rxBitMask;
624    const uint32_t curTick = microsToTicks(micros());
625    const bool empty = !self->m_isrBuffer->available();
626
627    // Store level and tick in the buffer unless we have an overflow
628    // tick's LSB is repurposed for the level bit
629    if (!self->m_isrBuffer->push((curTick | 1U) ^ !level)) self->m_isrOverflow.store(true);
630    // Trigger rx callback only when receiver is starved
631    if (empty && self->m_rxHandler) self->m_rxHandler();
632 }
```

I saw the issue created in the espsoftwareserial repository earlier today. Thank you, @mhightower83 and @mcspr for your guidance and patience. I'm completely out of me element here, but I learned a lot! There are a bunch of us that are experiencing this issue so it will help a lot of people in the community when resolved. I'm looking forward to testing whatever solution winds up being implemented.

closed 2023-01-30 by mcspr

ken830, 2023-01-30:
I see we're reverting to SoftwareSerial v6, but does this mean the breaking change in v7 cannot be made to work? Does this affect other Arduino libraries as well like ESP32?

mcspr, 2023-02-01:
ESP32 Does not ship SoftwareSerial, iirc it is up to user to keep it updated.

v7 has reworked SoftwareSerial::onReceive that changed execution context; there is no longer ::perform_work(), callback is executed in SoftwareSerial ISR directly. This is tangentially related, so far the decision is to revert.

PR #8838 (2023-01-30): Revert ESPSoftwareSerial from v7 to v6
Revert ESPSoftwareSerial  6.17.1, as release 7.0.0 has this comment:
["Breaking change, rx interrupt triggered receive callback instead of polling"](https://github.com/plerup/espsoftwareserial/releases/tag/7.0.0)
- should fix #8830 
- should close #8834