esp8266/Arduino #8831: Help identifying problem using Lib EBD + CORE 3.1.0

laercionit, 2023-01-26:
INSTRUCTIONS
============
Usando a 

LIB EBD :  https://github.com/jwhiddon/EDB

#### Platform
- Hardware:		              [ESP-12]
- Core Version:			[3.1.0]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Problem Description

The code example below has always worked great on previous cores. I'm having trouble identifying the problem using ESP8266 CORE 3.1.0. The code is working normally when the DB file is created, but when reusing the file an exeption occurs. I think it might be something related to Append or Write x SEEK of the file using SD CARD.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```#include <EDB.h>
#include <SPI.h>
#include <SD.h>
#define TABLE_SIZE 8192
#define RECORDS_TO_CREATE 100

String db_name = "/TESTE.db";
File dbFile;
struct LogEvent {
    int id;
    int temperature;
}
logEvent;

inline void writer (unsigned long address, const byte* data, unsigned int recsize) {
    dbFile.seek(address);
    dbFile.write(data,recsize);
    dbFile.flush();
}

inline void reader (unsigned long address, byte* data, unsigned int recsize) {
    dbFile.seek(address);
    dbFile.read(data,recsize);
}

// Create an EDB object with the appropriate write and read handlers
EDB db(&writer, &reader);

// Run the demo
void setup()
{
    Serial.begin(115200);
    Serial.println(" Extended Database Library + External SD CARD storage demo");
    Serial.println();
    if (!SD.begin(4)) {
        Serial.println("No SD-card.");
        return;
    }

    if (SD.exists(db_name.c_str())) {
        dbFile = SD.open(db_name.c_str(), FILE_WRITE);
        if (!dbFile) {
            dbFile = SD.open(db_name.c_str(), FILE_WRITE);
        }
        if (dbFile) {
            Serial.print("Openning current table... ");
            EDB_Status result = db.open(0);
            if (result == EDB_OK) {
                Serial.println("DONE");
            } else {
                Serial.println("ERROR");
                Serial.println("Did not find database in the file " + String(db_name));
                Serial.print("Creating new table... ");
                db.create(0, TABLE_SIZE, (unsigned int)sizeof(logEvent));
                Serial.println("DONE");
                return;
            }
        } else {
            Serial.println("Could not open file " + String(db_name));
            return;
        }
    } else {
        Serial.print("Creating table... ");
        // create table at with starting address 0
        dbFile = SD.open(db_name.c_str(), FILE_WRITE);
        db.create(0, TABLE_SIZE, (unsigned int)sizeof(logEvent));
        Serial.println("DONE");
    }

    //recordLimit();
    //countRecords();
    createRecords(RECORDS_TO_CREATE);
    //countRecords();
    selectAll();
    deleteOneRecord(RECORDS_TO_CREATE / 2);
    //countRecords();
    selectAll();
    appendOneRecord(RECORDS_TO_CREATE + 1);
    //countRecords();
    selectAll();
    insertOneRecord(RECORDS_TO_CREATE / 2);
    //countRecords();
    selectAll();
    updateOneRecord(RECORDS_TO_CREATE);
    selectAll();
    //countRecords();
    deleteAll();
    Serial.println("Use insertRec() and deleteRec() carefully, they can be slow");
    //countRecords();
    for (int i = 1; i <= 20; i++) insertOneRecord(1);  // inserting from the beginning gets slower and slower
    //countRecords();
    for (int i = 1; i <= 20; i++) deleteOneRecord(1);  // deleting records from the beginning is slower than from the end
    //countRecords();

    dbFile.close();
}

void loop()
{
}

// utility functions

void recordLimit()
{
    Serial.print("Record Limit: ");
    Serial.println(db.limit());
}

void deleteOneRecord(int recno)
{
    Serial.print("Deleting recno: ");
    Serial.println(recno);
    db.deleteRec(recno);
}

void deleteAll()
{
    Serial.print("Truncating table... ");
    db.clear();
    Serial.println("DONE");
}

void countRecords()
{
    Serial.print("Record Count: ");
    Serial.println(db.count());
}

void createRecords(int num_recs)
{
    Serial.print("Creating Records... ");
    for (int recno = 1; recno <= num_recs; recno++)
    {
        logEvent.id = recno;
        logEvent.temperature = random(1, 125);
        EDB_Status result = db.appendRec(EDB_REC logEvent);
        if (result != EDB_OK) printError(result);
    }
    Serial.println("DONE");
}

void selectAll()
{
    for (int recno = 1; recno <= db.count(); recno++)
    {
        EDB_Status result = db.readRec(recno, EDB_REC logEvent);
        if (result == EDB_OK)
        {
            Serial.print("Recno: ");
            Serial.print(recno);
            Serial.print(" ID: ");
            Serial.print(logEvent.id);
            Serial.print(" Temp: ");
            Serial.println(logEvent.temperature);
        }
        else printError(result);
    }
}

void updateOneRecord(int recno)
{
    Serial.print("Updating record at recno: ");
    Serial.print(recno);
    Serial.print("... ");
    logEvent.id = 1234;
    logEvent.temperature = 4321;
    EDB_Status result = db.updateRec(recno, EDB_REC logEvent);
    if (result != EDB_OK) printError(result);
    Serial.println("DONE");
}

void insertOneRecord(int recno)
{
    Serial.print("Inserting record at recno: ");
    Serial.print(recno);
    Serial.print("... ");
    logEvent.id = recno;
    logEvent.temperature = random(1, 125);
    EDB_Status result = db.insertRec(recno, EDB_REC logEvent);
    if (result != EDB_OK) printError(result);
    Serial.println("DONE");
}

void appendOneRecord(int id)
{
    Serial.print("Appending record... ");
    logEvent.id = id;
    logEvent.temperature = random(1, 125);
    EDB_Status result = db.appendRec(EDB_REC logEvent);
    if (result != EDB_OK) printError(result);
    Serial.println("DONE");
}

void printError(EDB_Status err)
{
    Serial.print("ERROR: ");
    switch (err)
    {
        case EDB_OUT_OF_RANGE:
            Serial.println("Recno out of range");
            break;
        case EDB_TABLE_FULL:
            Serial.println("Table full");
            break;
        case EDB_OK:
        default:
            Serial.println("OK");
            break;
    }
}
```

### Debug Messages

```
Debug messages go here
```

d-a-v, 2023-01-26:
> Debug messages go here

This would help alot to understand your issue:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

laercionit, 2023-01-26:
> > Debug messages go here
> 
> This would help alot to understand your issue:
> 
> https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html
> 
> https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

@d-a-v , 
Thanks for the guidance. Sorry for the insufficient detail. Include the information you requested.

laercionit, 2023-01-26:
I made the change from FILE_WRITE to "a+" and it solved the problem.

laercionit, 2023-01-26:
Apparently there was a change in CORE for the FILE_WRITE cursor position when opening the file.

mcspr, 2023-01-27:
> I made the change from FILE_WRITE to "a+" and it solved the problem.

SdFat <-> FS HAL mode translation is not correctly setting 'read' mode, as it turns out.
https://github.com/earlephilhower/ESP8266SdFat/blob/43a65b42d5a827092188cfe11fb27237da252692/src/common/FsApiConstants.h#L47
Our logic uses `O_READ` (aka `O_RDONLY`, aka `0`) as a mask, which **is always false** :/
https://github.com/esp8266/Arduino/blob/e25f9e95783d6de59d6b3b7b4ebc029428c9b539/libraries/SD/src/SD.h#L162-L164
https://github.com/esp8266/Arduino/blob/e25f9e95783d6de59d6b3b7b4ebc029428c9b539/libraries/SD/src/SD.h#L167
https://github.com/esp8266/Arduino/blob/e25f9e95783d6de59d6b3b7b4ebc029428c9b539/libraries/SD/src/SD.h#L169

> Apparently there was a change in CORE for the FILE_WRITE cursor position when opening the file.

Do you have an example of mode and position before and after?

PR #8833 (2023-01-31): SdFat -> FS HAL mode fixes & test
* re-use SdFat access mode through static const, no need to hard-code our own value w/ cast in the macro
* separate access-modes from flags
* simple compile-time tests in .cpp

resolve #8831
commit 6dfebec8 (2023-01-31): SdFat -> FS HAL mode fixes & test (#8833)

* re-use SdFat access mode through static const, no need to hard-code our own value w/ cast in the macro
* separate read-modes from flags; read, write and rw are distinct numbers
* simple compile-time tests in .cpp

resolve #8831
closed 2023-01-31 by mcspr