esp8266/Arduino #8836: Irrelevant WiFiClient disconnected by ESPhttpUpdate.update

snakeninny, 2023-01-29:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware: ESP-12F
- Core Version: 3.0.2 I guess?
- Development Env: PlatformIO
- Operating System: macOS

### Settings in platformio.ini

```
[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
monitor_speed = 115200
lib_deps = 
	knolleary/PubSubClient@^2.8
```

### Problem Description

In my code I have `WiFiClient espClient` and `WiFiClientSecure updateClient`, and they have irrelevant usage.
When `updateClient` is used by `ESPhttpUpdate.update`, `espClient` is disconnected in `ESPhttpUpdate.onProgress`, as you can test in the sketch:

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```
#include "Arduino.h"
#include <ESP8266WiFi.h>
#include <ESP8266httpUpdate.h>
#include <PubSubClient.h>

#define ssid "ssid" // Change this
#define password "password" // Change this

String upUrl = "https://bin.bemfa.com/b/3BcN2Q1NGY4NWFmNDI5NzZlZTNjMjY5M2U2OTJhNmJiNTk=light002.bin";
char serverName[] = "baidu.com";

WiFiClient espClient;

void onStart()
{
  Serial.println("onStart");
  Serial.printf("WiFiClient is %sconnected.\r\n", espClient.connected() ? "" : "NOT ");
}

void onEnd()
{
  Serial.println("onEnd");
}

void onProgress(int cur, int total)
{
  Serial.printf("onProgress at %d of %d bytes.\r\n", cur, total);
  Serial.printf("WiFiClient is %sconnected.\r\n", espClient.connected() ? "" : "NOT ");
}

void onError(int err)
{
  Serial.printf("onError, err code = %d.\r\n", err);
}

void updateBin()
{
  WiFiClientSecure updateClient;
  updateClient.setInsecure();

  ESPhttpUpdate.onStart(onStart);
  ESPhttpUpdate.onEnd(onEnd);
  ESPhttpUpdate.onProgress(onProgress);
  ESPhttpUpdate.onError(onError);

  t_httpUpdate_return ret = ESPhttpUpdate.update(updateClient, upUrl);
  switch (ret)
  {
  case HTTP_UPDATE_FAILED:
    Serial.println("HTTP_UPDATE_FAILED");
    break;
  case HTTP_UPDATE_NO_UPDATES:
    Serial.println("HTTP_UPDATE_NO_UPDATES");
    break;
  case HTTP_UPDATE_OK:
    Serial.println("HTTP_UPDATE_OK");
    break;
  }
}

void setup()
{
  Serial.begin(115200);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");

  if (espClient.connect(serverName, 443))
  {
    if (espClient.connected())
    {
      Serial.println("Connected to https://baidu.com.");
    }
    updateBin();
  }
}

void loop()
{
}
```

### Log

```
...........
Connected to https://baidu.com.
onStart
WiFiClient is connected.
onProgress at 0 of 305120 bytes.
WiFiClient is NOT connected.
onProgress at 0 of 305120 bytes.
WiFiClient is NOT connected.
onProgress at 4096 of 305120 bytes.
WiFiClient is NOT connected.
...
```

Did I miss anything or this is normal?
How do I keep `espClient` CONNECTED during `ESPhttpUpdate.update`? Thank you.

mcspr, 2023-01-29:
> How do I keep espClient CONNECTED during ESPhttpUpdate.update? Thank you.

Should be fixed ≥3.1.0, see https://github.com/esp8266/Arduino/pull/8598

closed 2023-01-29 by mcspr

snakeninny, 2023-01-29:
@mcspr Thanks for the info!
But the bug persists for me:
```
 *  Executing task: platformio run 

Processing nodemcuv2 (platform: espressif8266; board: esp12e; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 (4.1.0) > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 @ 3.30101.0 (3.1.1) 
 - tool-esptool @ 1.413.0 (4.13) 
 - tool-esptoolpy @ 1.30000.201119 (3.0.0) 
 - toolchain-xtensa @ 2.100300.220621 (10.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- PubSubClient @ 2.8.0
|-- ESP8266WiFi @ 1.0
|-- ESP8266httpUpdate @ 1.3
|   |-- ESP8266HTTPClient @ 1.2
|   |   |-- ESP8266WiFi @ 1.0
|   |-- ESP8266WiFi @ 1.0
Building in release mode
...
.......
Connected to https://baidu.com.
onStart
WiFiClient is connected.
onProgress at 0 of 305120 bytes.
WiFiClient is NOT connected.
onProgress at 0 of 305120 bytes.
WiFiClient is NOT connected.
...
```
Any ideas?

mcspr, 2023-01-29:
Ah, it is not server.

https://github.com/esp8266/Arduino/issues/3359 -> https://github.com/esp8266/Arduino/pull/6969
`ESPhttpUpdate.closeConnectionsOnUpdate(false);`

https://github.com/esp8266/Arduino/blob/e25f9e95783d6de59d6b3b7b4ebc029428c9b539/libraries/ESP8266httpUpdate/src/ESP8266httpUpdate.h#L100