esp8266/Arduino #8855: Issue loading LittleFS files  (ERR_CONTENT_LENGTH_MISMATCH)

infrafast, 2023-02-13:
### Basic Infos

- [X ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X ] I have tested that the issue is present in current master branch (aka latest git).
- [X ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ X] I have filled out all fields below.

#### Platform

- Hardware:				ESP8266
- Core Version:			framework-arduinoespressif8266 @ 3.30101.0 (3.1.1)
- Development Env:		PlatformIO
- Operating System:		Windows

### Settings in IDE

- Module:  			Espressif 8266 (4.1.0) > WeMos D1 R2 and mini
- Flash Mode:			
- Flash Size: 		4M
- lwip Variant:   default
- Reset Method:		
- Flash Frequency:	[40Mhz]
- CPU Frequency:		80Mhz
- Upload Using:		serial + OTA
- Upload Speed:   115200

### Problem Description

Detailed problem description goes here.

I have a device that works both with WiFi and Ethernet using W5500 library and esp8266webserver.
Serving page works ok with WiFI but when using W5500 I have the similar problem as the one described in #2121
This happens randomly n some page, one time it can load, the other it will not and another page will fail.

[image]

I follow all threads about this issue both in github and other places and I made test with and without gziping the document, with and without merging my javascript files, with small (2/3k) and bigger file (30k). 
I monitored the freeheap before and after pages are served and always have a reasonnable amount of memory available

The problem happens both when using "server.serveStatic" and "server.on"

I presume that the fix that was apply for the WiFi core has not been done for the W5500 part.

Hence I suspect the streamFile method template to be the cause.

I provide some extract of the code which are representative of the function calls (not the full one has it is too big).

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
#include <ESP8266WebServer.h>
#include <W5500lwIP.h>        

ESP8266WebServer server;
Wiznet5500lwIP ethDriver(D2);   

void streamJavascript(String path) {
  streamFile(path,"application/javascript");
}

void streamCSS(String path){
  streamFile(path,"text/css");
}

void handleIndexJS() {
  streamJavascript("/index.min.js.gz");
}

void handleStyleCSS() {
  streamCSS("/style.min.css.gz");
}

bool streamFile(String path, String dataType) {
  DEBUG_PRINT("\nFree heap() before call: %i",ESP.getFreeHeap());
  DEBUG_PRINT("\nRequested page -> %S",path.c_str());
  if (LittleFS.exists(path)){
      File dataFile = LittleFS.open(path, "r");
      if (!dataFile) {
          DEBUG_PRINT("\nNot found");
          return false;
      }
      uint16_t dataSentSize = server.streamFile(dataFile, dataType);
      if (dataSentSize != dataFile.size()) {
        DEBUG_PRINT("\nSent less data (%i) than expected! (%i)",dataSentSize,dataFile.size());

      }else{
          DEBUG_PRINT("\nPage served!");
      }
      DEBUG_PRINT("\nFree heap() after call: %i",ESP.getFreeHeap());
      dataFile.close();
  }else{
      DEBUG_PRINT("\nNot found");
      return false;
  }
  return true;
}

void setup() {
      ethDriver.begin(mac);         
      loopElapsedTime = millis();
      while (!ethDriver.connected() && (millis() - loopElapsedTime < ETHCONNECTTIMEOUT)) {
        //DEBUG_PRINT(".");          
      }
    server.on("/index.min.js", handleIndexJS);
    server.on("/style.min.css", handleStyleCSS);
}

void loop() {
   server.handleClient();
}

```

### Debug Messages

```
Free heap() before call: 11856
Requested page -> /style.min.css.gz
Page served!
Free heap() after call: 10232
Free heap() before call: 12504
Requested page -> /index.min.js.gz
Sent less data (4166) than expected! (6808)
Free heap() after call: 9832
```

infrafast, 2023-02-13:
Ok, I found the solution:
it's a timeout problem, 

on ESP8266WebServer.h => contentLength = file.sendAll(_currentClient,5000);  
instead of file.sendAll(_currentClient); 

one may consider this to be fixed for next releases ?

mcspr, 2023-02-13:
Do you mean that 5000ms timeout in the server code fixed the problem?
https://github.com/esp8266/Arduino/blob/d7cd4bef6c15165a181f9c65e549c2c5c356e014/libraries/ESP8266WebServer/src/ESP8266WebServer-impl.h#L510

Since neverExpires is implicitly added without argument, we end up with whatever set to the object on which we use sendAll()
https://github.com/esp8266/Arduino/blob/d7cd4bef6c15165a181f9c65e549c2c5c356e014/cores/esp8266/Stream.h#L178
https://github.com/esp8266/Arduino/blob/d7cd4bef6c15165a181f9c65e549c2c5c356e014/cores/esp8266/StreamSend.cpp#L235-L237
https://github.com/esp8266/Arduino/blob/d7cd4bef6c15165a181f9c65e549c2c5c356e014/cores/esp8266/Stream.h#L52

infrafast, 2023-02-13:
yes, I've put 5000 and this definitely fixed the problem

d-a-v, 2023-02-13:
Considering the "streamSend" API, timeout is currently the input stream's timeout.
The output stream's timeout is not considered (`Print* to` in StreamSend.cpp). In your case, the output is network (webserver's client) which default is already 5000ms.
Using a timeout of max(input-timeout,output-timeout) would be more coherent and should solve your issue.
I can prepare a proposal for this.

infrafast, 2023-02-16:
Great if you provide a fix, I'd be happy to retest my code with it and validate the use case

d-a-v, 2023-03-01:
If you are using the git version of this core, can you try with #8874 ?

infrafast, 2023-03-07:
I just had a look but I am using platformIO, not the Arduino IDE,  do oyu know by chance the lines I should add in my platformio.ini file to add the platform in a new environement ?

d-a-v, 2023-03-20:
@infrafast Have you had success trying with #8874 ?

mcspr, 2023-03-20:
> I just had a look but I am using platformIO, not the Arduino IDE, do oyu know by chance the lines I should add in my platformio.ini file to add the platform in a new environement ?

https://docs.platformio.org/en/stable/platforms/espressif8266.html#using-arduino-framework-with-staging-version

infrafast, 2023-03-21:
perfect thanks for the link, I will setup my staging environment and test then feedback here asap

infrafast, 2023-03-26:
> @infrafast Have you had success trying with #8874 ?

since I don't know how to connect my platformIO to the git repo with this fix, could you rather send me the 4 files that you changd and I will update my local folder with them ? => info@infrafast.com thx!

infrafast, 2023-03-27:
Thanks, with your instruction I learned how to setup my env and manage to test :-) !

Here are the results:
- on 2 differents PC, tested with 
     - chrome (macos and windows)
     - safari (ubuntu, macos, windows)
- using WiFi and Ethernet connectivity
- page load total ~12seconds
[image]
I've performed in total about 20 loads and never had the issue.
Without the fix, the problem manifest about 70% of the time.
I can state that this works very well and it seems to even faster the page loading. 
Thank you very much for this fix and looking forward to have it integrated into the main branch.

PR #8874 (2023-03-28): deprecate and update `Stream::send*(Print -> Stream)`
Deprecate and update `Stream::send*(Print -> Stream)` in order to benefit from, and use, output Stream's timeout value.

Fixes #8855
closed 2023-03-28 by d-a-v