esp8266/Arduino #8856: Boards Manager upgrade ESP8266 Core 3.0.2 to 3.1.0/3.1.1 fails to compile - error missing CommonHFile.h

TobbeG, 2023-02-13:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform
- Hardware:			       ESP-01
- Core Version:		       All works ok with 3.0.2 but fails to compile with 3.1.0/3.1.1
- Development Env:	          Arduino IDE 1.8.16
- Operating System:		Windows11

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			qio
- Flash Size: 		       4MB or 1MB
- lwip Variant:                  Higher Bandwidth
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	        40Mhz
- CPU Frequency:		160MHz
- Upload Using:		SERIAL
- Upload Speed:              921600

### Problem Description
Cannot compile code with core version 3.1.0 or 3.1.1
Core 3.0.2 works fine

Detailed problem description goes here.
When compling I get error message of missing file:       CommonHFile.h

Cannot even compile simples Blink.c example.

### Debug Messages
ERROR MESSAGE
cc1plus.exe: fatal error: C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.0\cores\esp8266\CommonHFile.h: No such file or directory
compilation terminated.
exit status 1
Error compiling for board Generic ESP8266 Module.

Rainbow-Cai, 2023-02-15:
Hi @igrr  , Could you please update this question?

Rainbow-Cai, 2023-02-15:
Hi  @TobbeG   This is a third party project, not something that is supported by Espressif.  esp8266/Arduino is a third party project, no one in Espressif is working on it.

mcspr, 2023-02-15:
Duplicate of #8811?
More build log lines please? Is it python error, compiler one, arduino-{builder,cli}?

mcspr, 2023-02-16:
Please show the full build log

TobbeG, 2023-02-16:
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Thorbjörn Geiser O\Dropbox\Otrobanda\IOT\MyArduinoSketches\libraries -fqbn=esp8266:esp8266:generic:xtal=160,vt=flash,exception=disabled,stacksmash=disabled,ssl=all,mmu=3232,non32xfer=fast,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,eesz=1M64,led=2,sdk=nonosdk_191122,ip=hb2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600 -ide-version=10816 -build-path C:\Users\THORBJ~1\AppData\Local\Temp\arduino_build_113882 -warnings=all -build-cache C:\Users\THORBJ~1\AppData\Local\Temp\arduino_cache_99336 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.python3.path=C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.mkspiffs.path=C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.mkspiffs-3.1.0-gcc10.3-e5f9fec.path=C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.mklittlefs.path=C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.mklittlefs-3.1.0-gcc10.3-e5f9fec.path=C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.xtensa-lx106-elf-gcc-3.1.0-gcc10.3-e5f9fec.path=C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.1.0-gcc10.3-e5f9fec -verbose C:\Users\Thorbjörn Geiser O\Dropbox\Otrobanda\IOT\MyArduinoSketches\OHMIGO.mini_Webserver\OHMIGO.mini_Webserver.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Thorbjörn Geiser O\Dropbox\Otrobanda\IOT\MyArduinoSketches\libraries -fqbn=esp8266:esp8266:generic:xtal=160,vt=flash,exception=disabled,stacksmash=disabled,ssl=all,mmu=3232,non32xfer=fast,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,eesz=1M64,led=2,sdk=nonosdk_191122,ip=hb2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600 -ide-version=10816 -build-path C:\Users\THORBJ~1\AppData\Local\Temp\arduino_build_113882 -warnings=all -build-cache C:\Users\THORBJ~1\AppData\Local\Temp\arduino_cache_99336 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.python3.path=C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.mkspiffs.path=C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.mkspiffs-3.1.0-gcc10.3-e5f9fec.path=C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.mklittlefs.path=C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.mklittlefs-3.1.0-gcc10.3-e5f9fec.path=C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.xtensa-lx106-elf-gcc-3.1.0-gcc10.3-e5f9fec.path=C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.1.0-gcc10.3-e5f9fec -verbose C:\Users\Thorbjörn Geiser O\Dropbox\Otrobanda\IOT\MyArduinoSketches\OHMIGO.mini_Webserver\OHMIGO.mini_Webserver.ino
Using board 'generic' from platform in folder: C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1
Using core 'esp8266' from platform in folder: C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1
"C:\\Users\\Thorbjörn Geiser O\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\python3\\3.7.2-post1/python3" -I "C:\\Users\\Thorbjörn Geiser O\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.1.1/tools/mkbuildoptglobals.py" "C:\\Program Files (x86)\\Arduino" 10816 "C:\\Users\\THORBJ~1\\AppData\\Local\\Temp\\arduino_build_113882" "C:\\Users\\THORBJ~1\\AppData\\Local\\Temp\\arduino_build_113882/core/build.opt" "C:\\Users\\Thorbjörn Geiser O\\Dropbox\\Otrobanda\ [... cut ...] 5\\packages\\esp8266\\hardware\\esp8266\\3.1.1\\variants\\generic" "C:\\Users\\THORBJ~1\\AppData\\Local\\Temp\\arduino_build_113882\\sketch\\OHMIGO.mini_Webserver.ino.cpp" -o nul
Alternatives for C: []
ResolveLibrary(C)
  -> candidates: []
cc1plus.exe: fatal error: C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\CommonHFile.h: No such file or directory
compilation terminated.
exit status 1
Error compiling for board Generic ESP8266 Module.

TobbeG, 2023-02-16:
[Arduino 1.8.16 Compiler output with ESP8266 core 3.11.txt](https://github.com/esp8266/Arduino/files/10754606/Arduino.1.8.16.Compiler.output.with.ESP8266.core.3.11.txt)

mcspr, 2023-02-16:
Thanks!
We'd expect that mkbuildoptsglobal.py script creates that file before building.

Does it work if you replace "C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1/tools/mkbuildoptglobals.py" with our latest version from https://raw.githubusercontent.com/esp8266/Arduino/master/tools/mkbuildoptglobals.py ?

TobbeG, 2023-02-16:
No difference.

Arduino:1.8.16 (Windows 10), Kort:"Generic ESP8266 Module, 160 MHz, Flash, Disabled (new aborts on oom), Disabled, All SSL ciphers (most compatible), 32KB cache + 32KB IRAM (balanced), Use pgm_read macros for IRAM/PROGMEM, dtr (aka nodemcu), 26 MHz, 40MHz, DOUT (compatible), 1MB (FS:64KB OTA:~470KB), 2, nonos-sdk 2.2.1+119 (191122), v2 Higher Bandwidth, Disabled, None, Only Sketch, 921600"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Thorbjörn Geiser O\Dropbox\Otrobanda\IOT\MyArduinoSketches\libraries -fqbn=esp8266:esp8266:generic:xtal=160,vt=flash,exception=disabled,stacksmash=disabled,ssl=all,mmu=3232,non32xfer=fast,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,eesz=1M64,led=2,sdk=nonosdk_191122,ip=hb2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600 -vid-pid=10C4_EA60 -ide-version=10816 -build-path C:\Users\THORBJ~1\AppData\Local\Temp\arduino_build_717766 -warnings=all -build-cache C:\Users\THORBJ~1\AppData\Local\Temp\arduino_cache_709172 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.python3.path=C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.xtensa-lx106-elf-gcc-3.1.0-gcc10.3-e5f9fec.path=C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.mkspiffs.path=C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.mkspiffs-3.1.0-gcc10.3-e5f9fec.path=C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.mklittlefs.path=C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.mklittlefs-3.1.0-gcc10.3-e5f9fec.path=C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\3.1.0-gcc10.3-e5f9fec -verbose C:\Users\Thorbjörn Geiser O\Dropbox\Otrobanda\IOT\MyArduinoSketches\OHMIGO.mini_Webserver\OHMIGO.mini_Webserver.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Thorbjörn Geiser O\Dropbox\Otrobanda\IOT\MyArduinoSketches\libraries -fqbn=esp8266:esp8266:generic:xtal=160,vt=flash,exception=disabled,stacksmash=disabled,ssl=all,mmu=3232,non32xfer=fast,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,eesz=1M64,led=2,sdk=nonosdk_191122,ip=hb2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600 -vid-pid=10C4_EA60 -ide-version=10816 -build-path C:\Users\THORBJ~1\AppData\Local\Temp\arduino_build_717766 -warnings=all -build-cache C:\Users\THORBJ~1\AppData\Local\Temp\arduino_cache_709172 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.python3.path=C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.xtensa-lx106-elf-gcc-3.1.0-gcc10.3-e5f9fec.path=C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.mkspiffs.path=C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.mkspiffs-3.1.0-gcc10.3-e5f9fec.path=C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.mklittlefs.path=C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.mklittlefs-3.1.0-gcc10.3-e5f9fec.path=C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\3.1.0-gcc10.3-e5f9fec -verbose C:\Users\Thorbjörn Geiser O\Dropbox\Otrobanda\IOT\MyArduinoSketches\OHMIGO.mini_Webserver\OHMIGO.mini_Webserver.ino

Using board 'generic' from platform in folder: C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1

Using core 'esp8266' from platform in folder: C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1

[... cut ...] ages\\esp8266\\hardware\\esp8266\\3.1.1\\variants\\generic" "C:\\Users\\THORBJ~1\\AppData\\Local\\Temp\\arduino_build_717766\\sketch\\OHMIGO.mini_Webserver.ino.cpp" -o nul

Alternatives for C: []

ResolveLibrary(C)

  -> candidates: []

cc1plus.exe: fatal error: C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\CommonHFile.h: No such file or directory

compilation terminated.

exit status 1

Error compiling for board Generic ESP8266 Module.

RastoH, 2023-02-16:
problem due to diacritics

C:\Users\Thorbjörn Geiser O\AppData\.........
C:\Users\THORBJ~1\AppData\.....

I had to make a new account without. Win11

TobbeG, 2023-02-16:
> problem due to diacritics
> 
> C:\Users\Thorbjörn Geiser O\AppData......... C:\Users\THORBJ~1\AppData.....
> 
> I had to make a new account without. Win11

I do not really understand - what do you mean?

mhightower83, 2023-02-16:
@TobbeG after a failed build, locate the file `...\core\build.opt` in your temp directory. I think it may help show if path strings are getting mangled.

TobbeG, 2023-02-16:
This is it - appreciate all the efforts in helping to solve this asap.

{
  "additionalFiles": "..,..,..,..,..,..,..,..",
  "builtInLibrariesFolders": "C:\\Program Files (x86)\\Arduino\\libraries",
  "builtInToolsFolders": "C:\\Program Files (x86)\\Arduino\\tools-builder,C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr,C:\\Users\\Thorbjörn Geiser O\\AppData\\Local\\Arduino15\\packages",
  "compiler.optimization_flags": "",
  "customBuildProperties": "build.warn_data_percentage=75,runtime.tools.xtensa-lx106-elf-gcc.path=C:\\Users\\Thorbjörn Geiser O\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\3.1.0-gcc10.3-e5f9fec,runtime.tools.xtensa-lx106-elf-gcc-3.1.0-gcc10.3-e5f9fec.path=C:\\Users\\Thorbjörn Geiser O\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\3.1.0-gcc10.3-e5f9fec,runtime.tools.python3.path=C:\\Users\\Thorbjörn Geiser O\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\python3\\3.7.2-post1,runtime.tools.python3-3.7.2-post1.path=C:\\Users\\Thorbjörn Geiser O\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\python3\\3.7.2-post1,runtime.tools.mkspiffs.path=C:\\Users\\Thorbjörn Geiser O\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\mkspiffs\\3.1.0-gcc10.3-e5f9fec,runtime.tools.mkspiffs-3.1.0-gcc10.3-e5f9fec.path=C:\\Users\\Thorbjörn Geiser O\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\mkspiffs\\3.1.0-gcc10.3-e5f9fec,runtime.tools.mklittlefs.path=C:\\Users\\Thorbjörn Geiser O\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\mklittlefs\\3.1.0-gcc10.3-e5f9fec,runtime.tools.mklittlefs-3.1.0-gcc10.3-e5f9fec.path=C:\\Users\\Thorbjörn Geiser O\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\mklittlefs\\3.1.0-gcc10.3-e5f9fec",
  "fqbn": "esp8266:esp8266:generic:xtal=160,vt=flash,exception=disabled,stacksmash=disabled,ssl=all,mmu=3232,non32xfer=fast,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,eesz=4M2M,led=2,sdk=nonosdk_191122,ip=hb2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600",
  "hardwareFolders": "C:\\Program Files (x86)\\Arduino\\hardware,C:\\Users\\Thorbjörn Geiser O\\AppData\\Local\\Arduino15\\packages",
  "otherLibrariesFolders": "C:\\Users\\Thorbjörn Geiser O\\Dropbox\\Otrobanda\\IOT\\MyArduinoSketches\\libraries",
  "runtime.ide.version": "10816",
  "sketchLocation": "C:\\Users\\Thorbjörn Geiser O\\Dropbox\\Otrobanda\\IOT\\MyArduinoSketches\\Ohmigo.ohmonwifi\\Ohmigo.ohmonwifi.ino"
}

[build.options.txt](https://github.com/esp8266/Arduino/files/10758871/build.options.txt)

mhightower83, 2023-02-16:
Almost, starting from the directory location of that file we need the file `.\core\build.opt`.

However, I think GCC is selectively having issues with long file names. From your previous post with the line strict through indicate embedded `~`s in the text. Some of the built-in paths I am passed are full paths and some are windows short paths.  I think the short paths with `~` are working. The long path that includes `ö` are failing.  This is what @RastoH was getting at, thanks!

TobbeG, 2023-02-16:
Is there an easy way renaming my W11 useraccount from "Thorbjörn..." to something without special characters ?

mhightower83, 2023-02-16:
There should be a way to get short names in python. I just don't know it yet and I need to set up a test machine.

If you are up for a quick test. You could add these two lines to `platform.txt`. I am hoping the extra `.windows` annotation will persuade Arduino to do more with the paths.
```
globals.h.source.fqfn.windows={build.source.path}\{build.project_name}.globals.h
commonhfile.fqfn.windows={build.core.path}\CommonHFile.h
```

TobbeG, 2023-02-16:
> There should be a way to get short names in python. I just don't know it yet and I need to set up a test machine.
>
> If you are up for a quick test. You could add these two lines to `platform.txt`. I [...]

I put it here - but same error.
 ESP8266 platform
# ------------------------------

# For more info:
# https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification

name=ESP8266 Boards (3.1.1)
version=3.1.1

# These will be removed by the packager script when doing a JSON release

**_globals.h.source.fqfn.windows={build.source.path}\{build.project_name}.globals.h
commonhfile.fqfn.windows={build.core.path}\CommonHFile.h_**

runtime.tools.signing={runtime.platform.path}/tools/signing.py
runtime.tools.elf2bin={runtime.platform.path}/tools/elf2bin.py
runtime.tools.sizes={runtime.platform.path}/tools/sizes.py

mcspr, 2023-02-17:
https://raw.githubusercontent.com/mhightower83/Arduino/pr-mkbuildopt-path/tools/mkbuildoptglobals.py ?
(from #8860)

TobbeG, 2023-02-17:
> https://raw.githubusercontent.com/mhightower83/Arduino/pr-mkbuildopt-path/tools/mkbuildoptglobals.py ? (from #8860)

What is this about?

mhightower83, 2023-02-17:
@TobbeG This should resolve your build issue. Save the file to your disk. Backup and replace your current copy at
`C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1/tools/mkbuildoptglobals.py`

RastoH, 2023-02-18:
win11pro
Arduino IDE 2.0.3

[mkbuildoptglobals.py](https://raw.githubusercontent.com/mhightower83/Arduino/pr-mkbuildopt-path/tools/mkbuildoptglobals.py)  (from https://github.com/esp8266/Arduino/pull/8860)

Using board 'd1_mini' from platform in folder: C:\Users\Rasťo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1
Using core 'esp8266' from platform in folder: C:\Users\Rasťo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1
"C:\\Users\\Rasťo\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\python3\\3.7.2-post1/python3" -I "C:\\Users\\Rasťo\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.1.1/tools/makecorever.py" --build_path "C:\\Users\\Rasťo\\AppData\\Local\\Temp\\arduino-sketch-66355F131017B1A8E3106F77A9989B6C" --platform_path "C:\\Users\\Rasťo\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.1.1" --version 3.2.0-dev
"C:\\Users\\Rasťo\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\python3\\3.7.2-post1/python3" -I "C:\\Users\\Rasťo\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.1.1/tools/mkbuildoptglobals.py" "C:\\Program Files\\Arduino IDE\\resources\\app\\node_modules\\arduino-ide-extension\\build" 10607 "C:\\Users\\Rasťo\\AppData\\Local\\Temp\\arduino-sketch-66355F131017B1A8E3106F77A9989B6C" "C:\\Users\\Rasťo\\AppData\\Local\\Temp\\arduino-sketch-66355F131017B1A8E3106F77A9989B6C/core/build.opt" "C:\\ARDUINO\\Kurnik/Kurnik.ino.globals.h" "C:\\Users\\Rasťo\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.1.1\\cores\\esp8266/CommonHFile.h"
Assume aggressive 'core.a' caching enabled.
Note: optional global include file 'C:\ARDUINO\Kurnik\Kurnik.ino.globals.h' does not exist.
  Read more at https://arduino-esp8266.readthedocs.io/en/latest/faq/a06-global-build-options.html
Detecting libraries used...
"C:\\Users\\Rasťo\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\3.1.0-gcc10.3-e5f9fec/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -D_GNU_SOURCE -DESP8266 "@C:\\Users\\Rasťo\\AppData\\Local\\Temp\\arduino-sketch-66355F131017B1A8E3106F77A9989B6C/core/build.opt" "-IC:\\Users\\Rasťo\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.1.1/tools/sdk/include" "-IC:\\Users\\Rasťo\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.1.1/tools/sdk/lwip2/include" "-IC:\\Users\\Rasťo\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.1.1/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\Rasťo\\AppData\\Local\\Temp\\arduino-sketch-66355F131017B1A8E3106F77A9989B6C/core" -c "@C:\\Users\\Rasťo\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.1.1/tools/warnings/none-cppflags" -Os -g -free -fipa-pta -Werror=return-type -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++17 -ffunction-sections -fdata-sections -fno-exceptions -DMMU_IRAM_SIZE=0x8000 -DMMU_ICACHE_SIZE=0x8000 -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=1460 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10607 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_WEMOS_D1MINI\"" "-DARDUINO_BOARD_ID=\"d1_mini\"" -DFLASHMODE_DIO "-IC:\\Users\\Rasťo\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.1.1\\cores\\esp8266" "-IC:\\Users\\Rasťo\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.1.1\\variants\\d1_mini" "C:\\Users\\Rasťo\\AppData\\Local\\Temp\\arduino-sketch-66355F131017B1A8E3106F77A9989B6C\\sketch\\Kurnik.ino.cpp" -o nul
Alternatives for C: []
ResolveLibrary(C)
  -> candidates: []
cc1plus.exe: fatal error: C:\Users\Rasťo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\CommonHFile.h: No such file or directory
compilation terminated.

exit status 1

Compilation error: exit status 1

always problem with diacritics.

mcspr, 2023-02-18:
C:\Users\*\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\CommonHFile.h is present, though? Script does create something at the target directory?

@mhightower83 
https://docs.python.org/3/library/os.html#os.fsencode
https://docs.python.org/3/library/sys.html#sys.getfilesystemencoding
perhaps the .opt file should have the fs encoding instead of utf8?

RastoH, 2023-02-18:
Present, is empty (size = 0), i dont use.
Works fine without diacritics (as another user, created for compilation only)

mhightower83, 2023-02-18:
Can you get to the `C:\Users\Rasťo\AppData\Local\Temp\arduino-sketch-66355F131017B1A8E3106F77A9989B6C/core/build.opt` file and post that?

RastoH, 2023-02-19:
-include "C:\\Users\\Rasťo\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.1.1\\cores\\esp8266\\CommonHFile.h"
-include "C:\\Users\\Rasťo\\AppData\\Local\\Temp\\arduino-sketch-66355F131017B1A8E3106F77A9989B6C\\core\\Kurnik.ino.globals.h"

mhightower83, 2023-02-20:
Creating an empty version of `Kurnik.ino.globals.h` simplifies getting dependencies right so that a rebuild occurs if/when the file is later added. `CommonHFile.h` is often empty the timestamps on the file help control triggering the rebuild of core files.

I am still looking for a way to encode the file paths such that they will work in more than one locale.

mhightower83, 2023-02-21:
I succeeded in creating an account with a diacritic in the user ID on a US version of Windows.  So the problem was much easier to reproduce and resolve.  

For Arduino IDE v2.0.3,  an additional issue shows up related to diacritics that did not show in Arduino v1.8. The older one used a mixture of Windows short names and full paths. Arduino IDE v2.0.3 uses only full paths exposing diacritics all the time.

`sizes.py` failure on Arduino IDE v2.0.3  
```
Traceback (most recent call last):
  File "C:\Users\testör\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1/tools/sizes.py", line 177, in <module>
    main()
  File "C:\Users\testör\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1/tools/sizes.py", line 158, in main
    sizes = get_segment_sizes(args.elf, args.path, args.mmu)
  File "C:\Users\testör\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1/tools/sizes.py", line 77, in get_segment_sizes
    lines = proc.stdout.readlines()
  File "D:\obj\Windows-Release\37win32_Release\msi_python\zip_win32\codecs.py", line 322, in decode
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xf6 in position 13: invalid start byte
```
There are two files to replace: 
* https://raw.githubusercontent.com/esp8266/Arduino/94a8c4e6befe55d23bffd53d8a077a2e7dc57d79/tools/mkbuildoptglobals.py
* https://raw.githubusercontent.com/esp8266/Arduino/94a8c4e6befe55d23bffd53d8a077a2e7dc57d79/tools/sizes.py

in `.\3.1.1\tools\`

@RastoH and @TobbeG could you give these a try?

RastoH, 2023-02-21:
Arduino IDE 2.0.3 & 1.8.19 

works :)

closed 2023-02-22 by mcspr