esp8266/Arduino #8861: Processing tons of data with pointers causes code to constantly crash.

XanCraft21, 2023-02-17:
### Basic Infos

- [X ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X ] I have tested that the issue is present in current master branch (aka latest git).
- [X ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it. (I will do this part soon.)
- [X ] I have filled out all fields below.

#### Platform

- Hardware:				ESP-12E
- Core Version:			3.1.1
- Development Env:		Arduino IDE
- Operating System:		Windows 10

### Settings in IDE

- Module:  			NodeMCU 1.0 (ESP-12E Module)
- Flash Mode:			DOUT  (I don't remember the default setting for NodeMCU since the option is not present for this board selection)

- Flash Size: 		4MB (FS:none OTA:~1019KB)
- lwip Variant:   v2 Lower Memory
- Reset Method:		dtr nodemcu (Not sure of setting for NodeMCU, not listed when this is selected)

- Flash Frequency:	40Mhz
- CPU Frequency:		160MHz
- Upload Using:		SERIAL
- Upload Speed:   256000 (Highest i can go before it gets unstable)

### Problem Description

I posted this in a discussion since the issue template is over complicated, but i got no answer, so i tried my hardest to make an issue with the given template. When i process tons of data using data pointers, the code crashes every time. I tried different settings, i made sure i have enough ram, i even tried force optimizing the code in the platform .txt file. Nothing makes it work. The code i am working with is basic 3D rendering, i even used a very lightweight header only code i found on codeberg by DrummyFish called Small3DLib. It works fine on my other controllers, and i only think it crashes when reading from a large array from a pointer, and maybe when writing to some other arrays using pointers. I'm surprised no-one else has this same problem, and this bug has been there for many past versions of the Arduino core. I have not decoded the stack dump yet but i will when i get to it. So far the errors i can see are Exception 3, reset reason 2. If i enable the debug output port to CORE mode i get a Load store error. If anyone can figure out the problem and fix it i will gladly appreciate it. Thank you for any help i can get.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

The sketch is too complicated and large to make a minimal version of it. if you need me to i will create a gist and link it here.

### Debug Messages

If this is the debug message using the debug setting in Arduino i have one, but if not i need to decode my stack dump.

```
SDK:2.2.2-dev(38a443e)/Core:3.1.1=30101000/lwIP:STABLE-2_1_3_RELEASE/glue:1.2-65-g06164fb/BearSSL:b024386
Fatal exception 3(LoadStoreErrorCause):
epc1=0x40201764, epc2=0x00000000, epc3=0x00000000, excvaddr=0x4024e0e8, depc=0x00000000
```

mcspr, 2023-02-17:
Does it work with `Non-32-bit access` menu option enabled?

The idea of template is to have as much info as possible. Exception address above is meaningless without some context like code and / or decoded stack trace, as it points to some of your functions (which we obviously don't known anything about)

XanCraft21, 2023-02-17:
No, it does not work with that setting enabled.

What other information do you think i should add? I will add a code soon, i hope it’s ok i use gist since it’s too complex for me to condense.

I will decode the stack trace when i get a chance, i was low on time when i wrote this.

d-a-v, 2023-02-17:
Software-only code can be tested and debugged on computers (linux, mac, windows/WSL2) [using emulation on host](https://github.com/esp8266/Arduino/discussions/8862#discussioncomment-5009722).
Tools like valgrind can help tracking issues.

XanCraft21, 2023-02-18:
Ok, it might take a while before i can get the decoded message done.

I still don’t understand what MCVE sketch is, is that my sketch i wrote my program on? Since my sketch is too complex to make smaller would you like me to send a link to a gist for each file, or should i attach to download?

mcspr, 2023-02-18:
Whatever you think is sufficient.

The goal here is to have some small example of the code that fails; it can be reproduced by anyone reading this issue, not just by you alone. If it is something related to core, we can fix it. If not, up to you I am afraid
Decoded exception usually points at the failing line, or at least somewhere in the vicinity