esp8266/Arduino #8867: Erreur de compilation pour la carte ESP32 Dev Module

ZakariaSell, 2023-02-24:
//Define libraries
#include <DHT.h>
#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>

//define Pins
RF24 radio(4, 5); // CE, CSN
const byte address[6] = "00001";// adresse du canal de communication

void setup(){
Serial.begin(9600); // initialisation du moniteur série
radio.begin();// initialisation du module NRF24L01
radio.openReadingPipe(0, address); // configuration du canal de communication du module NRF24L01
radio.setPALevel(RF24_PA_MIN);); // puissance max 0 dB
radio.startListening();// le module est en mode récepteur
}

void loop() {
String req;
if (radio.available()) { // si des données sont présentes
radio.read(&req, sizeof(req));
Serial.println(req);
}

}

In file included from C:\Users\zsell\OneDrive\Documents\Arduino\libraries\RF24/RF24.h:18,
from C:\Users\zsell\OneDrive\Desktop\code récepteur\Code__Recepteur\Code__Recepteur.ino:5:
C:\Users\zsell\OneDrive\Documents\Arduino\libraries\RF24/RF24_config.h:137:11: fatal error: avr/pgmspace.h: No such file or directory
#include <avr/pgmspace.h>
^~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Erreur de compilation pour la carte ESP32 Dev Module

earlephilhower, 2023-02-24:
Please stop spamming this repo.  You have ignored the template and are posting issues about a different Arduino core, this is the ESP8266 not ESP32 one.  Closing.

closed 2023-02-24 by earlephilhower

ZakariaSell, 2023-02-24:
Sorry, sir, I don't know use github
But I have a problem and I want to help how to do ?

earlephilhower, 2023-02-24:
Go to https://github.com/espressif/arduino-esp32, read their instructions, and answer all the questions in their issue template.

If you just post random code and ignore the instructions given, you will not get assistance.