esp8266/Arduino #8871: Core 3.1.1 - ESP8266HTTPClient.h getString() empty or truncated - read Timeout

TobbeG, 2023-02-27:
### Basic Infos

- [X ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ X] I have tested that the issue is present in current master branch (aka latest git).
- [X ] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X ] I have filled out all fields below.

#### Platform

- Hardware:				[ESP01|
- Core Version:			[3.1.1]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		       [1MB]
- lwip Variant:                 [v2 Higher Bandwidth]
- Reset Method:		[nodemcu]
- Flash Frequency:	        [40Mhz]
- CPU Frequency:		[160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:              [460800] (serial upload only)

### Problem Description
When I make https.GET not the all data received gets read into payload.

ESP8266Http library returms read Timeout
 - [HTTP-Client][returnError] error(-11): read Timeout

https.setTimeout(10000); does not help

Sending same query in web browser, reply is received within 2 seconds always.

Full XML response is around 3700 bytes, but only part of it is in payload (ie. some 1200 bytes but it varies from 0...3700)
despite that HTTP_CODE is 200 and [HTTP-Client] read chunk len is 3701

**CODE:**
    WiFiClientSecure *client = new WiFiClientSecure;
    int httpCode;
    if (client) {
      client->setInsecure();

          Serial.println(String(F("[ENTS]: Today url:")) + String(url));
          if (https.begin(*client, url)) {
            https.addHeader("Content-Type", "application/xml");
https.setTimeout(10000);
            httpCode = https.GET();
            Serial.println(String(F("[ENTS]: HTTP_CODE: ")) + String(httpCode));
            if (httpCode == HTTP_CODE_OK || httpCode == HTTP_CODE_MOVED_PERMANENTLY) {
              String payload = https.getString();
              Serial.println(String(F("[ENTS]: Length="))+String(payload.length()));
              Serial.println(payload);

**### Debug printed with myown and HTTP_CLIENT DEBUG** 
----------------------------------------------------------------------
[ENTS]: Today url:https://web-api.tp.entsoe.eu/api?securityToken=XXXXXXXXXXXXXX&documentType=A44&in_Domain=10Y1001A1001A46L&out_Domain=10Y1001A1001A46L&periodStart=202302262300&periodEnd=202302272300

[HTTP-Client][begin] url: https://web-api.tp.entsoe.eu/api?securityToken=XXXXXXXXXXXXXX&documentType=A44&in_Domain=10Y1001A1001A46L&out_Domain=10Y1001A1001A46L&periodStart=202302262300&periodEnd=202302272300

[HTTP-Client][begin] host: web-api.tp.entsoe.eu port: 443 url: /api?securityToken=XXXXXXXXXXXXXX&documentType=A44&in_Domain=10Y1001A1001A46L&out_Domain=10Y1001A1001A46L&periodStart=202302262300&periodEnd=202302272300

[HTTP-Client][sendRequest] type: 'GET' redirCount: 0
[HTTP-Client] connected to web-api.tp.entsoe.eu:443
[HTTP-Client] sending request header
-----
GET /api?securityToken=XXXXXXXXXXXXXX&documentType=A44&in_Domain=10Y1001A1001A46L&out_Domain=10Y1001A1001A46L&periodStart=202302262300&periodEnd=202302272300 HTTP/1.1
Host: web-api.tp.entsoe.eu
User-Agent: ESP8266HTTPClient
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
Connection: keep-alive
Content-Type: application/xml
Content-Length: 0

-----
[HTTP-Client][handleHeaderResponse] RX: 'HTTP/1.1 200 OK
'
[HTTP-Client][handleHeaderResponse] RX: 'Date: Mon, 27 Feb 2023 17:17:21 GMT
'
[HTTP-Client][handleHeaderResponse] RX: 'Server: Apache
'
[HTTP-Client][handleHeaderResponse] RX: 'Strict-Transport-Security: max-age=31536000 ; includeSubDomains
'
[HTTP-Client][handleHeaderResponse] RX: 'X-Powered-By: Undertow/1
'
[HTTP-Client][handleHeaderResponse] RX: 'Keep-Alive: timeout=5, max=100
'
[HTTP-Client][handleHeaderResponse] RX: 'Connection: Keep-Alive
'
[HTTP-Client][handleHeaderResponse] RX: 'Transfer-Encoding: chunked
'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Type: text/xml
'
[HTTP-Client][handleHeaderResponse] RX: '
'
[HTTP-Client][handleHeaderResponse] code: 200
[HTTP-Client][handleHeaderResponse] Transfer-Encoding: chunked
[ENTS]: HTTP_CODE: 200
[HTTP-Client] read chunk len: 3701
[HTTP-Client][returnError] error(-11): read Timeout
[HTTP-Client][returnError] tcp stop
[ENTS]: Length=1260
<?xml version="1.0" encoding="UTF-8"?>
<Publication_MarketDocument xmlns="urn:iec62325.351:tc57wg16:451-3:publicationdocument:7:0">
	<mRID>d2145fefc7324f33a62e9ce82a4f3a17</mRID>
	<revisionNumber>1</revisionNumber>
	<type>A44</type>
	<sender_MarketParticipant.mRID codingScheme="A01">10X1001A1001A450</sender_MarketParticipant.mRID>
	<sender_MarketParticipant.marketRole.type>A32</sender_MarketParticipant.marketRole.type>
	<receiver_MarketParticipant.mRID codingScheme="A01">10X1001A1001A450</receiver_MarketParticipant.mRID>
	<receiver_MarketParticipant.marketRole.type>A33</receiver_MarketParticipant.marketRole.type>
	<createdDateTime>2023-02-27T17:17:22Z</createdDateTime>
	<period.timeInterval>
		<start>2023-02-26T23:00Z</start>
		<end>2023-02-27T23:00Z</end>
	</period.timeInterval>
	<TimeSeries>
		<mRID>1</mRID>
		<businessType>A62</businessType>
		<in_Domain.mRID codingScheme="A01">10Y1001A1001A46L</in_Domain.mRID>
		<out_Domain.mRID codingScheme="A01">10Y1001A1001A46L</out_Domain.mRID>
		<currency_Unit.name>EUR</currency_Unit.name>
		<price_Measure_Unit.name>MWH</price_Measure_Unit.name>
		<curveType>A01</curveType>
			<Period>
				<timeInterval>
					<start>2023-02-26T23:00Z</start>
					<end>2023-02-27T23:00Z</end>
				</timeInterval>

d-a-v, 2023-02-27:
> `[HTTP-Client] read chunk len: 3701`
> `[HTTP-Client][returnError] error(-11): read Timeout`

Are there 10secs or only the default 5secs between these two lines ?

TobbeG, 2023-02-27:
It's 10 sec

19:23:02.769 -> [ENTS]: HTTP_CODE: 200
19:23:02.769 -> [HTTP-Client] read chunk len: 3701
19:23:12.735 -> [HTTP-Client][returnError] error(-11): read Timeout
19:23:12.735 -> [HTTP-Client][returnError] tcp stop
19:23:12.781 -> [ENTS]: Length=236

d-a-v, 2023-03-01:
Can you restart with SSL+HTTP_CLIENT debug option enabled ?

TobbeG, 2023-03-02:
**Same ERROR - Extract of full debug print (For full debug print, se end of this comment**

1. Http code=200 = OK
2. Read length=3697 OK
3. Timeout? error -11, why??
4. Then payload length only 1772??

**DEBUG extract**
00:34:37.799 -> [HTTP-Client][handleHeaderResponse] code: 200
00:34:37.799 -> [HTTP-Client][handleHeaderResponse] Transfer-Encoding: chunked
00:34:37.799 -> [ENTS]: HTTP_CODE: 200
00:34:37.799 -> [HTTP-Client] read chunk len: 3697
00:34:47.804 -> [HTTP-Client][returnError] error(-11): read Timeout
00:34:47.804 -> [HTTP-Client][returnError] tcp stop
00:34:47.804 -> [ENTS]: Length=1772

**CODE:**
          if (https.begin(*client, url)) {
            https.addHeader("Content-Type", "application/xml");
https.setTimeout(10000);
            httpCode = https.GET();
            printDEBUGln(String(F("[ENTS]: HTTP_CODE: ")) + String(httpCode));
            if (httpCode == HTTP_CODE_OK || httpCode == HTTP_CODE_MOVED_PERMANENTLY) {
              String payload = https.getString();
              printDEBUGln(String(F("[ENTS]: Length="))+String(payload.length()));

**Full Serial DEBUG print**
00:34:36.621 -> [ENTS]: Today url:https://web-api.tp.entsoe.eu/api?securityToken=XXXXXXXXXXXXXXXXXXXX&documentType=A44&in_Domain=10Y1001A1001A46L&out_Domain=10Y1001A1001A46L&periodStart=202303022300&periodEnd=202303032300
00:34:36.621 -> [HTTP-Client][begin] url: https://web-api.tp.entsoe.eu/api?securityToken=XXXXXXXXXXXXXXXXXXXX&documentType=A44&in_Domain=10Y1001A1001A46L&out_Domain=10Y1001A1001A46L&periodStart=202303022300&periodEnd=202303032300
00:34:36.668 -> [HTTP-Client][begin] host: web-api.tp.entsoe.eu port: 443 url: /api?securityToken=XXXXXXXXXXXXXXXXXXXX&documentType=A44&in_Domain=10Y1001A1001A46L&out_Domain=10Y1001A1001A46L&periodStart=202303022300&periodEnd=202303032300
00:34:36.668 -> [HTTP-Client][sendRequest] type: 'GET' redirCount: 0
00:34:36.715 -> BSSL:_connectSSL: start connection
00:34:36.762 -> BSSL:CERT: 30 82 06 35 30 82 05 1d a0 03 02 01 02 02 11 00 b0 5a 57 9a 0e 44 dc e3 94 aa ed cd 15 bd 8e af 30 0d 06 09 2a 86 48 86 f7 0d 01 01 0b 05 00 30 81 8f 31 0b 30 09 06 03 55 04 06 13 02 47 42 31 1b 30 19 06 03 55 04 08 13 12 47 72 65 61 74 65 72 20 4d 61 6e 63 68 65 73 74 65 72 31 10 30 0e 06 03 55 04 07 13 07 53 61 6c 66 6f 72 64 31 18 30 16 06 03 55 04 0a 13 0f 53 65 63 74 69 67 6f 20 4c 69 6d 69 74 65 64 31 37 30 35 06 03 55 04 03 13 2e 53 65 63 74 69 67 6f 20 52 53 41 20 44 6f 6d 61 69 6e 20 56 61 6c 69 64 61 74 69 6f 6e 20 53 65 63 75 72 65 20 53 65 72 76 65 72 20 43 41 30 1e 17 0d 32 32 30 33 32 33 30 30 30 30 30 30 5a 17 0d 32 33 30 34 32 33 32 33 35 39 35 39 5a 30 19 31 17 30 15 06 03 55 04 03 0c 0e 2a 2e 74 70 2e 65 6e 74 73 6f 65 2e 65 75 30 82 01 22 
00:34:36.809 -> BSSL:CERT: 30 0d 06 09 2a 86 48 86 f7 0d 01 01 01 05 00 03 82 01 0f 00 30 82 01 0a 02 82 01 01 00 ae 58 e2 6d 26 30 32 af c8 3a 7e e5 14 8b cf d4 4b 37 df 1d 31 af 95 fb 33 e3 62 ca c0 de 1e 2e f5 2b 11 5a 27 cb f6 b3 db 96 32 9b 6f 7b fc b6 53 13 7d 56 2d 5b 73 9b 1c 53 57 69 0d f0 da 9d 14 d6 fc 6a c0 52 1b 2d a3 5d ea f3 7e b2 24 03 d4 58 35 b3 a2 c8 e8 9e c9 4d 41 79 69 26 01 3a 52 86 8d 66 a8 0b 03 5b ef ad ae 67 10 df 53 98 3e d2 86 7e 69 40 e5 0f dd 25 58 7e 04 40 15 23 58 b1 84 6a 12 a5 ed 68 7c cc 59 e6 de b2 d6 c9 aa 36 67 22 1e e9 0b 5f 7e 6b 9a 8e 68 c6 79 c7 86 b2 5b f8 0c 7a fd 6c f0 6a bf 21 5b af f7 57 04 2a 36 53 89 a4 ad 07 dc b3 f5 47 40 a6 ff e4 b9 1d 13 51 c5 e7 6b a2 21 9c be 90 f6 3e 4f 84 8a 48 c1 6c ba 6a b5 4a 86 16 34 dd 28 1f 0a e8 de b9 6c 
00:34:36.904 -> BSSL:CERT: 8a 74 94 1f 01 28 cb f5 9f 11 40 c6 73 e1 fc b6 8f e0 a1 46 6d 46 aa 6a 6a b3 73 bf 3d 02 03 01 00 01 a3 82 02 ff 30 82 02 fb 30 1f 06 03 55 1d 23 04 18 30 16 80 14 8d 8c 5e c4 54 ad 8a e1 77 e9 9b f9 9b 05 e1 b8 01 8d 61 e1 30 1d 06 03 55 1d 0e 04 16 04 14 f4 d0 b1 4b e2 98 a5 44 13 a9 7b c0 e6 6e 1a 57 3d 93 00 f2 30 0e 06 03 55 1d 0f 01 01 ff 04 04 03 02 05 a0 30 0c 06 03 55 1d 13 01 01 ff 04 02 30 00 30 1d 06 03 55 1d 25 04 16 30 14 06 08 2b 06 01 05 05 07 03 01 06 08 2b 06 01 05 05 07 03 02 30 49 06 03 55 1d 20 04 42 30 40 30 34 06 0b 2b 06 01 04 01 b2 31 01 02 02 07 30 25 30 23 06 08 2b 06 01 05 05 07 02 01 16 17 68 74 74 70 73 3a 2f 2f 73 65 63 74 69 67 6f 2e 63 6f 6d 2f 43 50 53 30 08 06 06 67 81 0c 01 02 01 30 81 84 06 08 2b 06 01 05 05 07 01 01 04 
00:34:36.950 -> BSSL:CERT: 78 30 76 30 4f 06 08 2b 06 01 05 05 07 30 02 86 43 68 74 74 70 3a 2f 2f 63 72 74 2e 73 65 63 74 69 67 6f 2e 63 6f 6d 2f 53 65 63 74 69 67 6f 52 53 41 44 6f 6d 61 69 6e 56 61 6c 69 64 61 74 69 6f 6e 53 65 63 75 72 65 53 65 72 76 65 72 43 41 2e 63 72 74 30 23 06 08 2b 06 01 05 05 07 30 01 86 17 68 74 74 70 3a 2f 2f 6f 63 73 70 2e 73 65 63 74 69 67 6f 2e 63 6f 6d 30 27 06 03 55 1d 11 04 20 30 1e 82 0e 2a 2e 74 70 2e 65 6e 74 73 6f 65 2e 65 75 82 0c 74 70 2e 65 6e 74 73 6f 65 2e 65 75 30 82 01 7f 06 0a 2b 06 01 04 01 d6 79 02 04 02 04 82 01 6f 04 82 01 6b 01 69 00 76 00 ad f7 be fa 7c ff 10 c8 8b 9d 3d 9c 1e 3e 18 6a b4 67 29 5d cf b1 0c 24 ca 85 86 34 eb dc 82 8a 00 00 01 7f b7 3e e6 44 00 00 04 03 00 47 30 45 02 20 18 94 48 0b c7 0a 64 f9 77 13 a1 38 21 80 75 
00:34:37.045 -> BSSL:CERT: 8d 8f b2 f2 f9 cf 0a 44 9f ef 69 9f e9 36 1f 65 59 02 21 00 f5 b4 3e ff 15 e1 e9 21 c1 05 9d 51 98 48 c6 d8 50 38 84 2f df e9 62 15 d8 3c 0f 19 15 c5 68 37 00 77 00 7a 32 8c 54 d8 b7 2d b6 20 ea 38 e0 52 1e e9 84 16 70 32 13 85 4d 3b d2 2b c1 3a 57 a3 52 eb 52 00 00 01 7f b7 3e e6 1a 00 00 04 03 00 48 30 46 02 21 00 f0 93 dd a6 65 59 dd e6 0a 51 3b c9 14 7d b9 5 [... cut ...]
00:34:48.134 -> 						<position>3</position>
00:34:48.134 -> 						<price.amount>22.00</price.amount>
00:34:48.134 -> 					</Point>
00:34:48.134 -> 					<Point>
00:34:48.134 -> 						<position>4</position>
00:34:48.134 -> 						<price.amount>22.52</price.amount>
00:34:48.134 -> 					</Point>
00:34:48.134 -> 					<Point>
00:34:48.134 -> 						<position>5</position>
00:34:48.134 -> 						<price.amount>23.49</price.amount>
00:34:48.134 -> 					</Po[ENTS]: Today payload contained: 5
00:34:48.134 -> [HTTP-Client][end] tcp is closed

d-a-v, 2023-03-10:
I tried to reproduce this bug without success.
#8885 permits to run ssl servers with 4KB chunked answers that are correctly read by the BasicHttpsClient example.
If you have another ssl+chunked webserver that we can try ouselves (or a way to get a temporary token with https://web-api.tp.entsoe.eu), we'll be happy to try and understand what's going on.

TobbeG, 2023-03-11:
Thanks for continuing to work on this issue.

Can it be something in the dataproviders data that is not expected or?

I cannot share my Token but use this guide to obtain your own to continue test.

You can register and obtain API token free of charge, and here is the procedure:

  1.  Go to https://transparency.entsoe.eu/ and click "Login" in upper right corner.
  2.  A new window opens, click on "Register" at the bottom. Fill in your email and select a password, which must be at least 14 character long and must contain at least one special character (other than letter, digit or '@').
  3.  You'll receive an email requesting you to confirm activation. Click the link in the email, and you're logged in.
  4.  So far so good - normal stuff.
Now you need to send them an email to request API token, that information is somewhat hidden in paragraph 2 on this page:<https://transparency.entsoe.eu/content/static_content/Static%20content/web%20api/Guide.html>
Send an email to ***@***.*** with “Restful API access” in the subject line. Indicate the email address you entered during registration in the email body. The ENTSO-E Helpdesk will make their best efforts to respond to your request within 3 working days.
  5.  They will normally respond next working day, I guess 3 days is to be on the safe side.

When you receive answer from them, there will be a button on your "My account settings" page. Click it to generate API token.

Regards
Thorbjörn

otrobanda ab
Mob. +46(0)705182318
SE Knaparydsvägen 15, 43993 Onsala
DE Pfalzburger Strasse 67, 10719 Berlin

MSKRoss, 2023-03-15:
I think I'm wrestling with the same issue; I have a simple Get request that pulls back less than 800 bytes. Under v3.0.2 of the library it works fine. Upgrade to the latest (3.1.1) and it's truncating the result (which initially made it look like a JSON parsing error). Reverted back to 3.0.2 and it works fine again. Very basic code doing the work - even ChatGPT can spit this out ;) 

WiFiClientSecure client;
...
    client.print(String("GET ") + apiName + apiToken + " HTTP/1.1\r\n" +
              "Host: " + serverName + "\r\n" +
              "Connection: close\r\n\r\n");

    // Read (and toss) the response header
    while (client.connected()) {
      String line = client.readStringUntil('\n');
      if (line == "\r") {
        break;
      }
    }

    // Read the JSON response data
    String rawJSON = client.readStringUntil('\n');

Under the 3.0.2 version, that rawJSON variable will hold the complete JSON string. Under 3.1.x the string is truncated or missing entirely.

mcspr, 2023-03-16:
> while (client.connected())

context switch race happening? if tcp client is disconnected, tls engine is treated as disabled. if engine is disabled, we are not connected. available() would tell whether remaining data is available.
(see #8330, same as basic WiFiClient method)

https://github.com/esp8266/Arduino/blob/84a59aa97fc3881231579127c5a0235b3e6cef3f/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp#L264-L267
https://github.com/esp8266/Arduino/blob/84a59aa97fc3881231579127c5a0235b3e6cef3f/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp#L260-L262
https://github.com/esp8266/Arduino/blob/84a59aa97fc3881231579127c5a0235b3e6cef3f/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp#L252-L258

TobbeG, 2023-03-16:
Let me know if there is a problem getting the TOKEN for tests.
Appreciate the continuation of finding why Entso-e data is lost or truncated with getString().
Entso-e is a EU website providing data for electricity spotprices and I do really need to find a solution or workaround.
Thank you!

mcspr, 2023-03-20:
Building w/ `-DDEBUG_ESP_CORE -DDEBUG_ESP_SSL -DDEBUG_ESP_TLS_MEM -DDEBUG_ESP_HTTP_CLIENT` would give us some more info about the connection.

Not sure if above is related, since we check both available() and connected() in HTTP client code.
API and a simple sketch work just fine printing to Serial, either as Stream or saving a temporary String.
(also note that you probably want to use `const String&` as object type, string data is duplicated to your object so it is twice the mem used)

TobbeG, 2023-03-20:
> Building w/ `-DDEBUG_ESP_CORE -DDEBUG_ESP_SSL -DDEBUG_ESP_TLS_MEM -DDEBUG_ESP_HTTP_CLIENT` would give us some more info about the connection.
>
> Not sure if above is related, since we check both [...]

Is the above related to my original issue?

Did you get and test with your own TOKEN and the Entsoe server?

https://web-api.tp.entsoe.eu/api?securityToken=XXXXXXXXXXXXXXXXXXXX&documentType=A44&in_Domain=10Y1001A1001A46L&out_Domain=10Y1001A1001A46L&periodStart=202303022300&periodEnd=202303032300

mcspr, 2023-03-20:
Yes to both. Debug opts above should also dump ClientContext transfers info.

TobbeG, 2023-03-20:
Can you explain how to use const String& instead of String - do you mean "String payload" should be "const String payload"?

Debug data is enourmous and I get Exception(29)  - tried to capture some here:

00:11:11.159 -> [ENTS]: Today url:https://web-api.tp.entsoe.eu/api?securityToken=ab4797bd-8010-4078-9325-4f3af043410f&documentType=A44&in_Domain=10Y1001A1001A46L&out_Domain=10Y1001A1001A46L&periodStart=202303202300&periodEnd=202303212300
00:11:11.159 -> [HTTP-Client][begin] url: https://web-api.tp.entsoe.eu/api?securityToken=ab4797bd-8010-4078-9325-4f3af043410f&documentType=A44&in_Domain=10Y1001A1001A46L&out_Domain=10Y1001A1001A46L&periodStart=202303202300&periodEnd=202303212300
00:11:11.205 -> [HTTP-Client][begin] host: web-api.tp.entsoe.eu port: 443 url: /api?securityToken=ab4797bd-8010-4078-9325-4f3af043410f&documentType=A44&in_Domain=10Y1001A1001A46L&out_Domain=10Y1001A1001A46L&periodStart=202303202300&periodEnd=202303212300
00:11:11.205 -> [HTTP-Client][sendRequest] type: 'GET' redirCount: 0
00:11:11.205 -> [hostByName] request IP for: web-api.tp.entsoe.eu
00:11:11.205 -> [hostByName] Host: web-api.tp.entsoe.eu IP: 62.209.222.31
00:11:11.205 -> :ref 1
00:11:11.252 -> BSSL:_connectSSL: start connection
00:11:11.252 -> 
00:11:11.252 -> _iobuf_in:       0x3fff67f4
00:11:11.252 -> _iobuf_out:      0x3fffa94c
00:11:11.252 -> _iobuf_in_size:  16709
00:11:11.252 -> _iobuf_out_size: 597
00:11:11.298 -> :wr 225 0
00:11:11.298 -> :wrc 225 225 0
00:11:11.298 -> :ack 225
00:11:11.345 -> :rn 1460
00:11:11.345 -> :rch 1460, 1460
00:11:11.345 -> :rd 5, 2920, 0
00:11:11.345 -> :rdi 1460, 5
00:11:11.345 -> :rd 98, 2920, 5
00:11:11.345 -> :rdi 1455, 98
00:11:11.345 -> :rd 5, 2920, 103
00:11:11.345 -> :rdi 1357, 5
00:11:11.345 -> :rd 512, 2920, 108
00:11:11.345 -> :rdi 1352, 512
00:11:11.345 -> BSSL:CERT: 30 82 06 35 30 82 05 1d a0 03 02 01 02 02 11 00 b0 5a 57 9a 0e 44 dc e3 94 aa ed cd 15 bd 8e af 30 0d 06 09 2a 86 48 86 f7 0d 01 01 0b 05 00 30 81 8f 31 0b 30 09 06 03 55 04 06 13 02 47 42 31 1b 30 19 06 03 55 04 08 13 12 47 72 65 61 74 65 72 20 4d 61 6e 63 68 65 73 74 65 72 31 10 30 0e 06 03 55 04 07 13 07 53 61 6c 66 6f 72 64 31 18 30 16 06 03 55 04 0a 13 0f 53 65 63 74 69 67 6f 20 4c 69 6d 69 74 65 64 31 37 30 35 06 03 55 04 03 13 2e 53 65 63 74 69 67 6f 20 52 53 41 20 44 6f 6d 61 69 6e 20 56 61 6c 69 64 61 74 69 6f 6e 20 53 65 63 75 72 65 20 53 65 72 76 65 72 20 43 41 30 1e 17 0d 32 32 30 33 32 33 30 30 30 30 30 30 5a 17 0d 32 33 30 34 32 33 32 33 35 39 35 39 5a 30 19 31 17 30 15 06 03 55 04 03 0c 0e 2a 2e 74 70 2e 65 6e 74 73 6f 65 2e 65 75 30 82 01 22 
00:11:11.392 -> :rd 5, 2920, 620
00:11:11.392 -> :rdi 840, 5
00:11:11.392 -> :rd 512, 2920, 625
00:11:11.392 -> :rdi 835, 512
00:11:11.437 -> BSSL:CERT: 30 0d 06 09 2a 86 48 86 f7 0d 01 01 01 05 00 03 82 01 0f 00 30 82 01 0a 02 82 01 01 00 ae 58 e2 6d 26 30 32 af c8 3a 7e e5 14 8b cf d4 4b 37 df 1d 31 af 95 fb 33 e3 62 ca c0 de 1e 2e f5 2b 11 5a 27 cb f6 b3 db 96 32 9b 6f 7b fc b6 53 13 7d 56 2d 5b 73 9b 1c 53 57 69 0d f0 da 9d 14 d6 fc 6a c0 52 1b 2d a3 5d ea f3 7e b2 24 03 d4 58 35 b3 a2 c8 e8 9e c9 4d 41 79 69 26 01 3a 52 86 8d 66 a8 0b 03 5b ef ad ae 67 10 df 53 98 3e d2 86 7e 69 40 e5 0f dd 25 58 7e 04 40 15 23 58 b1 84 6a 12 a5 ed 68 7c cc 59 e6 de b2 d6 c9 aa 36 67 22 1e e9 0b 5f 7e 6b 9a 8e 68 c6 79 c7 86 b2 5b f8 0c 7a fd 6c f0 6a bf 21 5b af f7 57 04 2a 36 53 89 a4 ad 07 dc b3 f5 47 40 a6 ff e4 b9 1d 13 51 c5 e7 6b a2 21 9c be 90 f6 3e 4f 84 8a 48 c1 6c ba 6a b5 4a 86 16 34 dd 28 1f 0a e8 de b9 6c 
00:11:11.483 -> BSSL:CERT: 8a 74 94 1f 01 28 cb f5 9f 11 40 c6 73 e1 fc b6 8f e0 a1 46 6d 46 aa 6a 6a b3 73 bf 3d 02 03 01 00 01 a3 82 02 ff 30 82 02 fb 30 1f 06 03 55 1d 23 04 18 30 16 80 14 8d 8c 5e c4 54 ad 8a e1 77 e9 9b f9 9b 05 e1 b8 01 8d 61 e1 30 1d 06 03 55 1d 0e 04 16 04 14 f4 d0 b1 4b e2 98 a5 44 13 a9 7b c0 e6 6e 1a 57 3d 93 00 f2 30 0e 06 03 55 1d 0f 01 01 ff 04 04 03 02 05 a0 30 0c 06 03 55 1d 13 01 01 ff 04 02 30 00 30 1d 06 03 55 1d 25 04 16 30 14 06 08 2b 06 01 05 05 07 03 01 06 08 2b 06 01 05 05 07 03 02 30 49 06 03 55 1d 20 04 42 30 40 30 34 06 0b 2b 06 01 04 01 b2 31 01 02 02 07 30 25 30 23 06 08 2b 06 01 05 05 07 02 01 16 17 68 74 74 70 73 3a 2f 2f 73 65 63 74 69 67 6f 2e 63 6f 6d 2f 43 50 53 30 08 06 06 67 81 0c 01 02 01 30 81 84 06 08 2b 06 01 05 05 07 01 01 04 
00:11:11.576 -> :rd 5, 2920, 1137
00:11:11.576 -> :rdi 323, 5
00:11:11.576 -> :rd 512, 2920, 1142
00:11:11.576 -> :rdi 318, 318
00:11:11.576 -> :c 318, 1460, 2920
00:11:11.576 -> :rdi 1460, 194
00:11:11.576 -> BSSL:CERT: 78 30 76 30 4f 06 08 2b 06 01 05 05 07 30 02 86 43 68 74 74 70 3a 2f 2f 63 72 74 2e 73 65 63 74 69 67 6f 2e 63 6f 6d 2f 53 65 63 74 69 67 6f 52 53 41 44 6f 6d 61 69 6e 56 61 6c 69 64 61 74 69 6f 6e 53 65 63 75 72 65 53 65 72 76 65 72 43 41 2e 63 72 74 30 23 06 08 2b 06 01 05 05 07 30 01 86 17 68 74 74 70 3a 2f 2f 6f 63 73 70 2e 73 65 63 74 69 67 6f 2e 63 6f 6d 30 27 06 03 55 1d 11 04 20 30 1e 82 0e 2a 2e 74 70 2e 65 6e 74 73 6f 65 2e 65 75 82 0c 74 70 2e 65 6e 74 73 6f 65 2e 65 75 30 82 01 7f 06 0a 2b 06 01 04 01 d6 79 02 04 02 04 82 01 6f 04 82 01 6b 01 69 00 76 00 ad f7 be fa 7c ff 10 c8 8b 9d 3d 9c 1e 3e 18 6a b4 67 29 5d cf b1 0c 24 ca 85 86 34 eb dc 82 8a 00 00 01 7f b7 3e e6 44 00 00 04 03 00 47 30 45 02 20 18 94 48:rch 1460, 697
00:11:11.623 ->  0b c7 0a 64 f9 77 13 a1 38 21 80 75 
[... cut ...]
00:11:13.068 -> 3fffee10:  696d686f 34326f67 3fff1900 40100f0e  
00:11:13.068 -> 3fffee20:  00000000 3ffe860c 3fff19f8 40100f0e  
00:11:13.115 -> 3fffee30:  40261fe9 3ffe860c 0000001c 40101168  
00:11:13.115 -> 3fffee40:  ffffffda 3ffef4fc 3ffed890 3ffefb40  
00:11:13.115 -> 3fffee50:  3ffeeae0 0000003a 00000000 4026698b  
00:11:13.115 -> 3fffee60:  00000000 3fff1a94 ffffffda 00000000  
00:11:13.115 -> 3fffee70:  00000000 3ffefb40 00000010 40101508  
00:11:13.115 -> 3fffee80:  0000001c 00000000 00000000 402417b0  
00:11:13.115 -> 3fffee90:  00000000

mcspr, 2023-03-20:
> Can you explain how to use const String& instead of String - do you mean "String payload" should be "const String payload"?

`String payload = https.getString();` line. I mean the `&` (ampersand). `http.getString()` return type is const reference string, which you copy into payload object.

> Debug data is enourmous and I get Exception(29) - tried to capture some here:

Disable extra stuff like MDNS? :u... logs, and its responder callbacks seem to happen simultaneously with the request.

TobbeG, 2023-03-21:
> `String payload = https.getString();` line. I mean the `&` (ampersand). `http.getString()` return type is const reference string, which you copy into payload object.
>
>
> Disable extra stuff like MDNS? [...]

So it should be... or?
**const String payload = http.getString();**

mcspr, 2023-03-21:
```cpp
const String& payload = http.getString();
```

TobbeG, 2023-03-21:
**Code now: with const String& payload and also prints ESP.getFreeHeap()**
          if (https.begin(*client, url)) {
            https.addHeader("Content-Type", "application/xml");
            https.setTimeout(10000);
            httpCode = https.GET();
            printDEBUGln(String(F("[ENTS]: HTTP_CODE: ")) + String(httpCode));
            if (httpCode == HTTP_CODE_OK || httpCode == HTTP_CODE_MOVED_PERMANENTLY) {
              Serial.println(String(F("0 - Free HEAP:")) + String(ESP.getFreeHeap()) ); 
              const String& payload = https.getString();
              printDEBUGln(String(F("[ENTS]: Length="))+String(payload.length()));
              printDEBUGln(payload);

**This is repeats as an infinite loop until device crashes after timeout 10s**
17:25:51.310 -> [String] Reallocating large String(1260 -> 1772 bytes) '<?xml vers ... erval>

**Debug print**
17:25:45.482 -> [SYS ]: BOOT>>
17:25:45.482 -> [INFO]: DeviceType:    Ohmigo.mini
17:25:45.482 -> [INFO]: DeviceID:      1129931
17:25:45.482 -> [INFO]: Version:       1.1.1 BUILD: Mar 21 2023 17:22:06
17:25:45.482 -> [INFO]: Flash:         1048576
17:25:45.482 -> [INFO]: FlashReal:     1048576
17:25:45.482 -> [INFO]: FlashSpeed MHz 40
17:25:45.529 -> 
17:25:45.529 -> lfs_file_close: fd=0x3fff2910
17:25:45.529 -> [INFO]: DeviceName:    mini
17:25:45.529 -> lfs_file_close: fd=0x3fff2910
17:25:45.529 -> lfs_file_close: fd=0x3fff28c0
17:25:45.529 -> [String] Reallocating large String(143 -> 144 bytes) 'ENTSOE_URL ... =<START>&p'
17:25:45.529 -> lfs_file_close: fd=0x3fff28c0
17:25:45.576 -> lfs_file_close: fd=0x3fff28c0
17:25:45.576 -> lfs_file_close: fd=0x3fff28c0
17:25:45.576 -> [WIFI]: Try 1 connecting to WiFi: ohmigo24wifi evt: 2
17:25:46.470 -> ...wifi evt: 0
17:25:49.567 -> wifi evt: 3
17:25:49.614 -> 	OK with RSSI:-36
17:25:49.614 -> :urn 48
17:25:49.614 -> :urd 48, 48, 0
17:25:49.661 -> lfs_file_close: fd=0x3fff28a0
17:25:49.661 -> [NTP ]: NTP = 1679419550
17:25:49.661 -> lfs_file_close: fd=0x3fff28a0
17:25:49.661 -> [NTP ]: Sync interval set to: 3600 seconds
17:25:49.661 -> lfs_file_close: fd=0x3fff28a0
17:25:49.661 -> [NTP ]: 2023-03-21 17:25:50
17:25:49.661 -> [FILE]: Total:65536
17:25:49.661 -> [FILE]: Used: 12288
17:25:49.661 -> [FILE]: 1129931.cfg - 1024 bytes
17:25:49.661 -> lfs_file_close: fd=0x3fff2968
17:25:49.661 -> [RUN ]: Runsecond is: 14
17:25:49.661 -> [SYS ]: IP:192.168.7.85 - Ready >>
17:25:49.661 -> lfs_file_close: fd=0x3fff3778
17:25:49.709 -> lfs_file_close: fd=0x3fff37a8
17:25:49.709 -> [CHK ]: License valid until: 2037-12-31	2023-03-21
17:25:49.709 -> [ENTS]: File today=0	tomorrow=0
17:25:49.709 -> 
17:25:49.709 -> StackThunk malloc(6200)
17:25:49.709 -> StackThunk stack_thunk_ptr: 0x3fff392c
17:25:49.709 -> lfs_file_close: fd=0x3fff5180
17:25:49.709 -> [String] Reallocating large String(143 -> 144 bytes) 'ENTSOE_URL ... =<START>&p'
17:25:49.755 -> lfs_file_close: fd=0x3fff5180
17:25:49.755 -> [String] Reallocating large String(143 -> 144 bytes) 'ENTSOE_URL ... =<START>&p'
17:25:49.755 -> lfs_file_close: fd=0x3fff5230
17:25:49.755 -> [String] Reallocating large String(147 -> 176 bytes) 'https://we ... End=<STOP>'
17:25:49.755 -> lfs_file_close: fd=0x3fff5250
17:25:49.755 -> [String] Reallocating large String(176 -> 192 bytes) 'https://we ... End=<STOP>'
17:25:49.803 -> lfs_file_close: fd=0x3fff5340
17:25:49.803 -> lfs_file_close: fd=0x3fff5340
17:25:49.803 -> [ENTS]: Today url:https://web-api.tp.entsoe.eu/api?securityToken=XXXXXXXXXXXXTOKENXXXXXXXXXXXX&documentType=A44&in_Domain=10Y1001A1001A46L&out_Domain=10Y1001A1001A46L&periodStart=202303202300&periodEnd=202303212300
17:25:49.803 -> [HTTP-Client][begin] url: https://web-api.tp.entsoe.eu/api?securityToken=XXXXXXXXXXXXTOKENXXXXXXXXXXXX&documentType=A44&in_Domain=10Y1001A1001A46L&out_Domain=10Y1001A1001A46L&periodStart=202303202300&periodEnd=202303212300
17:25:49.849 -> [HTTP-Client][begin] host: web-api.tp.entsoe.eu port: 443 url: /api?securityToken=XXXXXXXXXXXXTOKENXXXXXXXXXXXX&documentType=A44&in_Domain=10Y1001A1001A46L&out_Domain=10Y1001A1001A46L&periodStart=202303202300&periodEnd=202303212300
[... 297 lines cut ...]
17:26:01.725 -> BSSL:CERT: 30 82 06 35 30 82 05 1d a0 03 02 01 02 02 11 00 b0 5a 57 9a 0e 44 dc e3 94 aa ed cd 15 bd 8e af 30 0d 06 09 2a 86 48 86 f7 0d 01 01 0b 05 00 30 81 8f 31 0b 30 09 06 03 55 04 06 13 02 47 42 31 1b 30 19 06 03 55 04 08 13 12 47 72 65 61 74 65 72 20 4d 61 6e 63 68 65 73 74 65 72 31 10 30 0e 06 03 55 04 07 13 07 53 61 6c 66 6f 72 64 31 18 30 16 06 03 55 04 0a 13 0f 53 65 63 74 69 67 6f 20 4c 69 6d 69 74 65 64 31 37 30 35 06 03 55 04 03 13 2e 53 65 63 74 69 67 6f 20 52 53 41 20 44 6f 6d 61 69 6e 20 56 61 6c 69 64 61 74 69 6f 6e 20 53 65 63 75 72 65 20 53 65 72 76 65 72 20 43 41 30 1e 17 0d 32 32 30 33 32 33 30 30 30 30 30 30 5a 17 0d 32 33 30 34 32 33 32 33 35 39 35 39 5a 30 19 31 17 30 15 06 03 55 04 03 0c 0e 2a 2e 74 70 2e 65 6e 74 73 6f 65 2e 65 75 30 82 01 22 
17:26:01.819 -> :rd 5, 3617, 620
17:26:01.819 -> :rdi 840, 5
17:26:01.819 -> 
17:26:01.819 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
17:26:01.819 -> 
17:26:01.819 -> Exception (29):
17:26:01.819 -> epc1=0x4000df64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
17:26:01.819 -> 
17:26:01.819 -> >>>stack>>>
17:26:01.819 -> 
17:26:01.819 -> ctx: sys
17:26:01.819 -> sp: 3fffec10 end: 3fffffb0 offset: 0190
17:26:01.819 -> 3fffeda0:  00000139 ffffffff ffffffff 00000000  
[... cut ...]
17:26:01.866 -> 3fffede0:  40267a45 3fffee90 3ffefb40 0000001a  
17:26:01.866 -> 3fffedf0:  402654a8 3fffee90 3ffef9e8 3ffef374  
17:26:01.866 -> 3fffee00:  3ffec5b4 3fffee90 3fffee90 3fffee50  
17:26:01.866 -> 3fffee10:  696d686f 34326f67 3fff1900 40100f0e  
17:26:01.866 -> 3fffee20:  00000002 3ffe860c 3fff19f8 40100f0e  
17:26:01.866 -> 3fffee30:  40261801 3ffe860c 0000001c 40101168  
17:26:01.866 -> 3fffee40:  ffffffe1 3ffef4fc 3ffec5c4 3ffefb40  
17:26:01.866 -> 3fffee50:  3ffeea90 00000041 00000000 402661a3

TobbeG, 2023-03-25:
After minimizing memory use, the ESP8266 still does not get the data with getString() but it does not crash
It just hangs waiting for timeout with "Reallocating large String..

16:11:43.323 -> [SYS ]: BOOT>>
16:11:43.323 -> [INFO]: DeviceType:    Ohmigo.mini
16:11:43.323 -> [INFO]: DeviceID:      1129931
16:11:43.323 -> [INFO]: Version:       1.1.1 BUILD: Mar 25 2023 16:10:56
16:11:43.323 -> [INFO]: Flash:         1048576
16:11:43.323 -> [INFO]: FlashReal:     1048576
16:11:43.323 -> [INFO]: FlashSpeed MHz 40
16:11:43.323 -> 
16:11:43.323 -> lfs_file_close: fd=0x3fff2910
16:11:43.323 -> [INFO]: DeviceName:    mini
16:11:43.371 -> lfs_file_close: fd=0x3fff2910
16:11:43.371 -> lfs_file_close: fd=0x3fff28c0
16:11:43.371 -> [String] Reallocating large String(143 -> 144 bytes) 'ENTSOE_URL ... =<START>&p'
16:11:43.371 -> lfs_file_close: fd=0x3fff28c0
16:11:43.417 -> lfs_file_close: fd=0x3fff28c0
16:11:43.417 -> lfs_file_close: fd=0x3fff28c0
16:11:43.417 -> [WIFI]: Try 1 connecting to WiFi: ohmigo24wifi evt: 2
16:11:44.314 -> ...wifi evt: 0
16:11:47.898 -> wifi evt: 3
16:11:47.946 -> 	OK with RSSI:-32
16:11:47.994 -> :urn 48
16:11:47.994 -> :urd 48, 48, 0
16:11:47.994 -> lfs_file_close: fd=0x3fff28a0
16:11:47.994 -> [NTP ]: NTP = 1679760708
16:11:47.994 -> lfs_file_close: fd=0x3fff28a0
16:11:47.994 -> [NTP ]: Sync interval set to: 3600 seconds
16:11:47.994 -> lfs_file_close: fd=0x3fff28a0
16:11:47.994 -> [NTP ]: 2023-03-25 16:11:48
16:11:47.994 -> [FILE]: Total:65536
16:11:47.994 -> [FILE]: Used: 12288
16:11:47.994 -> [FILE]: 1129931.cfg - 1024 bytes
16:11:48.042 -> lfs_file_close: fd=0x3fff2968
16:11:48.042 -> [RUN ]: Runsecond is: 14
16:11:48.042 -> [SYS ]: IP:192.168.7.85 - Ready >>
16:11:48.042 -> lfs_file_close: fd=0x3fff3778
16:11:48.042 -> lfs_file_close: fd=0x3fff37a8
16:11:48.042 -> [CHK ]: License valid until: 2037-12-31	2023-03-25
16:11:48.042 -> 
16:11:48.042 -> StackThunk malloc(6200)
16:11:48.042 -> StackThunk stack_thunk_ptr: 0x3fff392c
16:11:48.042 -> [ENTS]: Today url:https://web-api.tp.entsoe.eu/api?securityToken=XXXXXXXXXXXXXXTOKENXXXXXXXXXXX&documentType=A44&in_Domain=10Y1001A1001A46L&out_Domain=10Y1001A1001A46L&periodStart=202303242300&periodEnd=202303252300
16:11:48.089 -> [HTTP-Client][begin] url: https://web-api.tp.entsoe.eu/api?securityToken=XXXXXXXXXXXXXXTOKENXXXXXXXXXXX&documentType=A44&in_Domain=10Y1001A1001A46L&out_Domain=10Y1001A1001A46L&periodStart=202303242300&periodEnd=202303252300
16:11:48.089 -> [HTTP-Client][begin] host: web-api.tp.entsoe.eu port: 443 url: /api?securityToken=XXXXXXXXXXXXTOKENXXXXXXXXXXXX&documentType=A44&in_Domain=10Y1001A1001A46L&out_Domain=10Y1001A1001A46L&periodStart=202303242300&periodEnd=202303252300
16:11:48.137 -> [HTTP-Client][sendRequest] type: 'GET' redirCount: 0
16:11:48.137 -> [hostByName] request IP for: web-api.tp.entsoe.eu
16:11:48.137 -> [hostByName] Host: web-api.tp.entsoe.eu IP: 62.209.222.31
16:11:48.137 -> :ref 1
16:11:48.184 -> BSSL:_connectSSL: start connection
16:11:48.184 -> 
16:11:48.184 -> _iobuf_in:       0x3fff630c
16:11:48.184 -> _iobuf_out:      0x3fffa464
16:11:48.184 -> _iobuf_in_size:  16709
16:11:48.184 -> _iobuf_out_size: 597
16:11:48.231 -> :wr 225 0
16:11:48.231 -> :wrc 225 225 0
16:11:48.231 -> :ack 225
16:11:48.231 -> :rn 1460
16:11:48.231 -> :rd 5, 1460, 0
16:11:48.231 -> :rdi 1460, 5
16:11:48.231 -> :rch 1460, 1460
16:11:48.278 -> :rch 2920, 697
16:11:48.278 -> :rd 98, 3617, 5
16:11:48.278 -> :rdi 1455, 98
16:11:48.278 -> :rd 5, 3617, 103
16:11:48.278 -> :rdi 1357, 5
16:11:48.278 -> :rd 512, 3617, 108
16:11:48.278 -> :rdi 1352, 512
[... 134 lines cut ...]
16:11:49.509 -> [String] Reallocating large String(1772 -> 2284 bytes) '<?xml vers ... 		</Point>'
16:11:49.509 -> [String] Reallocating large String(1772 -> 2284 bytes) '<?xml vers ... 		</Point>'
16:11:49.509 -> [String] Reallocating large String(1772 -> 2284 bytes) '<?xml vers ... 		</Point>'
16:11:49.556 -> [String] Reallocating large String(1772 -> 2284 bytes) '<?xml vers ... 		</Point>'
16:11:49.556 -> [String] Reallocating large String(1772 -> 2284 bytes) '<?xml vers ... 		</Point>'
16:11:49.556 -> [String] Reallocating large String(1772 -> 2284 bytes) '<?xml vers ... 		</Point>'
16:11:49.556 -> [String] Reallocating large String(1772 -> 2284 bytes) '<?xml vers ... 		</Point>'
16:11:49.556 -> [String] Reallocating large String(1772 -> 2284 bytes) '<?xml vers ... 		</Point>'
16:11:49.556 -> [String] Reallocating large String(1772 -> 2284 bytes) '<?xml vers ... 		</Point>'
16:11:49.556 -> [String] Reallocating large String(1772 -> 2284 bytes) '<?xml vers ... 		</Point>'
16:11:49.603 -> [String] Reallocating large String(1772 -> 2284 bytes) '<?xml vers ... 		</Point>'
16:11:49.603 -> [String] Reallocating large String(1772 -> 2284 bytes) '<?xml vers ... 		</Point>'
16:11:49.603 -> [String] Reallocating large String(1772 -> 2284 bytes) '<?xml vers ... 		</Point>'
16:11:49.603 -> [String] Reallocating large String(1772 -> 2284 bytes) '<?xml vers ... 		</Point>'
16:11:49.603 -> [String] Reallocating large String(1772 -> 2284 bytes) '<?xml vers ... 		</Point>'
16:11:49.603 -> [String] Reallocating large String(1772 -> 2284 bytes) '<?xml vers ... 		</Point>'
16:11:49.651 -> [String] Reallocating large String(1772 -> 2284 bytes) '<?xml vers ... 		</Point>'
16:11:49.651 -> [String] Reallocating large String(1772 -> 2284 bytes) '<?xml vers ... 		</Point>'
16:11:49.651 -> [String] Reallocating large String(1772 -> 2284 bytes) '<?xml vers ... 		</Point>'
[... cut ...] int>'
16:11:49.651 -> [String] Reallocating large String(1772 -> 2284 bytes) '<?xml vers ... 		</Point>'
16:11:49.651 -> [String] Reallocating large String(1772 -> 2284 bytes) '<?xml vers ... 		</Point>'
16:11:49.698 -> [String] Reallocating large String(1772 -> 2284 bytes) '<?xml vers ... 		</Point>'
16:11:49.698 -> [String] Reallocating large String(1772 -> 2284 bytes) '<?xml vers ... 		</Point>'
16:11:49.698 -> [String] Reallocating large String(1772 -> 2284 bytes) '<?xml vers ... 		</Point>'

TobbeG, 2023-03-25:
Appreciate if we can get this solved asap, it is causing me lots of trouble.

... or is there another way - maybe a workaround to get extract the price data from Entose?

THANKS for getting this solved soon!

TobbeG, 2023-03-27:
Thanks – I will wait for your green light 😊 once you solved it.

I tried also using stream with 128bytes chunks and that works ok.

Let me know if I can assist you with more info.

TobbeG, 2023-03-27:
**Just an odd thing, maybe it has no relevance or maybe it has?**

When I am using the following code that works using streaming data.
Just before the XML data, I get 3 characters **"e6e"**

CODE:
          if (https.begin(*client, url)) {
            https.addHeader("Content-Type", "application/xml");
            httpCode = https.GET();
            printDEBUGln(String(F("[ENTS]: HTTP_CODE: ")) + String(httpCode));
            if (httpCode == HTTP_CODE_OK || httpCode == HTTP_CODE_MOVED_PERMANENTLY) {
              int len = https.getSize();
              memset(buff, '\0', sizeof buff);
              while(https.connected() && (len>0 || len==-1)){
                size_t size = client->available();
                if( size ){
                  int j = client->readBytes(buff, ((size > sizeof(buff)) ? sizeof(buff) : size));   // read up to 128 byte
                  Serial.write(buff,j);
                  if (len > 0)  len-=j;
                }
                delay(1);
              }

15:00:38.212 -> [ENTS]: Today url:https://web-api.tp.entsoe.eu/api?securityToken=XXXXXXXTOKENXXXXXXX&documentType=A44&in_Domain=10Y1001A1001A46L&out_Domain=10Y1001A1001A46L&periodStart=202303262200&periodEnd=202303272200
15:00:39.013 -> [ENTS]: HTTP_CODE: 200
15:00:39.013 -> e6e
15:00:39.013 -> <?xml version="1.0" encoding="UTF-8"?>
15:00:39.013 -> <Publication_MarketDocument xmlns="urn:iec62325.351:tc57wg16:451-3:publicationdocument:7:0">
15:00:39.013 -> 	<mRID>6613a721515c4715bb49e7e8b82fd35d</mRID>
15:00:39.013 -> 	<revisionNumber>1</revisionNumber>
15:00:39.013 -> 	<type>A44</type>
15:00:39.013 -> 	<sender_MarketParticipant.mRID codingScheme="A01">10X1001A1001A450</sender_MarketParticipant.mRID>

d-a-v, 2023-03-29:
#8874 is merged, #8885 and #8899 should solve your issue.

TobbeG, 2023-03-30:
Hi again, sorry for asking - but how can I get what I need ínto my Arduino IDE 1.8.16 ?
Or is there a new core version or any specific library to download?

mcspr, 2023-03-31:
> Hi again, sorry for asking - but how can I get what I need ínto my Arduino IDE 1.8.16 ? Or is there a new core version or any specific library to download?

https://arduino-esp8266.readthedocs.io/en/latest/installing.html#using-git-version

d-a-v, 2023-04-01:
@TobbeG You can also try the unofficial nightly release for Arduino IDE.

It is located at https://d-a-v.github.io/esp8266/Arduino/index.html under version 0.0.2
(not 0.0.1 because all proposals that are of interest for you are not yet integrated in the master branch)
please take time to read there first.

You can then update timeouts (during handshake or normal operations) also use ssl ram usage optimization in your code as added [in this example](https://github.com/esp8266/Arduino/blob/cc1951aff6f6519f7a4718e3ee26735681e44a94/libraries/ESP8266HTTPClient/examples/BasicHttpsClient/BasicHttpsClient.ino#L63-L71).

I have now no issue reading entsoe data with a sketch similar to the above one.

d-a-v, 2023-04-04:
No but there is an esp8266/Arduino nightly release (v0.0.2) which is v3.1.2 + something for you to try, described in my previous post.

d-a-v, 2023-04-07:
@TobbeG Did you have a chance to try the proposed changes ?

TobbeG, 2023-04-10:
Not yet. I changed my code to use streaming and it works fine for now. 
How long before this fix will be part of a new core version?