esp8266/Arduino #8876: Enabling, but not even actually using ESPNOW on Wemos D1 Mini clone cause random H/W resets

brnrds-git, 2023-03-01:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[be02af0504869e2bf119a013fe8fd6cf5ed8e411]
- Development Env:		[Platformiopio project]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[Wemos D1 mini clone]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

The code listed below will cause random h/w (I think) resets (ets Jan  8 2013,rst cause:4, boot mode:(3,6)). Commenting out everything in setup() below the the empty println() will make the problem away, in the sense that the counter will run for hours without crashes, while when 

```cpp
  WiFi.mode(WIFI_STA);
  if (esp_now_init() != 0)
  {
    Serial.println("Error initializing ESP-NOW");
    return;
  }
  esp_now_set_self_role(ESP_NOW_ROLE_CONTROLLER);
  esp_now_register_send_cb(OnDataSent);
  esp_now_add_peer(broadcastAddress, ESP_NOW_ROLE_SLAVE, 1, NULL, 0);
```
is included it never takes more than a minute or so to crash.

Note that the code in loop(), listed below, doesn't even make use of ESPNow or wifi, it just configures them in setup(). Something in the framework code seems to be triggering the watchdog. Either that or I'm missing additional steps to configure ESPNow properly.

Here is my platformio.ini

```
[env:d1_mini]
platform = espressif8266
platform_packages =
    platformio/framework-arduinoespressif8266 @ https://github.com/esp8266/Arduino.git
board = d1_mini
framework = arduino
lib_deps = 
monitor_speed = 115200
monitor_filters = esp8266_exception_decoder
```

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <espnow.h>

uint8_t broadcastAddress[] = {0x8C, 0xAA, 0xB5, 0xB5, 0x90, 0x60};

typedef struct struct_message
{
  char channel[16];
  char msg[32];
  bool active;
} struct_message;

struct_message myData;

void OnDataSent(uint8_t *mac, uint8_t sendStatus)
{
  Serial.print("Last Packet Send Status: ");
  if (sendStatus == 0)
  {
    Serial.println("Delivery success");
  }
  else
  {
    Serial.println("Delivery fail");
  }
}

void setup()
{
  Serial.begin(115200);
  delay(1000);
  Serial.println();

  // removing the code below will make the crashes go away
  WiFi.mode(WIFI_STA);
  if (esp_now_init() != 0)
  {
    Serial.println("Error initializing ESP-NOW");
    return;
  }
  esp_now_set_self_role(ESP_NOW_ROLE_CONTROLLER);
  esp_now_register_send_cb(OnDataSent);
  esp_now_add_peer(broadcastAddress, ESP_NOW_ROLE_SLAVE, 1, NULL, 0);
  // removing the code above will make the crashes go away
}

int counter = 0;

void loop()
{
  if (counter % 10000 == 0)
  {
    Serial.println(counter);
  }
  counter += 1;
}
```

### Debug Messages

```
8540000
8550000
8560000

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v000429e0
~ld
```

lozziboy, 2023-09-14:
I just run a test of the sketch using:
- board package esp8266 version 3.1.2
- firmware version ESP8266-IDF-AT_V2.2.1.0
- Wemos D1 R2 that mounts a soc ESP8266MOD = the same of Wemos D1 Mini

and it runs really smoth (no crashes at all)

mcspr, 2023-09-14:
**Very** likely to be related to our change between 2.7.4 and 3.0.0 that disables WiFi on boot and removes default that causes STA SSID & pass to be saved to flash. Make sure you are not connecting, right after `mode()` either `disconnect()` or `wifi_station_disconnect()` and clean-up flash contents, can't reproduce either with a fresh device.

Re. above, we don't use [IDF aka RTOS SDK](https://github.com/espressif/ESP8266_RTOS_SDK) underneath, it is still [NONOS](https://github.com/espressif/ESP8266_NONOS_SDK)
ESP8266-IDF-AT_V2.2.1.0 is built by Espressif based on RTOS (which also means there is a different network lib stack used, which may or may not have some differences with wifi and espnow implementations)

closed 2023-09-14 by mcspr

lozziboy, 2023-09-15:
> **Very** likely to be related to our change between 2.7.4 and 3.0.0 that disables WiFi on boot and removes default that causes STA SSID & pass to be saved to flash. Make sure you are not connecting, [...]

Thanks for the feedback!
Still a question about IDF Vs. NONOS: there is the needs to "flash" the NONOS firmware to the device using EspressIf flash_download_tool_3.9.5 before to program it using the IDE, or is it this firmware things are already embedded into the Core when we make a download?

lozziboy, 2023-09-15:
> > **Very** likely to be related to our change between 2.7.4 and 3.0.0 that disables WiFi on boot and removes default that causes STA SSID & pass to be saved to flash. Make sure you are not connecting, right after `mode()` either `disconnect()` or `wifi_station_disconnect()` and clean-up flash contents, can't reproduce either with a fresh device.
> > Re. above, we don't use [IDF aka RTOS SDK](https://github.com/espressif/ESP8266_RTOS_SDK) underneath, it is still [NONOS](https://github.com/espressif/ESP8266_NONOS_SDK) ESP8266-IDF-AT_V2.2.1.0 is built by Espressif based on RTOS (which also means there is a different network lib stack used, which may or may not have some differences with wifi and espnow implementations)
> 
> Thanks for the feedback! Still a question about RTOS_SDK Vs. NONOS_SDK: is there the needs to "flash" the NONOS firmware into the SOC ESP8266 using EspressIf "flash_download_tool" before to load the arduino-esp8266 code using Arduino-IDE, or are these firmware things already embedded into the Core (and so automatically loaded into the SOC) when we perform a Upload?