esp8266/Arduino #8877: AttributeError: module 'time' has no attribute 'time_ns'

denis-stepanov, 2023-03-04:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					ESP-01
- Core Version:			3.1.1
- Development Env:		Arduino IDE
- Operating System:		Ubuntu

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			dio
- Flash Size: 		1MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

I have a problem compiling any sketch with the release 3.1.1. The last working one was 3.0.2.

```
Using board 'generic' from platform in folder: /home/dns/.arduino15/packages/esp8266/hardware/esp8266/3.1.1
Using core 'esp8266' from platform in folder: /home/dns/.arduino15/packages/esp8266/hardware/esp8266/3.1.1
/home/dns/.arduino15/packages/esp8266/tools/python3/3.7.2-post1/python3 -I /home/dns/.arduino15/packages/esp8266/hardware/esp8266/3.1.1/tools/mkbuildoptglobals.py /home/dns/arduino-1.8.19 10819 /tmp/arduino_build_46880 /tmp/arduino_build_46880/core/build.opt /home/dns/arduino-1.8.19/examples/01.Basics/BareMinimum/BareMinimum.ino.globals.h /home/dns/.arduino15/packages/esp8266/hardware/esp8266/3.1.1/cores/esp8266/CommonHFile.h
Traceback (most recent call last):
  File "/home/dns/.arduino15/packages/esp8266/hardware/esp8266/3.1.1/tools/mkbuildoptglobals.py", line 846, in <module>
    sys.exit(main())
  File "/home/dns/.arduino15/packages/esp8266/hardware/esp8266/3.1.1/tools/mkbuildoptglobals.py", line 759, in main
    if time.time_ns() < os.stat(commonhfile_fqfn).st_mtime_ns:
AttributeError: module 'time' has no attribute 'time_ns'
exit status 1
Error compiling for board Generic ESP8266 Module.
```

I saw that a similar error #8803 was merged in 3.1.1, but I still have it.

$ cat /etc/issue
Ubuntu 18.04.6 LTS \n \l
$ python --version
Python 2.7.17
$ python3 --version
Python 3.6.9

Let me know if you need any other info.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

`BareMinimum` from Arduino IDE examples.

d-a-v, 2023-03-04:
It is not "merged" in 3.1.1 but in current `git` version.
It will be included in 3.1.2 when it is released.
In the meantime you can try with the [git version](https://arduino-esp8266.readthedocs.io/en/latest/installing.html#using-git-version)  (arduino or pio), or the unofficial [snapshot](https://d-a-v.github.io/esp8266/Arduino/index.html) release

mcspr, 2023-03-04:
(typing too slow)

We do call system `env python3` on Linux though (#6960)
#8804 fixed the issue with bundled versions, but we can't do anything about users with <3.7 besides asking to update it. And not sure whether it is good idea to have us depend on versions older than we expect.

*btw https://devguide.python.org/versions/ says 3.7 end-of-life is June 2023 :>*

denis-stepanov, 2023-03-05:
Thanks. Yes, the Linux box is somewhat old but not extremely old.

I could not find anywhere in the docs or in the release notes that Python 3.7+ is now a requirement.

closed 2023-03-17 by mcspr

denis-stepanov, 2023-03-18:
OK I found out that on Ubuntu 18.04 Python 3.7 and 3.8 are available in repos. It is OK now after installation of 3.8 and putting it in front of 3.6 in PATH.

PR #8886 (2023-03-17): mkbuildoptglobals.py: assert python version
resolve #8877
also fix spurious warning when getdefaultencoding() is used with versions ≥3.11