esp8266/Arduino #8882: std::mutex is not available

Insomniac80, 2023-03-07:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:				ESP-12F
- Core Version:			PIO Core 6.1.6
- Development Env:		PlatformIO
- Operating System:		Windows

### Settings in IDE

- Module:  			WeMos D1 R2 and mini / 
- Flash Mode:			dio
- Flash Size: 		4MB
- lwip Variant:   v1.4
- Reset Method:		ck
- Flash Frequency:	[40Mhz]
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

std::mutex is not available. Have set up a Win10 VM and fresh installed VS Code and PIO to ensure it is not related to one of my libraries or such. Still persists.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

main.cpp:
```
#include <Arduino.h>

#include <mutex>

using std::mutex;

void setup() {}

void loop() {}
```

platformio.ini:
```
[env:d1_mini]
platform = espressif8266
board = d1_mini
framework = arduino
```

### Output

```
src\main.cpp:5:12: error: 'mutex' has not been declared in 'std'
    5 | using std::mutex;
      |            ^~~~~
*** [.pio\build\d1_mini\src\main.cpp.o] Error 1
```

d-a-v, 2023-03-10:
Our toolchain is not configured to build and provide the thread library (`_GLIBCXX_HAS_GTHREADS` is not defined).
There is currently no multithreading enabled in this architecture because this is useless until proven wrong:
The scheduling model of NONOS-SDK firmware on top of which we are running is single-task even though we know that freertos is running under the hood.

The arduino API `cli()` and `sei()` is implemented and accessible for global / critical locking.

dok-net, 2023-03-12:
According to https://www.arduino.cc/reference/en/, the portable calls are `interrupts()` and `nointerrupts()`, are they not?