esp8266/Arduino #8900: Generic esp8266 does not set i2c pins for ESP01

keremers, 2023-03-30:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-01]
- Core Version:			[3.0.2]
- Development Env:		[Arduino IDE 2.0.4]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		        [1MB]
- lwip Variant:                  [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	        [40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:              [115200] (serial upload only)

### Problem Description

the library does not set i2c pins for ESP-01 correctly. It comes with default 4 SDA/5 SCK which ESP01 is missing. It should set them GP2 and GP0. 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {
.
.
// requires manual initialization
Wire.begin (2,0);
.
.
}

void loop() {

}

```

### Debug Messages

```
Debug messages go here
```

d-a-v, 2023-04-01:
The default (sda,scl)=(4,5) choice is historical and it is not advised to change them now.
Sadly, there is no esp01 in the available board list.
What can be proposed is to add a new board variant (= a new board esp01 or esp01s or both) with proper documentation.
Would you be keen to make such a proposal ?

Changes would [happen there](https://github.com/esp8266/Arduino/blob/a76ef290eaa445f0ad8603121b9a87745a24b072/tools/boards.txt.py#L64) where you can amend the board list displayed in the IDE and allow selecting the matching variant. Its include files would go to [`variants/`](https://github.com/esp8266/Arduino/tree/master/variants)`esp01/`.

enjoyneering, 2023-04-03:
I often use ESP-01 with AHT10 in my projects and they work fine if SDA is on GPIO0 and SCL is on GPIO2.

`Wire.begin(0, 2); //SDA= GPIO0, SCL=GPIO2`