esp8266/Arduino #8904: ESP8266WebServer: The problem with frequent connections

jjsuwa-sys3175, 2023-04-05:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[other (ESP-12 compatible)] (AE-ESP-WROOM-02: see https://akizukidenshi.com/catalog/g/gK-09758/ )]
- Core Version:			[a76ef290eaa445f0ad8603121b9a87745a24b072]
- Development Env:		[Arduino IDE (1.8.19)]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Higher Bandwidth]
- Reset Method:		[nodemcu]
- Flash Frequency:	[80Mhz]
- CPU Frequency:		[160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:   [512000]

### Problem Description

A moderate rate of unexpected HTTP (or socket) disconnections/timeouts is observed when trying to connect to the ESP8266WebServer at a high frequency.
But other HTTP servers belonging to the same WiFi do not see such disconnections/timeouts.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#define WIFI_SSID     "YOUR-WIFI-SSID"
#define WIFI_PASSWORD "YOUR-WIFI-PASSWORD"

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <ESP8266WebServer.h>

ESP8266WebServer webServer(80);

void setup() {
  Serial.begin(74880);
  delay(1000);
  Serial.print(F("\n\nconnecting to WiFi"));
  WiFi.mode(WIFI_STA);
  WiFi.begin(F(WIFI_SSID), F(WIFI_PASSWORD));
  for (; ; Serial.print('.'), delay(333))
    if (WiFi.status() == WL_CONNECTED)
      break;
  Serial.println(F(", done."));
  String addr = WiFi.localIP().toString();
  webServer.onNotFound([]() {
    webServer.send(404);
  });
  webServer.on(F("/"), HTTP_GET, []() {
    StreamString s;
    s.reserve(40);
    s.print(F("http://"));
    WiFi.localIP().printTo(s);
    s.print(F("/index.html"));
    webServer.sendHeader(F("Location"), s);
    webServer.send(302);
  });
  Serial.printf_P(PSTR("webServer: http://%s/\n"), addr.c_str());
  webServer.on(F("/index.html"), HTTP_GET, []() {
    webServer.send(200, F("text/html"), F("<!DOCTYPE html><html>"
                                          "<head><title>WiFi HTTP server</title></head>"
                                          "<body><h1>WiFi HTTP server</h1><p>hello world</p></body>"
                                          "</html>"));
  });
  Serial.printf_P(PSTR("webServer: http://%s/index.html\n"), addr.c_str());
  webServer.on(F("/blob.bin"), HTTP_GET, []() {
    size_t length = (ESP.random() & 4095) + 4096;
    auto buffer = new uint8_t[length];
    ESP.random(buffer, length);
    webServer.send(200, PSTR("application/octet-stream"), buffer, length);
    delete[] buffer;
  });
  Serial.printf_P(PSTR("webServer: http://%s/blob.bin\n"), addr.c_str());
  webServer.begin();
}

void loop() {
  webServer.handleClient();
}
```

### Detailed Procedure

1. Build the above MVCE and upload it to the chip.
2. After the chip reboots and connects to your WiFi, you should get a serial output similar to:
```
connecting to WiFi................, done.
webServer: http://192.168.2.83/
webServer: http://192.168.2.83/index.html
webServer: http://192.168.2.83/blob.bin
```
3. Now, let's use the HTTP load test program "siege" from a linux PC belonging to the same WiFi network to send high-frequency HTTP requests to the chip.  First, write something like this in a text file:
```
http://192.168.2.83/
http://192.168.2.83/index.html
http://192.168.2.83/blob.bin
```
4.  Next, run "siege" in benchmark mode for a some time (a one hour, for example). You'll probably see some alerts in the near future...
```
vboxuser@Bullseye64:/tmp$ siege -qbt 1H -f siege.txt 
[alert] socket: read check timed out(30) sock.c:273: Connection timed out
[alert] socket: select and discovered it's not ready sock.c:384: Connection timed out
[alert] socket: read check timed out(30) sock.c:273: Connection timed out
[alert] socket: select and discovered it's not ready sock.c:384: Connection timed out
...
```
5. Just to be clear, other HTTP servers (Apache, nginx, or even a simple server written in python) don't behave this way. Also, changing the "lwip Variant" setting does the same.

d-a-v, 2023-04-05:
The good news is that the esp did not freeze :)

There is no appropriate api in webserver to call `wificlient::begin(port, tcp-backlog)` but you have tried with other lwIP variants, did this include the "(no features)" variant which has tcp-backlog disabled ?

Otherwise you can try with an increased value of this define :
https://github.com/esp8266/Arduino/blob/a76ef290eaa445f0ad8603121b9a87745a24b072/libraries/ESP8266WiFi/src/WiFiServer.cpp#L39

What I guess happens is that there are too many requests waiting. Starting from the 6th client requesting, lwIP does not anymore answer "please come in and wait for your turn"  so the client sees as if there is no server behind the door. It is normally keeping trying for a little while (`time telnet 192.168.42.42 1234` gives 2mn10s on ubuntu22.04) then fails because lwIP has not opened the door during that time. This should be the origin of the 0.19% of failure.

jjsuwa-sys3175, 2023-04-05:
> > > so let's define MAX_PENDING_CLIENTS_PER_PORT to a number (say 100)
> > 
> > 
> > 100 is like in the "no feature" selection, like infinity, and will fail / freeze / reboot because of OOM on high load.
> 
> Oh, aha, I see, the backlog obviously needs memory :)

It doesn't look like memory is a constraint... Modify the above MVCE `loop()` as follows:
```
void loop() {
  webServer.handleClient();
  static size_t prev_l = 0;
  size_t l = (ESP.getFreeHeap() + 1024) / 2048;
  if (l != prev_l) {
    prev_l = l;
    String s;
    s.reserve(40 * 2);
    size_t i;
    for (i = 0; i < l; ++i)
      s += '#';
    for (; i < 40; ++i)
      s += ' ';
    for (i = 0; i < 40; ++i)
      s += '\b';
    Serial.print(s);
  }
}
```
This allows you to visually track the remaining memory (requires a terminal that can handle backspace). But it seems that there is still enough memory left even during timeouts or other connection failures...

jjsuwa-sys3175, 2023-04-06:
Well, I apologize for not being able to present clear evidence here, but I would like to assert the following:
- It is quite rare (in my experience, about less than 0.1%) but certain that connections are abruptly closed before all responses are received from ESP8266WebServer, possibly caused by early-arriving FINs due to IP packet reordering or dropouts.
- I came up with a simple workaround for the above, which I'm testing right now, but it looks fairly good.
  ```
  --- a/libraries/ESP8266WebServer/src/ESP8266WebServer-impl.h
  +++ b/libraries/ESP8266WebServer/src/ESP8266WebServer-impl.h
  @@ -769,6 +769,7 @@ void ESP8266WebServerTemplate<ServerType>::_finalizeResponse() {
     if (_chunked) {
       sendContent(emptyString);
     }
  +  _currentClient.flush();
   }
   
   template <typename ServerType>
  ```

jjsuwa-sys3175, 2023-04-06:
Finally, only one early-closing server response has ever been occurred with a quickly-made load testing about 1.6 million HTTP transactions by 5 concurrent connections.

I'm still not able to personally convinced, but I'm going to close the issue. Thanks all.

jjsuwa-sys3175, 2023-04-12:
There seems to be some kind of problem with the ESP8266WebServer, so I've reopened this issue.

In general, timeouts in opening connections can be safely resolved by retries, but early closed responses after connection establishment cannot.
With that in mind, let's go back to the first MVCE and prepare an HTTP connection load test using [the famous Python requests](https://github.com/psf/requests).
```
#
if __name__ == "__main__":
    import itertools
    import sys
    import requests
    url = sys.argv[1]
    headers = None  # {"Connection":"close"}
    for n in itertools.count():
        response = requests.get(url, timeout=None, headers=headers)
        print("#%d" % n, end="\r")
```
No special explanation needed, just a normal Python example of many HTTP connections - in fact, we can issue millions of requests on a single connection without any problems (except for timeouts).
However, if we do more than one of the above, all but one will eventually abort due to early closing of the response - parhaps taking no more than an hour:
```
Traceback (most recent call last):
  File "C:\Program Files\python-3.8.10-embed\lib\site-packages\urllib3-1.26.15.zip\urllib3\connectionpool.py", line 703, in urlopen
  File "C:\Program Files\python-3.8.10-embed\lib\site-packages\urllib3-1.26.15.zip\urllib3\connectionpool.py", line 449, in _make_request
  File "<string>", line 3, in raise_from
  File "C:\Program Files\python-3.8.10-embed\lib\site-packages\urllib3-1.26.15.zip\urllib3\connectionpool.py", line 444, in _make_request
  File "http\client.py", line 1344, in getresponse
  File "http\client.py", line 307, in begin
  File "http\client.py", line 276, in _read_status
http.client.RemoteDisconnected: Remote end closed connection without response

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Program Files\python-3.8.10-embed\lib\site-packages\requests-2.28.2.zip\requests\adapters.py", line 489, in send
  File "C:\Program Files\python-3.8.10-embed\lib\site-packages\urllib3-1.26.15.zip\urllib3\connectionpool.py", line 787, in urlopen
  File "C:\Program Files\python-3.8.10-embed\lib\site-packages\urllib3-1.26.15.zip\urllib3\util\retry.py", line 550, in increment
  File "C:\Program Files\python-3.8.10-embed\lib\site-packages\urllib3-1.26.15.zip\urllib3\packages\six.py", line 769, in reraise
  File "C:\Program Files\python-3.8.10-embed\lib\site-packages\urllib3-1.26.15.zip\urllib3\connectionpool.py", line 703, in urlopen
  File "C:\Program Files\python-3.8.10-embed\lib\site-packages\urllib3-1.26.15.zip\urllib3\connectionpool.py", line 449, in _make_request
  File "<string>", line 3, in raise_from
  File "C:\Program Files\python-3.8.10-embed\lib\site-packages\urllib3-1.26.15.zip\urllib3\connectionpool.py", line 444, in _make_request
  File "http\client.py", line 1344, in getresponse
  File "http\client.py", line 307, in begin
  File "http\client.py", line 276, in _read_statusurllib3.exceptions.ProtocolError: ('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "Z:\torture_requests.py", line 9, in <module>
    response = requests.get(url, timeout=None, headers=headers)
  File "C:\Program Files\python-3.8.10-embed\lib\site-packages\requests-2.28.2.zip\requests\api.py", line 73, in get
  File "C:\Program Files\python-3.8.10-embed\lib\site-packages\requests-2.28.2.zip\requests\api.py", line 59, in request
  File "C:\Program Files\python-3.8.10-embed\lib\site-packages\requests-2.28.2.zip\requests\sessions.py", line 587, in request
  File "C:\Program Files\python-3.8.10-embed\lib\site-packages\requests-2.28.2.zip\requests\sessions.py", line 701, in send
  File "C:\Program Files\python-3.8.10-embed\lib\site-packages\requests-2.28.2.zip\requests\adapters.py", line 547, in send
requests.exceptions.ConnectionError: ('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))
```
What this exception trace suggests is that the connection was lost while trying to receive the status line (yes, the first line) of the HTTP response.

FYI, adding the HTTP/1.1 `Connection: close` header doesn't help anything.

closed 2023-05-23 by jjsuwa-sys3175

jjsuwa-sys3175, 2023-06-15:
https://github.com/esp8266/Arduino/pull/8944 seems to fix this issue as well :)