esp8266/Arduino #8910: Crash on WiFi.disconnect(true) if called a few millis after start

olonsoft, 2023-04-18:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[Espressif 8266 (2023.3.0) > WeMos D1 R2 and mini]
- Development Env:		[Platformio]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[platformio default]
- Flash Size: 		[4MB]
- lwip Variant:   [platformio default]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description

When I start my program I do some other non wifi related stuff like loading settings from flash.
Then before I do anything with wifi, I call
```c++
  WiFi.persistent(false); 
  WiFi.disconnect(true);
```
but my program crashes at `WiFi.disconnect(true)`

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {
  Serial.begin(115200);
  delay(500);
 // some flash loading code here.
  WiFi.persistent(false); 
  WiFi.disconnect(true); // <- crash here/
}

void loop() {

}

```

If the first lines of my code are 

```c++
void setup() {
  WiFi.persistent(false); 
  WiFi.disconnect(true);
  Serial.begin(115200);
  delay(500);
 // some flash loading code here.
}
```

everything works normally.

### Debug Messages

```
Exception Cause: 28  [LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads]

0x40241f05: scan_cancel at ??:?
0x40241f01: scan_cancel at ??:?
0x402323c1: sys_timeout_abs at /local/users/gauchard/arduino/arduino_esp8266/esp8266-lwip/tools/sdk/lwip2/builder/lwip2-src/src/core/timeouts.c:189
0x40241e3a: scan_start at ??:?
0x40243d77: chm_end_op at ??:?
0x40243d85: chm_end_op at ??:?
0x40243d98: chm_end_op at ??:?
0x40241d98: scan_start at ??:?
0x40252969: ets_timer_handler_isr at ??:?
0x402529ae: ets_timer_handler_isr at ??:?
0x40000f49: ?? ??:0
0x40000f49: ?? ??:0
0x40000f49: ?? ??:0
0x40000e19: ?? ??:0
0x40104b11: call_user_start_local at ??:?
```

mcspr, 2023-04-22:
> framework-arduinoespressif8266 @ 2.7.4+9

I believe this is the version used by Tasmota. Not the version we have here, 3.1.2 is the latest one. 
https://registry.platformio.org/tools/platformio/framework-arduinoespressif8266

2.x.x defaults to auto-connect, so you are in STA mode and already have some kind of connection attempt happening. Disable autoconnect, disable autoreconnect?
https://github.com/esp8266/Arduino/blob/65579d29081cb8501e4d7f786747bf12e7b37da2/doc/esp8266wifi/station-class.rst#setautoconnect

3.x.x Cores default to disabled WiFi, so you would already have the right state at boot

olonsoft, 2023-04-22:
You are right.
I did not notice the version number.
I  updated to 3.1.2 and now my project runs without a problem.
Thank you !!!

closed 2023-04-22 by olonsoft