esp8266/Arduino #8916: C++20 compatibility: a lot of warnings

MacDada, 2023-04-23:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-8266]
- Core Version:			[v3.0.2]
- Development Env:		[Platformio]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[nodemcuv2]

### Problem Description

I've set a flag to compile with the latest (available) standard: `-std=gnu++20`.

It works, but it produces a lot of warnings, for example:

```
.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_si2c.cpp:576:33: warning: '++' expression of 'volatile'-qualified type is deprecated [-Wvolatile]
  576 |         twi_data = twi_txBuffer[twi_txBufferIndex++];
      |                                 ^~~~~~~~~~~~~~~~~
.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_si2c.cpp:579:9: warning: '--' expression of 'volatile'-qualified type is deprecated [-Wvolatile]
  579 |         bitCount--;
      |         ^~~~~~~~
.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_si2c.cpp:588:18: warning: compound assignment with 'volatile'-qualified left operand is deprecated [-Wvolatile]
  588 |         twi_data <<= 1;
      |         ~~~~~~~~~^~~~~
.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_si2c.
```

They are all connected to the fact that starting with C++20, some of the `volatile` operations are deprecated: https://blog.feabhas.com/2021/05/modern-embedded-c-deprecation-of-volatile/

1. Is there a plan for better support of C++20?
2. Are you open for PRs in this matter?

mcspr, 2023-04-23:
C++20, probably not before also switching to GCC12 or latter
https://en.cppreference.com/w/cpp/compiler_support
https://github.com/earlephilhower/arduino-pico/issues/1379

afaik there is a paper about de-deprecation of these operations
https://www.open-std.org/jtc1/sc22/wg21/docs/papers/2021/p2327r1.pdf

Notably, gcc already implements it in master version (12, 13 include it already?)
https://github.com/gcc-mirror/gcc/commit/136029059686fed2d99c755baf35f98553fc0232

Also, https://docs.platformio.org/en/latest/projectconf/sections/env/options/build/build_src_flags.html and https://docs.platformio.org/en/latest/scripting/index.html should limit build flag to just your project files (unless our headers somehow interfere)

MacDada, 2023-04-23:
> C++20, probably not before also switching to GCC12 or latter

Well, I was confused at first, but C++20 _is_ working with the current setup (GCC 10.3.0), but it's experimental.

>  -std=gnu++20                Conform to the ISO 2020 C++ draft standard with
>                              GNU extensions (experimental and incomplete
>                             support).  Same as -std=gnu++2a.

https://community.platformio.org/t/std-gnu-20-is-shown-as-201709-nodemcuarduino/32514/3

> https://docs.platformio.org/en/latest/projectconf/sections/env/options/build/build_src_flags.html

I've tried it and I still still get the warnings.

Actually, yeah – [I wish I could somehow "separate" flags for _my_ and _external_ code…](https://community.platformio.org/t/silence-warnings-for-dependencies-external-libraries/33387?u=dvdnwk)

MacDada, 2023-04-30:
I have _temporarily_ disabled all `volatile` warnings: https://github.com/MacDada/DnWiFiDoorLock/commit/5533426e6a3e2b36b690b7ae90310587715b989a

mcspr, 2023-04-30:
> or if they fix the issues in the libs (not gonna happen ;p)

Feel free to provide a PR :)

Volatile use in I2C master seems to originate from old SDK behaviour where OS (aka ETS) timers were serviced as soon as the peripheral timer expired. This is no longer the case, timers are handled in a task instead. So, I would guess it might be pretty safe to just remove volatile code in those sections.

I2C slave code is using pin ISR, so some sync logic still needs to happen somewhere.
But we could also just mask interrupts while inside some of critical sections; manual `memw` and asm `memory` to sync vars still applies here, probably don't need to use `volatile` to enforce desired behaviour

Which libraries other than I2C are affected?

MacDada, 2023-04-30:
> Feel free to provide a PR :)

Ok, will try – I was asking about that [at the beginning: _2. Are you open for PRs in this matter?_](https://github.com/esp8266/Arduino/issues/8916#issue-1680007211) :-)

Mostly because after playing with C++/uC/Arduino/ESP8266/PIO for a year already, I can see that a lot of the projects are… kinda abandonware… Even simple issues/PRs get not answered/merged for many months… Hence asking if I'm gonna waste my time, trying to do anything – hopefully not!

> I would guess it might be pretty safe to just remove volatile code in those sections.

Oooof, my C++ brain is too small for that, yet. I might try simple substitution from the "short" `foo++` into extended `foo = foo + 1` version, but more complicated analysis are most likely outside of my abilities ;-)