esp8266/Arduino #8920: Weird compilation error related to typedef dns_found_callback since version 3.2

lmarmisa, 2023-05-08:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[since version 3.2]
- Development Env:		[Arduino IDE]
- Operating System:		[Ubuntu]

### Settings in IDE

- Module:  			[Wemos D1 mini r2 | Nodemcu]
- Flash Mode:			[all]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[160MHz]
- Upload Using:		[OTA | SERIAL]
- Upload Speed:   [115200]

### Problem Description

A weird compilation problem has appeared since version 3.2. It is related to _typedef void (*dns_found_callback)(const char *name, ipv4_addr_t *ipaddr, void *callback_arg);_ 

I get several compilation errors of this type: _error: conflicting declaration 'typedef void (* dns_found_callback)(const char*, ip4_addr_t*, void*)'_

Two files seem involved:
1) $HOME/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/tools/sdk/lwip2/include/lwip/dns.h:103:16
2) $HOME/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/tools/sdk/include/espconn.h:447:16

No compilation problem at all using release 3.0.2. None of my programs use the typedef  _dns_found_.

I would like to migrate to new releases, but I am unable to compile my project with the new toolchain & Arduino IDE.

The information provided by the compiler related to this error seem **_useless:_**
```
In file included from /home/luis/Arduino/CUCA2.1.rc068/modsyscommands.h:11,
                 from /home/luis/Arduino/CUCA2.1.rc068/modbutton.h:7,
                 from /home/luis/Arduino/CUCA2.1.rc068/modcommand.h:15,
                 from /home/luis/Arduino/CUCA2.1.rc068/modio.h:9,
                 from /home/luis/Arduino/CUCA2.1.rc068/CUCA.h:19,
                 from /home/luis/Arduino/CUCA2.1.rc068/modapps.h:5,
                 from /home/luis/Arduino/CUCA2.1.rc068/modapps.cpp:3:
/home/luis/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/tools/sdk/include/espconn.h:447:16: error: conflicting declaration 'typedef void (* dns_found_callback)(const char*, ip4_addr_t*, void*)'
  447 | typedef void (*dns_found_callback)(const char *name, ipv4_addr_t *ipaddr, void *callback_arg);
      |                ^~~~~~~~~~~~~~~~~~
In file included from /home/luis/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/libraries/ESP8266WiFi/src/ESP8266WiFiGeneric.h:29,
                 from /home/luis/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.h:28,
                 from /home/luis/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/libraries/ESP8266WiFi/src/ESP8266WiFi.h:34,
                 from /home/luis/Arduino/CUCA2.1.rc068/modsyscommands.h:7,
                 from /home/luis/Arduino/CUCA2.1.rc068/modbutton.h:7,
                 from /home/luis/Arduino/CUCA2.1.rc068/modcommand.h:15,
                 from /home/luis/Arduino/CUCA2.1.rc068/modio.h:9,
                 from /home/luis/Arduino/CUCA2.1.rc068/CUCA.h:19,
                 from /home/luis/Arduino/CUCA2.1.rc068/modapps.h:5,
                 from /home/luis/Arduino/CUCA2.1.rc068/modapps.cpp:3:
/home/luis/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/tools/sdk/lwip2/include/lwip/dns.h:103:16: note: previous declaration as 'typedef void (* dns_found_callback)(const char*, const ip_addr_t*, void*)'
  103 | typedef void (*dns_found_callback)(const char *name, const ip_addr_t *ipaddr, void *callback_arg);
      |                ^~~~~~~~~~~~~~~~~~
In file included from /home/luis/Arduino/CUCA2.1.rc068/modsyscommands.h:11,
                 from /home/luis/Arduino/CUCA2.1.rc068/modbutton.h:7,
                 from /home/luis/Arduino/CUCA2.1.rc068/modcommand.h:15,
[... 144 lines cut ...]
                 from /home/luis/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.h:28,
                 from /home/luis/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/libraries/ESP8266WiFi/src/ESP8266WiFi.h:34,
                 from /home/luis/Arduino/CUCA2.1.rc068/modwifi.h:7,
                 from /home/luis/Arduino/CUCA2.1.rc068/modsys.h:13,
                 from /home/luis/Arduino/CUCA2.1.rc068/modload.h:7,
                 from /home/luis/Arduino/CUCA2.1.rc068/modbuiltinfn.h:11,
                 from /home/luis/Arduino/CUCA2.1.rc068/modbuiltin.h:9,
                 from /home/luis/Arduino/CUCA2.1.rc068/modapps.h:7,
                 from /home/luis/Arduino/CUCA2.1.rc068/modcommand.h:19,
                 from /home/luis/Arduino/CUCA2.1.rc068/modio.h:9,
                 from /home/luis/Arduino/CUCA2.1.rc068/CUCA.h:19,
                 from /home/luis/Arduino/CUCA2.1.rc068/modbutton.h:5,
                 from /home/luis/Arduino/CUCA2.1.rc068/modbutton.cpp:1:
/home/luis/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/tools/sdk/lwip2/include/lwip/dns.h:103:16: note: previous declaration as 'typedef void (* dns_found_callback)(const char*, const ip_addr_t*, void*)'
  103 | typedef void (*dns_found_callback)(const char *name, const ip_addr_t *ipaddr, void *callback_arg);
      |                ^~~~~~~~~~~~~~~~~~
Se encontraron varias bibliotecas para "ArduinoOTA.h"
Usado: /home/luis/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/libraries/ArduinoOTA
 No usado: /home/luis/Arduino/libraries/ArduinoOTA
Se encontraron varias bibliotecas para "SoftwareSerial.h"
Usado: /home/luis/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/libraries/SoftwareSerial
 No usado: /home/luis/Arduino/libraries/EspSoftwareSerial
exit status 1
Error compilando para la tarjeta LOLIN(WEMOS) D1 R2 & mini.
```

d-a-v, 2023-05-09:
Everything should be back to normal after you comment the declaration on this line:
$HOME/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/tools/sdk/include/espconn.h:447:16

espconn backported from espressif is not widely used so we are not aware of every issues with it.

Please tell us if this solves your issue.

lmarmisa, 2023-05-09:
Hi d-a-v,

Thanks for the workaround. Working pretty well now.

I hope that this issue will be fixed in future releases.

Best regards,

Luis

```
. . . . . . 

/******************************************************************************
 * TypedefName : dns_found_callback
 * Description : Callback which is invoked when a hostname is found.
 * Parameters  : name -- pointer to the name that was looked up.
 *               ipaddr -- pointer to an ipv4_addr_t containing the IP address of
 *               the hostname, or NULL if the name could not be found (or on any
 *               other error).
 *               callback_arg -- a user-specified callback argument passed to
 *               dns_gethostbyname
*******************************************************************************/
 
//typedef void (*dns_found_callback)(const char *name, ipv4_addr_t *ipaddr, void *callback_arg);

. . . . . . 

```

closed 2023-05-09 by lmarmisa