esp8266/Arduino #8921: Feature Request: Add SLIP support to LWIP

ssshake, 2023-05-10:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					ESP-12

### Problem Description

I wish to use the SLIP protocol using an ESP8266 and the Arduino IDE, however LWIP for this repository does not have the SLIP code that does come with the standard ESP library found here: https://github.com/espressif/esp-lwip/blob/2.1.2-esp/src/netif/slipif.c

I am asking if you could please add this support so that I can continue development of my project in the Arduino ecosystem.

mcspr, 2023-05-10:
> https://github.com/espressif/esp-lwip/blob/2.1.2-esp/src/netif/slipif.c

repository is here, we do not use Espressif LWIP repo
* https://github.com/lwip-tcpip/lwip is our upstream
* https://github.com/d-a-v/esp82xx-nonos-linklayer is the builder + some patches

things just have to be built, don't see any dependencies and it has to be manually initialized so should be pretty safe to include

* slipif.c -> slipif.o missing in the object file generator - https://github.com/d-a-v/esp82xx-nonos-linklayer/blob/8e3c4eec00ec7f3962705ee095a310642953b5fb/makefiles/Makefile.lwip2#L4-L12 (same goes for cmake one)
* relevant slipif.h and sio.h missing from our local tools/sdk/lwip2/include/
* necessary boilerplate code for uart setup - like in esp-idf https://github.com/espressif/esp-protocols/tree/master/examples/esp_netif/slip_custom_netif - but, for HardwareSerial / SoftwareSerial / whatever that sends and receives bytes

ssshake, 2023-05-10:
OK thank you very much for the information! 

What I am trying to do is port this existing code based (https://github.com/martin-ger/esp_slip_router) so that is can be used within the Arduino environment. I am trying to use martin-ger's implementation of a SLIP router in my existing Arduino code base.

To be honest I am just learning as I go and this is quite a bit over my head, but if I could get the basics of SLIP working in Arduino that would get me moving in the right direction.