esp8266/Arduino #8922: round() macro in Arduino.h causes build errors

TD-er, 2023-05-10:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Core Version:			[2.7.4]
- Development Env:		[Platformio]
- Operating System:		[Ubuntu]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

Starting yesterday, I am no longer able to let GHActions build my project.

```
In file included from src/src/Helpers_tmp/../../ESPEasy_common.h:19:0,
                 from src/src/Helpers_tmp/__tmpfile.cpp:114:
/home/runner/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/bits/random.tcc: In member function 'void std::poisson_distribution<_IntType>::param_type::_M_initialize()':
/home/runner/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Arduino.h:137:22: error: expected unqualified-id before '(' token
 #define round(x)     ((x)>=0?(long)((x)+0.5):(long)((x)-0.5))
                      ^
/home/runner/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/bits/random.tcc: In member function 'void std::binomial_distribution<_IntType>::param_type::_M_initialize()':
/home/runner/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Arduino.h:137:22: error: expected unqualified-id before '(' token
 #define round(x)     ((x)>=0?(long)((x)+0.5):(long)((x)-0.5))
                      ^
/home/runner/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Arduino.h:137:22: error: expected unqualified-id before '(' token
 #define round(x)     ((x)>=0?(long)((x)+0.5):(long)((x)-0.5))
                      ^
*** [.pio/build/climate_ESP8266_4M1M/src/src/Helpers_tmp/__tmpfile.cpp.o] Error 1
========================= [FAILED] Took 95.87 seconds =========================
Environment           Status    Duration
--------------------  --------  ------------
climate_ESP8266_4M1M  FAILED    00:01:35.868
==================== 1 failed, 0 succeeded in 00:01:35.868 ====================

Traceback (most recent call last):
  File "/home/runner/work/ESPEasy/ESPEasy/tools/ci/build-and-archive.py", line 49, in <module>
    cmd(**opts)
  File "/home/runner/work/ESPEasy/ESPEasy/tools/ci/build-and-archive.py", line 18, in cmd
    subprocess.check_call(["platformio", "run", "-e", env])
  File "/opt/hostedtoolcache/Python/3.11.3/x64/lib/python3.11/subprocess.py", line 413, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['platformio', 'run', '-e', 'climate_ESP8266_4M1M']' returned non-zero exit status 1.
```

See for example [this build](https://github.com/letscontrolit/ESPEasy/actions/runs/4935471659/jobs/8821831438#step:8:780)

As far as I can see, this `round()` macro is defined in `Arduino.h` whenever `__cplusplus` isn't defined.

However I have absolutely no clue right now where this may be called from, so I can isolate the code calling it.

The problem is that I need to have some .cpp files concatenated to build on Windows (due to linker argument max. length) and it also helps to reduce the build size by roughly 2 - 3 kB.
Essentially this is no different from how the Arduino toolchain deals with .ino files.

When I try to keep out the `Helpers` folder of my code from being concatenated, the build doesn't complete due to size issues.
However I plan on trying to exclude some more code to see if something in that folder may cause these build failures.

Does anyone have an idea on how to find out what code may call for this template object to be compiled?
Problem is, on Windows it is building just fine and my WSL Ubuntu env is now being updated to the same platform as GHActions uses so I could hopefully reproduce it instead of abusing the GH Actions build system to find what's breaking my builds.

I got the impression it may have something to do with the latest PlatformIO, as I also had to update some of my Python build scripts since `env['CPPDEFINES']` no longer is a list, but a deque.

mcspr, 2023-05-10:
See https://github.com/esp8266/Arduino/issues/5787#issuecomment-466186521

Update to 3.1.x?
https://github.com/esp8266/Arduino/blob/57fa6cdc9295c30cbde7aa2e4205f8d52dbaba33/cores/esp8266/Arduino.h#L218-L224

If you going to use old Core version... either repeat the fix or just `undef round` before including any stl code that happens to use round() / using round() on different types in your own code. Might be tricky with libs, but the is also [`-U` flag](https://gcc.gnu.org/onlinedocs/gcc/Preprocessor-Options.html#index-U)

closed 2023-05-10 by mcspr

TD-er, 2023-05-10:
But the problem is that I have absolutely no idea where this is included in such a way that the compiler is not considering it as C++ code.
I've spent an incredible amount of hours in the last 2 days trying to replicate it locally and I really can't.

So that leaves me to either completely make a fork of the older core only to replace a single Arduino.h file, or move away from the old core to 3.x
But to be honest, then I think I will have to seriously considering to abandon supporting ESP8266 as it will simply be too much of a struggle to get things working on there.

mcspr, 2023-05-10:
I think we still misunderstand each other. Compiler does not parse things as C++ or C in individual code sections, it is for the whole file.

Error log above tells you about some kind of code from `<random>`. Whenever you include it **after** `Arduino.h`, template inline code using round() is replaced with our macro, which is invalid. Surely you could inspect the resulting tmpfile and see include order? Preprocessor runs top to bottom, so line to line order really really matters when dealing with C and C++ header files

TD-er, 2023-05-10:
Problem is that I am not even 100% sure it happens in the concat of the Helpers/*.cpp tmp file.
Purely when judging that concat file, Arduino.h should already have been included right at the top of the file.
But apparently somewhere `round` is being defined while it shouldn't as Arduino.h was already included and thus the wrapping #ifndef checks should have prevented further includes.

So like I said, I have no clue where to look and now GH Actions is (again) down....

mcspr, 2023-05-10:
Look out for something like this
```cpp
#include <cmath>
#include <Arduino.h>

void setup() {
    auto x = std::round(5.55);
}

void loop() {
}
```

`<cmath>` undefs round
`Arduino.h` defines it
oops when building

TD-er, 2023-05-10:
OK, that's a good point for further searching, thanks

mcspr, 2023-05-10:
...`math.h` too. When Arduino.h is at the top of the file, cmath undefs the macro and everything works correctly. If not, include cmath and undef does not happen 2nd time because of header guards.

Suppose tmpfile generator can just force Arduino.h to be the very first line. Also note that [file list order is arbitrary](https://github.com/letscontrolit/ESPEasy/blob/e14450ccbbde9decee6be0a9e5aae3eebef67131/tools/pio/concat_cpp_files.py#L31); using some kind of [manual sorting from python side](https://docs.python.org/3/library/os.html#os.listdir) would keep it consistent between CI and local builds.

TD-er, 2023-05-10:
I was thinking of something like this, which is quite like what you describe:

```c++
#include <Arduino.h> // See: https://github.com/esp8266/Arduino/issues/8922#issuecomment-1542301697
#include <cmath>
```

mcspr, 2023-05-10:
Right. Just to repeat the reasoning - Arduino *should* be first to include cmath / math.h, so our macro is unde'fed and replaced with a func
https://github.com/TD-er/ESPEasy/compare/feature/ESP32_C3...mcspr:ESPEasy:arduino8922

Doing a build at https://github.com/mcspr/ESPEasy/actions/runs/4938387997, waiting for queue...

(nb. random include was really misleading. it happens via bits/stl_algo.h, algorithms and containers want to include it so it should be a pretty common occurrence. sadly, it does not tell the originating file, and instead gives that weird error about brackets :/)

TD-er, 2023-05-10:
[image]
Yay!
The finalizing of the build failed, but that's probably related to the GH Actions outage and that it only was restarted to build the failed jobs due to this same outage.

But thanks (as always) for helping out finding this very unclear issue... (well now it is clear :) )