esp8266/Arduino #8923: Linking error since last post mortem improvments.

everslick, 2023-05-13:
Since commit https://github.com/esp8266/Arduino/pull/8839 my program fails to link:

```
Linking ../build/3.0.99/caroline/eng/alpha/emonio.bin
  Versions: 3.0.51-22-g8105810c-dirty, 3.1.2-1-g01d1c8e4-dirty
../build/3.0.99/caroline/eng/alpha/arduino.ar(core_esp8266_postmortem.cpp.o): in function `__wrap_system_restart_local':
core_esp8266_postmortem.cpp:(.text.__wrap_system_restart_local+0x2): dangerous relocation: j: cannot encode: (.text.postmortem_report+0x8c)
collect2: error: ld returned 1 exit status
make[1]: *** [/home/clemens/Devel/ESP/emonio-fw/sdk/makeEspArduino-6.7.0/makeEspArduino.mk:319: ../build/3.0.99/caroline/eng/alpha/emonio.bin] Error 1
make: *** [Makefile:86: src] Error 2
```

I use latest makeEspArduino and Arduino Core with the updated toolchain (tools/get.py).

Going back to the commit before the offending patch, it starts working again:

```
Creating core archive
Creating object archive
Linking ../build/3.0.99/caroline/eng/alpha/emonio.bin
  Versions: 3.0.51-22-g8105810c-dirty, 3.1.1-13-g1beca6f4-dirty
Creating BIN file "../build/3.0.99/caroline/eng/alpha/emonio.bin" using "/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP8266/bootloaders/eboot/eboot.elf" and "../build/3.0.99/caroline/eng/alpha/emonio.elf"
. Variables and constants in RAM (global, static), used 40948 / 80192 bytes (51%)
║   SEGMENT  BYTES    DESCRIPTION
╠══ DATA     1912     initialized variables
╠══ RODATA   8308     constants       
╚══ BSS      30728    zeroed variables
. Instruction RAM (IRAM_ATTR, ICACHE_RAM_ATTR), used 63489 / 65536 bytes (96%)
║   SEGMENT  BYTES    DESCRIPTION
╠══ ICACHE   32768    reserved space for flash instruction cache
╚══ IRAM     30721    code in IRAM    
. Code in flash (default, ICACHE_FLASH_ATTR), used 668888 / 1048576 bytes (63%)
║   SEGMENT  BYTES    DESCRIPTION
╚══ IROM     668888   code in flash   

Memory summary
  RAM:    40948 bytes
  Flash: 709829 bytes

LwIPVariant: v2 Lower Memory (no features)
Flash size: 4MB (FS:2MB OTA:~1019KB)

Build complete. Elapsed time: 5 seconds
```

Is there something I do wrong, or has anybody an idea what the real problem is?

mhightower83, 2023-05-13:
I don't have a failure case to test against; however, I think if you replace
https://github.com/esp8266/Arduino/blob/57fa6cdc9295c30cbde7aa2e4205f8d52dbaba33/cores/esp8266/core_esp8266_postmortem.cpp#L128
With 
```cpp
    "j.l          postmortem_report, a3\n\t"
```
that will resolve the link failure.

everslick, 2023-05-15:
I can happily confirm, that your PR fixes my problem. Thank you very much! I will close this issue, as soon as it gets merged.

everslick, 2023-06-16:
Fix merged.

closed 2023-06-16 by everslick