esp8266/Arduino #8926: analogWrite LED intensity with ESP8266

askondro, 2023-05-14:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:			[ESP8266 device]
- Core Version:			[3.1.2]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

The built-in (2) LED intensity goes down with the increased value.

The analogWrite(LED, 100); is dimmer than the analogWrite(LED, 254);

It is like it is inverted.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ESP8266WiFi.h>

#define LED 2

void setup() {
  Serial.begin(115200);
  pinMode(LED, OUTPUT);
  delay(100);
  Serial.println(F("\nESP8266 WiFi scan example"));

  // Set WiFi to station mode
  WiFi.mode(WIFI_STA);

  // Disconnect from an AP if it was previously connected
  WiFi.disconnect();
  delay(100);
}

void loop() {
  String ssid;
  int32_t rssi;
  uint8_t encryptionType;
  uint8_t *bssid;
  int32_t channel;
  bool hidden;
  int scanResult;

  Serial.println(F("Starting WiFi scan..."));

  Serial.printf(PSTR("Setting LED to %d"), 100);
  analogWrite(LED, 100);

  delay(10000);

  Serial.printf(PSTR("Setting LED to %d"), 254);
  analogWrite(LED, 254);

  // Wait a bit before scanning again
  delay(10000);
}
```

### Debug Messages

```
No debug messages,
```

askondro, 2023-05-14:
Sorry, i mean that analogWrite(LED, 100) has more light intensity than e.g. analogWrite(LED, 254).

earlephilhower, 2023-05-14:
That is the way the HW is built on your board.  The LED is negative logic.

For example, `digitalWrite(LED, LOW)` should turn the LED on full, while `digitalWrite(LED, HIGH)`  will turn it off.  The GPIO is used only to sink current, not source it, in this case.

closed 2023-05-14 by earlephilhower

askondro, 2023-05-14:
Thank you for your answer. Is there any logic behind that?

earlephilhower, 2023-05-14:
Most MCUs can sink more current than they can source, so it makes sense to connect the anode of the LED to VCC through a resistor, and the cathode to the MCU pin.  There's no real standard for the Arduino clone boards on this.

askondro, 2023-05-14:
Makes sense. Thank you for the explanation!