esp8266/Arduino #8938: Map function - IntegerDivideByZero

FernandoGarcia, 2023-06-06:
Hi!

Someone did a regression in map function and I'm getting this error.

[image]

This issue has been fixed [here](https://github.com/esp8266/Arduino/pull/2397) before.

Best regards.

mcspr, 2023-06-16:
So https://github.com/esp8266/Arduino/pull/7027 is the culprit. But code above does not check the value regardless, so our exception is replaced with a run-time error.

Looking at esp32 where PR was also merged, [it was eventually reverted back to original implementation](https://github.com/espressif/arduino-esp32/pull/6218) because of errors with larger values. With the provided test (0...180 to 1000...2000 and back), error rate is 0. With 0 and 10000 as min and max - error rate becomes 99%, mapping value and mapping value back are off-by-one :/

Suppose in servo case it makes sense, but not in general func context. Or, calc should take a different value (and btw, divide-by-2 effect is not explained even in servo case besides that it is 'better' :/)
https://github.com/esp8266/Arduino/blob/521ae60a89e64bb0d1eb7a0b7addf620ced5cad3/libraries/Servo/src/Servo.cpp#L36-L38

func|result|expected|
-|-|-
map(7500, 0, 10000, 10000, 0)|2501|2500
map(2501, 10000, 0, 0, 10000)|7499|7499
map(7499, 0, 10000, 10000, 0)|2502|2501
map(2502, 10000, 0, 0, 10000)|7498|7498
...and so on...|...|...

TD-er, 2023-07-16:
Shouldn't a simple fix like this be enough to prevent these crashes?
```c++
if (divisor == 0) return in_min;
```

Increasing resolution and int overflows with large values are much harder to fix.
But my gut feeling tells me there should be 3 separate cases:
- dividend > divisor
- dividend < divisor
- dividend == divisor

TD-er, 2023-07-16:
OK, just thought of something like this (still untested, just some gut feeling programming)

```c++
long map(long x, long in_min, long in_max, long out_min, long out_max) {
  const long divident = out_max - out_min;
  const long divisor = in_max - in_min;
  if (divisor == 0) return in_min;

  const long delta = x - in_min;

  if (divident == divisor) {
    return out_min + delta;
  }
  if (divident == (-1 * divisor)) {
    return out_max + delta;
  }
  if (divident > divisor) {
	const long error = divident % divisor;
	if (error == 0 || error == 1) {
      const long factor = divident / divisor;
      return (delta * factor) + out_min;
	} else {
      const long factor = (divident * error) / divisor;
      return (delta * factor) / error + out_min;
	}
  }
  // divident < divisor
  const long factor = divisor / divident;
  const long error = divisor % divident;
  return (delta + error) / factor + out_min;
}
```

TD-er, 2023-07-16:
Just did a quick test on [cpp.sh](https://cpp.sh/) with the last suggested option:

```c++

long map(long x, long in_min, long in_max, long out_min, long out_max) {
  const long divident = out_max - out_min;
  const long divisor = in_max - in_min;
  if (divisor == 0) return in_min;

  const long delta = x - in_min;

  if (divident == divisor) {
    return out_min + delta;
  }
  if (divisor < 0 && divident < 0) {
    return map(x, in_max, in_min, out_max, out_min);
  } else if (divisor < 0) {
    return map(in_max - x, in_max, in_min, out_min, out_max);
  } else if (divident < 0) {
    return map(in_max - x, in_min, in_max, out_max, out_min);
  }

  if (divident > divisor) {
    const long error = divident % divisor;
    const long factor = (divident + error) / divisor;
    return (delta * factor) + out_min;
  }
  // divident < divisor
  const long factor = divisor / divident;
  const long error = out_max - (in_max / factor + out_min);
  return (delta + error) / factor + out_min;
}

int main()
{
    long value = 0;
    for (size_t i = 1; i < 15; ++i) {
        value = (i/3) * 2500 - 1 + i%3;
        long output = map(value,  0, 10000, 10000, 0);
        long output2 = map(value,  0, 100, 10000, 0);
        long output3 = map(value,  0, 100, 0, 10000);
        std::cout << value << ":\t" << output << "\t" << output2 << "\t" << output3 << "\n";
    }
    
}
```

```
0:	10000	10000	0
1:	9999	9900	100
2499:	7501	-239900	249900
2500:	7500	-240000	250000
2501:	7499	-240100	250100
4999:	5001	-489900	499900
5000:	5000	-490000	500000
5001:	4999	-490100	500100
7499:	2501	-739900	749900
7500:	2500	-740000	750000
7501:	2499	-740100	750100
9999:	1	-989900	999900
10000:	0	-990000	1000000
10001:	-1	-990100	1000100
 ```

TD-er, 2023-07-17:
Made some more progress, but there's still something funky going on when going outside the input range and mapping to a longer range.

```c++
// Example program
# include <iostream>
# include <string>

long map(long x, long in_min, long in_max, long out_min, long out_max) {
  const long out_length = out_max - out_min;
  const long in_length  = in_max - in_min;

  if (in_length == 0) { return in_min; }

  if (out_length == 0) { return out_min; }

  const long delta = x - in_min;

  if (out_length == in_length) {
    return out_min + delta;
  }

  if (out_length == (-1 * in_length)) {
    return out_max + delta;
  }

  if (out_length < 0 && in_length < 0) {
    return map(x, in_max, in_min, out_max, out_min);
  } else if (out_length < 0) {
    return map(in_max - delta, in_min, in_max, out_max, out_min);
  } else if (in_length < 0) {
    return map(in_max - delta, in_max, in_min, out_min, out_max);
  }
  
  // We now know in_min < in_max and out_min < out_max
  // Make sure x is within range of in_min ... in_max
  if (x < in_min) {
    const long before_min = in_min - x;	
	long shift_factor = -1;
    shift_factor -= (before_min / in_length);
    const long in_shift = shift_factor * in_length;
    const long out_shift = shift_factor * out_length;
    return map(x, in_min + in_shift, in_max + in_shift, out_min + out_shift, out_max + out_shift);
  } else if (x > in_max) {
    const long passed_max = x - in_max;
	long shift_factor = 1;
    shift_factor += (passed_max / in_length);
    const long in_shift = shift_factor * in_length;
    const long out_shift = shift_factor * out_length;
    return map(x, in_min + in_shift, in_max + in_shift, out_min + out_shift, out_max + out_shift);
  }

  //  if (abs(delta) > abs(in_max - delta)) {
  //    return map(in_max - delta, in_max, in_min, out_min, out_max);
  //  }

  const long abs_out_length = abs(out_length);

  if (abs_out_length > abs(in_length)) {
    const long intermediate_divisor = delta == 0 ? 1 : abs(delta);
    const long intermediate_max     = (0x1FFFFFFF - abs(in_length / 2)) / intermediate_divisor;

    if ((2 * intermediate_max) > abs_out_length) {
      const long intermediate = (delta * intermediate_max * 2 + (in_length / 2)) / in_length - intermediate_max;
      return map(intermediate, -intermediate_max, intermediate_max, out_min, out_max);
    }
  }

  // abs(out_length) < abs(in_length)
  return (delta * out_length + (in_length / 2)) / in_length + out_min;
}

int main()
{
  long value = 0;

  for (size_t i = 1; i < 15; ++i) {
    value = (i / 3) * 2500 - 1 + i % 3;
    const long values[] = {
      map(value, 0,     1000, 255,   0),
      map(value, 0,     1000, 0,     255),
      map(value, 10000, 0,     0,     1024),
      map(value, 10000, 0,     1024,  0),
      map(value, 1000,  0,     0,     10240),
      map(value, 1000,  0,     10240, 0),
      map(value, 0,     1000,  0,     10240),
      map(value, 0,     1000,  10240, 0),
      map(value, 0,     10000, 10240, 0),
      map(value, 10000, 0,     10240, 0)
    };
    std::cout << value << ":";
    constexpr size_t nrvalues = sizeof(values) / sizeof(values[0]);

    for (size_t i = 0; i < nrvalues; ++i) {
      std::cout << "\t" << values[i];
    }
    std::cout << "\n";
  }
}
```

These cases show some funky behavior:
```c++
      map(value, 1000,  0,     10240, 0),
      map(value, 0,     1000,  0,     10240),
```

```
0:	255	0	1024	0	-1759	0	0	-1759	10240	0
1:	255	0	1024	0	-1757	-1	-1	-1757	10239	1
2499:	-382	637	768	256	-21361	19603	19603	-21361	7681	2559
2500:	-382	638	768	256	-21359	19601	19601	-21359	7680	2560
2501:	-383	638	768	256	-21357	19599	19599	-21357	7679	2561
4999:	-1020	1275	512	512	-40961	39203	39203	-40961	5121	5119
5000:	-1020	1275	512	512	-52959	51200	51200	-52959	5120	5120
5001:	-1020	1275	512	512	-52957	51199	51199	-52957	5119	5121
7499:	-1657	1912	256	768	-72561	70803	70803	-72561	2561	7679
7500:	-1657	1913	256	768	-72559	70801	70801	-72559	2560	7680
7501:	-1658	1913	256	768	-72557	70799	70799	-72557	2559	7681
9999:	-2295	2550	0	1024	-92161	90403	90403	-92161	1	10239
10000:	-2295	2550	0	1024	-104159	102400	102400	-104159	0	10240
10001:	-2295	2550	0	1024	-104157	102399	102399	-104157	-1	10241
```

TD-er, 2023-07-17:
And some extra checks for int overflow:

```c++
// Example program
# include <iostream>
# include <string>

long map(long x, long in_min, long in_max, long out_min, long out_max) {
  const long out_length = out_max - out_min;
  const long in_length  = in_max - in_min;

  if (in_length == 0) { return in_min; }

  if (out_length == 0) { return out_min; }

  const long delta = x - in_min;

  if (out_length == in_length) {
    return out_min + delta;
  }

  if (out_length == (-1 * in_length)) {
    return out_max + delta;
  }

  if ((out_length < 0) && (in_length < 0)) {
    return map(x, in_max, in_min, out_max, out_min);
  } else if (out_length < 0) {
    return map(in_max - delta, in_min, in_max, out_max, out_min);
  } else if (in_length < 0) {
    return map(in_max - delta, in_max, in_min, out_min, out_max);
  }

  // We now know in_min < in_max and out_min < out_max
  // Make sure x is within range of in_min ... in_max
  if ((x < in_min) || (x > in_max)) {
    long shift_factor = 0;

    if (x < in_min) {
      const long before_min = in_min - x;
      shift_factor = -1 - (before_min / in_length);
    } else if (x > in_max) {
      const long passed_max = x - in_max;
      shift_factor = 1 + (passed_max / in_length);
    }

    if (shift_factor != 0) {
      const long in_shift  = shift_factor * in_length;
      const long out_shift = shift_factor * out_length;
      return map(x, in_min + in_shift, in_max + in_shift, out_min + out_shift, out_max + out_shift);
    }
  }

  const long abs_out_length = abs(out_length);

  if (abs_out_length > abs(in_length)) {
    const long intermediate_max = (0x1FFFFFFF - abs(in_length / 2)) / abs_out_length;

    if (intermediate_max > (abs_out_length >> 2)) {
      const long intermediate = (delta * intermediate_max * 2 + (in_length / 2)) / in_length - intermediate_max;
      return map(intermediate, -intermediate_max, intermediate_max, out_min, out_max);
    }

    // Probably running into integer overflow, so use the slightly less accurate factor approach.
    const long factor = out_length / in_length;
    return (delta * factor) + out_min;
  }

  // abs(out_length) < abs(in_length)

  const unsigned long overflow_estimator =
    (static_cast<unsigned long>(delta) >> 8) *
    (static_cast<unsigned long>(out_length) >> 8) +
    (static_cast<unsigned long>(in_length / 2) >> 8);

  if (overflow_estimator > (1u << 16)) {
    // Would result in integer overflow, so use the slightly less accurate factor approach.
    const long factor = in_length / out_length;

    if (factor != 0) {
      return delta / factor + out_min;
    }
  }
  return (delta * out_length + (in_length / 2)) / in_length + out_min;
}

int main()
{
  long input = 0;

  for (size_t i = 1; i < 15; ++i) {
    input = (i / 3) * 2500 - 1 + i % 3;
    const long values[] = {
      map(input, 0,     1000,  255,      0),
      map(input, 0,     1000,  0,        255),
      map(input, 10000, 0,     0,        1024),
      map(input, 10000, 0,     1024,     0),
      map(input, 1000,  0,     0,        10240),
      map(input, 1000,  0,     10240,    0),
      map(input, 0,     1000,  0,        10240),
      map(input, 0,     1000,  10240,    0),
      map(input, 0,     10000, 10240,    0),
      map(input, 10000, 0,     10240,    0),
      map(input, 10000, 0,     10240000, 0),
      map(input, 100,   0,     10240000, 0)
    };
    std::cout << input << ":";
    constexpr size_t nrvalues = sizeof(values) / sizeof(values[0]);

    for (size_t i = 0; i < nrvalues; ++i) {
      std::cout << "\t" << values[i];
    }
    std::cout << "\n";
  }
}
```

Result:
```
0:	255	0	1024	0	10240	0	0	10240	10240	0	0	0
1:	255	0	1024	0	10230	10	10	10230	10239	1	1024	102400
2499:	-382	637	768	256	-15350	25590	25590	-15350	7681	2559	2558976	255897600
2500:	-382	638	768	256	-15360	25600	25600	-15360	7680	2560	2560000	256000000
2501:	-383	638	768	256	-15370	25610	25610	-15370	7679	2561	2561024	256102400
4999:	-1020	1275	512	512	-40950	51190	51190	-40950	5121	5119	5118976	511897600
5000:	-1020	1275	512	512	-40960	51200	51200	-40960	5120	5120	5120000	512000000
5001:	-1020	1275	512	512	-40970	51210	51210	-40970	5119	5121	5121024	512102400
7499:	-1657	1912	256	768	-66550	76790	76790	-66550	2561	7679	7678976	767897600
7500:	-1657	1913	256	768	-66560	76800	76800	-66560	2560	7680	7680000	768000000
7501:	-1658	1913	256	768	-66570	76810	76810	-66570	2559	7681	7681024	768102400
9999:	-2295	2550	0	1024	-92150	102390	102390	-92150	1	10239	10238976	1023897600
10000:	-2295	2550	0	1024	-92160	102400	102400	-92160	0	10240	10240000	1024000000
10001:	-2295	2550	0	1024	-92170	102410	102410	-92170	-1	10241	10241024	1024102400
```

TD-er, 2023-07-17:
Fixed another issue with "inverted shift" (same range, one counting up, one down) and added a test function to show the "error" compared to the interpolation function using `double`

```c++
// Example program
# include <iostream>
# include <string>

long map(long x, long in_min, long in_max, long out_min, long out_max) {
  const long out_length = out_max - out_min;
  const long in_length  = in_max - in_min;

  if (in_length == 0) { return in_min; }

  if (out_length == 0) { return out_min; }

  const long delta = x - in_min;

  if (out_length == in_length) {
    return out_min + delta;
  }

  if ((out_length < 0) && (in_length < 0)) {
    return map(x, in_max, in_min, out_max, out_min);
  } else if (out_length < 0) {
    return map(in_max - delta, in_min, in_max, out_max, out_min);
  } else if (in_length < 0) {
    return map(in_max - delta, in_max, in_min, out_min, out_max);
  }

  // We now know in_min < in_max and out_min < out_max
  // Make sure x is within range of in_min ... in_max
  if ((x < in_min) || (x > in_max)) {
    long shift_factor = 0;

    if (x < in_min) {
      const long before_min = in_min - x;
      shift_factor = -1 - (before_min / in_length);
    } else if (x > in_max) {
      const long passed_max = x - in_max;
      shift_factor = 1 + (passed_max / in_length);
    }

    if (shift_factor != 0) {
      const long in_shift  = shift_factor * in_length;
      const long out_shift = shift_factor * out_length;
      return map(x, in_min + in_shift, in_max + in_shift, out_min + out_shift, out_max + out_shift);
    }
  }

  const long abs_out_length = abs(out_length);

  if (abs_out_length > abs(in_length)) {
    const long intermediate_max = (0x1FFFFFFF - abs(in_length / 2)) / abs_out_length;

    if (intermediate_max > (abs_out_length >> 2)) {
      const long intermediate = (delta * intermediate_max * 2 + (in_length / 2)) / in_length - intermediate_max;
      return map(intermediate, -intermediate_max, intermediate_max, out_min, out_max);
    }

    // Probably running into integer overflow, so use the slightly less accurate factor approach.
    const long factor = out_length / in_length;
    return (delta * factor) + out_min;
  }

  // abs(out_length) < abs(in_length)

  const unsigned long overflow_estimator =
    (delta > 256 ? (static_cast<unsigned long>(delta) >> 8) : 1) *
    (out_length > 256 ? (static_cast<unsigned long>(out_length) >> 8) : 1) +
    (static_cast<unsigned long>(in_length / 2) >> 8);

  if (overflow_estimator > (1u << 16)) {
    // Would result in integer overflow, so use the slightly less accurate factor approach.
    const long factor = in_length / out_length;

    if (factor != 0) {
      return delta / factor + out_min;
    }
  }
  return (delta * out_length + (in_length / 2)) / in_length + out_min;
}

long test_map(long x, long in_min, long in_max, long out_min, long out_max) {
  const double out_length = out_max - out_min;
  const double in_length  = in_max - in_min;

  if (in_length == 0) { return in_min; }

  if (out_length == 0) { return out_min; }

  const double delta = x - in_min;
  const double value =  (delta * out_length + (in_length / 2)) / in_length + out_min;
//  return std::round(value);    
  return value;    
}

long check(long x, long in_min, long in_max, long out_min, long out_max) {
    const long floatvalue = test_map(x, in_min, in_max, out_min, out_max);
    const long map_value = map(x, in_min, in_max, out_min, out_max);
    return floatvalue - map_value;
    
}

int main()
{
  long input = 0;

  for (size_t i = 0; i < 15; ++i) {
    input = (i / 3) * 2500 - 1 + i % 3;
    const long values[] = {
      check(input, 0,     1000,  255,      0),
      check(input, 0,     1000,  0,        255),
      check(input, 10000, 0,     100,      10100),
      check(input, 10000, 0,     10100,    100),
      check(input, 10000, 0,     0,        1024),
      check(input, 10000, 0,     1024,     0),
      check(input, 1000,  0,     0,        10240),
      check(input, 1000,  0,     10240,    0),
      check(input, 0,     1000,  0,        10240),
      check(input, 0,     1000,  10240,    0),
      check(input, 0,     10000, 10240,    0),
      check(input, 10000, 0,     10240,    0),
      check(input, 10000, 0,     10240000, 0),
      check(input, 50,   0,     10240000, 0)
    };
    std::cout << input << ":";
    constexpr size_t nrvalues = sizeof(values) / sizeof(values[0]);

    for (size_t i = 0; i < nrvalues; ++i) {
      std::cout << "\t" << values[i];
    }
    std::cout << "\n";
  }
}
```

The "error" for each calculation.
```
-1:	0	0	0	0	0	0	0	1	1	0	0	1	1	1
0:	0	0	0	0	0	0	0	0	0	0	0	0	0	0
1:	0	0	0	0	0	0	0	0	0	0	0	0	0	0
2499:	1	0	0	0	0	0	1	0	0	1	0	0	0	0
2500:	0	0	0	0	0	0	1	0	0	1	0	0	0	0
2501:	1	0	0	0	0	0	1	0	0	1	0	0	0	0
4999:	1	0	0	0	0	0	1	0	0	1	0	0	0	0
5000:	1	0	0	0	0	0	1	0	0	1	0	0	0	0
5001:	1	0	0	0	0	0	1	0	0	1	0	0	0	0
7499:	1	0	0	0	0	0	1	0	0	1	0	0	0	0
7500:	0	0	0	0	0	0	1	0	0	1	0	0	0	0
7501:	1	0	0	0	0	0	1	0	0	1	0	0	0	0
9999:	1	0	0	0	0	0	1	0	0	1	0	0	0	0
10000:	1	0	0	0	0	0	1	0	0	1	0	0	0	0
10001:	1	0	0	0	0	0	1	0	0	1	1	0	0	0
 ```

I think this is as accurate as can be done using only integer calculations :)

TD-er, 2023-07-17:
OK, still could be improved :)

Now even more accurate with large numbers
```c++
// Example program
# include <iostream>
# include <string>

long map(long x, long in_min, long in_max, long out_min, long out_max) {
  const long out_length = out_max - out_min;
  const long in_length  = in_max - in_min;

  if (in_length == 0) { return in_min; }

  if (out_length == 0) { return out_min; }

  const long delta = x - in_min;

  if (out_length == in_length) {
    return out_min + delta;
  }

  if ((out_length < 0) && (in_length < 0)) {
    return map(x, in_max, in_min, out_max, out_min);
  } else if (out_length < 0) {
    return map(in_max - delta, in_min, in_max, out_max, out_min);
  } else if (in_length < 0) {
    return map(in_max - delta, in_max, in_min, out_min, out_max);
  }

  // We now know in_min < in_max and out_min < out_max
  // Make sure x is within range of in_min ... in_max
  if ((x < in_min) || (x > in_max)) {
    long shift_factor = 0;

    if (x < in_min) {
      const long before_min = in_min - x;
      shift_factor = -1 - (before_min / in_length);
    } else if (x > in_max) {
      const long passed_max = x - in_max;
      shift_factor = 1 + (passed_max / in_length);
    }

    if (shift_factor != 0) {
      const long in_shift  = shift_factor * in_length;
      const long out_shift = shift_factor * out_length;
      return map(x, in_min + in_shift, in_max + in_shift, out_min + out_shift, out_max + out_shift);
    }
  }

  const long abs_out_length = abs(out_length);

  if (abs_out_length > abs(in_length)) {
    const long intermediate_max = (0x2FFFFFFF - abs(in_length)) / abs_out_length;

    if (intermediate_max > (abs_out_length >> 2)) {
      const long intermediate = (delta * intermediate_max * 2 + (in_length / 2)) / in_length - intermediate_max;
      return map(intermediate, -intermediate_max, intermediate_max, out_min, out_max);
    }

    // Probably running into integer overflow, so use the slightly less accurate factor approach.
    const long factor = out_length / in_length;
    const long error_mod =  out_length % in_length;
    const long error = (delta * error_mod) / in_length;
    return (delta * factor) + out_min + error;
  }

  // abs(out_length) < abs(in_length)
  const unsigned long overflow_estimator =
    (delta > 256 ? (static_cast<unsigned long>(delta) >> 8) : 1) *
    (out_length > 256 ? (static_cast<unsigned long>(out_length) >> 8) : 1) +
    (static_cast<unsigned long>(in_length / 2) >> 8);

  if (overflow_estimator > (1u << 16)) {
    // Would result in integer overflow, so use the slightly less accurate factor approach.
    const long factor = in_length / out_length;
    const long error_mod = in_length % out_length;
    const long error = (error_mod == 0 || delta == 0) ? 0 : out_length / (delta * error_mod);

    return delta / factor + out_min + error;
  }
  return (delta * out_length + (in_length / 2)) / in_length + out_min;
}

long test_map(long x, long in_min, long in_max, long out_min, long out_max) {
  const double out_length = out_max - out_min;
  const double in_length  = in_max - in_min;

  if (in_length == 0) { return in_min; }

  if (out_length == 0) { return out_min; }

  const double delta = x - in_min;
  const double value =  (delta * out_length + (in_length / 2)) / in_length + out_min;
//  return std::round(value);    
  return value;    
}

long check(long x, long in_min, long in_max, long out_min, long out_max) {
    const long floatvalue = test_map(x, in_min, in_max, out_min, out_max);
    const long map_value = map(x, in_min, in_max, out_min, out_max);
    return floatvalue - map_value;
    
}

int main()
{
  long input = 0;

  for (size_t i = 0; i < 15; ++i) {
    input = (i / 3) * 2500 - 1 + i % 3;
    const long values[] = {
      check(input, 0,     1000,  255,      0),
      check(input, 0,     1000,  0,        255),
      check(input, 10000, 0,     100,      10100),
      check(input, 10000, 0,     10100,    100),
      check(input, 10000, 0,     0,        1024),
      check(input, 10000, 0,     1024,     0),
      check(input, 1000,  0,     0,        10240),
      check(input, 1000,  0,     10240,    0),
      check(input, 0,     1000,  0,        10240),
      check(input, 0,     1000,  10240,    0),
      check(input, 0,     10000, 10240,    0),
      check(input, 10234567, 0,  1234567,   100),
      check(input, 10000, 0,     10234567, 0),
      check(input, 50,    1,     10234567, 0)
    };
    std::cout << input << ":";
    constexpr size_t nrvalues = sizeof(values) / sizeof(values[0]);

    for (size_t i = 0; i < nrvalues; ++i) {
      std::cout << "\t" << values[i];
    }
    std::cout << "\n";
  }
}
```

Output:
```
-1:	0	0	0	0	0	0	0	1	1	0	0	-44853	2	2
0:	0	0	0	0	0	0	0	0	0	0	0	0	0	1
1:	0	0	0	0	0	0	0	0	0	0	0	0	0	0
2499:	1	0	0	0	0	0	1	0	0	1	0	418	0	0
2500:	0	0	0	0	0	0	1	0	0	1	0	419	1	0
2501:	1	0	0	0	0	0	1	0	0	1	0	419	0	1
4999:	1	0	0	0	0	0	1	0	0	1	0	-21	0	0
5000:	1	0	0	0	0	0	1	0	0	1	0	-22	1	1
5001:	1	0	0	0	0	0	1	0	0	1	0	-22	1	0
7499:	1	0	0	0	0	0	1	0	0	1	0	-32	1	1
7500:	0	0	0	0	0	0	1	0	0	1	0	-32	0	0
7501:	1	0	0	0	0	0	1	0	0	1	0	-32	1	0
9999:	1	0	0	0	0	0	1	0	0	1	0	-43	1	0
10000:	1	0	0	0	0	0	1	0	0	1	0	-44	0	0
10001:	1	0	0	0	0	0	1	0	0	1	1	-44	0	1
```