esp8266/Arduino #8939: Feature Request: TCP Keepalive in WiFiClientSecure

mobizt, 2023-06-08:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Core Version:			latest (3.1.2)

### Problem Description

The TCP Keepalive was already implemented in WiFiClient class but it was missing from WiFiClientSecure class. 

Can you add support TCP Keepalive in WiFiClientSecure class?

https://github.com/esp8266/Arduino/blob/57fa6cdc9295c30cbde7aa2e4205f8d52dbaba33/libraries/ESP8266WiFi/src/WiFiClient.cpp#L417-L440

earlephilhower, 2023-06-09:
WiFiClientSecure already implements this through the magic of C++ inheritance, no?  The following snippet builds fine:
````
#include <ESP8266WiFi.h>

void setup() {
  WiFiClientSecure s;
  s.keepAlive(1,2,3);
}

void loop() { }
````

mobizt, 2023-06-09:
I test with your simple code and device crashed with this error in Serial monitor.

```
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x40207056 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffe30 end: 3fffffd0 offset: 0150
3fffff80:  40207558 00000000 00001388 feefeffe  
3fffff90:  00000000 00000000 00000000 3ffef504  
3fffffa0:  3ffef504 3fff0e1c feefeffe feefeffe  
3fffffb0:  feefeffe feefeffe 3ffee6f8 40204510  
3fffffc0:  feefeffe feefeffe 3fffdab0 40100c39  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v0005c970
~ld

```

And this is the decoded stack trace.

```
Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
PC: 0x40207056
EXCVADDR: 0x00000000

Decoding stack results
0x40204510: loop_wrapper() at C:\Users\Suwatchai\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266\core_esp8266_main.cpp line 255

```
I think WiFiClientSecure should provide the public functions to calling this private WiFiClientSecure context.
https://github.com/esp8266/Arduino/blob/57fa6cdc9295c30cbde7aa2e4205f8d52dbaba33/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.h#L362

mobizt, 2023-06-09:
I have tested this before to make sure that we can't call `keepAlive` directly from WiFiClientSecure class. The error is the same as I post above.

I can call `keepAlive` from `_ctx` without error if I edit [WiFiClientSecureBearSSL.h](https://github.com/esp8266/Arduino/blob/57fa6cdc9295c30cbde7aa2e4205f8d52dbaba33/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.h#L362) to make this public access.
`std::shared_ptr<WiFiClientSecureCtx> _ctx; `

That is why I request this feature here for WiFiClientSecure.

mobizt, 2023-06-09:
Ok, I will submit a PR for this.

commit e05656bd (2023-06-11): Add support WiFiClientSecure TCP KeepAlive (#8940)

* Add support WiFiClientSecure TCP KeepAlive
* Make TCP keepalive and related functions virtual.
* Make TCP keepalive and related functions override.

Fixes #8939
closed 2023-06-11 by earlephilhower