esp8266/Arduino #8945: python3 Parameters are incorrect?

JTzLinux, 2023-06-17:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:			LOLIN D1
- Core Version:			3.1.1
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			Wemos D1 mini r2
- Flash Mode:			other
- Flash Size: 		        Unknown
- lwip Variant:                   Unknown
- Reset Method:		Button on hw
- Flash Frequency:	        40Mhz?
- CPU Frequency:		80Mhz?
- Upload Using:		SERIAL
- Upload Speed:               115200

### Problem Description

When I try to upload it I receive that the parameters are incorrect 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}

```

### Debug Messages

```

fork/exec C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1/python3.exe: The parameter is incorrect.

Compilation error: fork/exec C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1/python3.exe: The parameter is incorrect.
```

JTzLinux, 2023-06-17:
Had to downgrade from 3.X.X to 2.X.X

closed 2023-06-17 by JTzLinux

mcspr, 2023-06-18:
> Had to downgrade from 3.X.X to 2.X.X

> 3.1.1

What about 3.1.2 aka latest version?

JTzLinux, 2023-06-19:
Same jizz aka same error

JTzLinux, 2023-06-19:
> Anything before python.exe line? Usually python produces a lengthy log
>
> Traceback (most recent call last):
> ...

no only what i wrote came back

mcspr, 2023-06-19:
Win+R - C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1\python3.exe does work though?

Is it portable IDE, Microsoft Store, or globally installed one? 1.x or 2.x?
Antivirus / 'security' sw interfering, maybe?