esp8266/Arduino #8951: Cannot use secure web server and secure wifi client together

PreyMa, 2023-07-03:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git). *
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

\* Latest version availble through the Arduino IDE

#### Platform

- Hardware:					ESP-12F (ESP8266MOD)
- Core Version:			3.1.2
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			NodeMCU 1.0 (ESP-12E Module)
- Flash Mode:			?
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		?
- Flash Frequency:	?
- CPU Frequency:		80MHz & 160MHz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

I want to use both a `ESP8266WebServerSecure` for serving a webpage and a `WiFiClientSecure` for sending MQTT messages to a broker securely. However, when the wifi client is used once, the web server stops working. As a work-around I tried destroying and restarting the server and client after every request. This unblocks the webserver, but the system hangs after a short while, rendering this solution unfit. It seems like, accessing the webpage while the MQTT client tries to establish a connection blocks the system.

This problem first occured in a larger project of mine, where I applied the same work-around, but here the system crashes entirely after some time. I wrote [a post](https://www.esp8266.com/viewtopic.php?f=8&t=24241) on the forum about this issue, but did not get any response. 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

The test code has two defined flags (`SECURE_CLIENT` and `RESET_AFTER_CLIENT_REQUEST`) that allow for different test cases. If `SECURE_CLIENT` is set to `0`, the web server uses TLS, but the client does not. Setting `RESET_AFTER_CLIENT_REQUEST` to `1` enables the work-around described above. There are also other defines for user specific configuration values, like wifi SSID, which are left blank.

A secure web server and a (secure) wifi client are created, with a MQTT pubsub client. Every 10 seconds a MQTT message is sent to a broker.

```cpp

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>
#include <PubSubClient.h>

#include <ESP8266WebServerSecure.h>

unsigned char privateKeyBinaryData[1191] = {
	/** Private Key **/
};

unsigned char certificateBinaryData[901] = {
	/** Certificate **/
};

#define WIFI_NAME "..."
#define WIFI_PWD "..."
#define BROKER_FINGERPRINT "..."
#define BROKER_IP "..."
#define MQTT_ID "..."
#define MQTT_USER "..."
#define MQTT_PWD "..."
#define MQTT_PATH "..."

#define SECURE_CLIENT 0
#define RESET_AFTER_CLIENT_REQUEST 0

[... 416 lines cut ...]
Route not found
HTTPServer: error: sent 0 on 122 bytes
HTTPServer: error: short send after timeout (0 < 11)
webserver: peer has closed after served
Drop client
New client
http-server loop: conn=1 avail=0 status=wait-read
Sending mqtt message
BSSL:_run_until: Not connected
Connecting to mqttBSSL:_connectSSL: start connection
BSSL:_connectSSL: OOM error
.BSSL:_connectSSL: start connection
BSSL:_connectSSL: OOM error
.BSSL:_connectSSL: start connection
BSSL:_connectSSL: OOM error
.BSSL:_connectSSL: start connection
BSSL:_connectSSL: OOM error
.BSSL:_connectSSL: start connection
BSSL:_connectSSL: OOM error
.BSSL:_connectSSL: start connection
BSSL:_connectSSL: OOM error
.BSSL:_connectSSL: start connection
BSSL:_connectSSL: OOM error

```

earlephilhower, 2023-07-03:
The error message is pretty clear.  You're Out Of Memory OOM.  You can't have both running in the limited RAM on the 8266.

closed 2023-07-03 by earlephilhower

PreyMa, 2023-07-04:
Hello, thank you for your quick response and clarifying the issue. Yes, the error message is pretty clear. But as far as I know the documentation does not discourage this use-case, which is why I wanted to make sure I did not do anything else wrong. :^)