esp8266/Arduino #8952: No poison after block with current git version

hmueller01, 2023-07-06:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				[ESP8266 device]
- Core Version:			[framework-arduinoespressif8266 @ 3.2.0-dev+sha.521ae60]
- Development Env:		[Platformio]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[Shelly 1]
- Flash Mode:			[qio]
- Flash Size: 			[2MB]
- lwip Variant:			[default, not set]
- Reset Method:		[nodemcu]
- Flash Frequency:		[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:		[115200] (serial upload only)

### Problem Description

After connecting to the AP the module panics
```
add 1
aid 1
station: a4:83:e7:d2:34:b7 join, AID = 1
No poison after block at: 0xe5a4e455, actual data: 0x80 0x0 0x0 0x0

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Panic umm_malloc.cpp:751 umm_malloc_core

>>>stack>>>
```
Decoded stack dump see below.

**The same code works well with Arduino Release 3.1.2.**

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

There is a lot of code I wrote, so it is not really possible to show it here. But showing the lines noted in the stack dump might help.

/Users/hmueller/src/PlatformIO/Projects/rhts-platformio/src/rhts-arduino.cpp:142
is a global definition of https://github.com/hmueller01/ESP_WiFiManager_Lite
```cpp
ESP_WiFiManager_Lite m_wifi_manager;
```

0x4020dc73: ESP_WiFiManager_Lite::run() at /Users/hmueller/src/PlatformIO/Projects/rhts-platformio/lib/ESP_WiFiManager_Lite.hmueller01/src/ESP_WiFiManager_Lite.h:921
```cpp
            if ( String(ARDUINO_BOARD) == "ESP32S2_DEV" )
```

ARDUINO_BOARD expands to "PLATFORMIO_SHELLY1"

So it looks like the new git platform doesn't like this. Or am I wrong? If you need more code details please let me know.

### Debug Messages

```
[... 43 lines cut ...]
0x3ffef690: _Pri_3_Stack at /Users/hmueller/src/PlatformIO/Projects/rhts-platformio/src/rhts-arduino.cpp:142
0x4020dc73: ESP_WiFiManager_Lite::run() at /Users/hmueller/src/PlatformIO/Projects/rhts-platformio/lib/ESP_WiFiManager_Lite.hmueller01/src/ESP_WiFiManager_Lite.h:921
0x3ffefa84: _Pri_3_Stack at /Users/hmueller/src/PlatformIO/Projects/rhts-platformio/include/dynamicParams.h:51
0x3ffefa84: _Pri_3_Stack at /Users/hmueller/src/PlatformIO/Projects/rhts-platformio/include/dynamicParams.h:51
0x3ffef690: _Pri_3_Stack at /Users/hmueller/src/PlatformIO/Projects/rhts-platformio/src/rhts-arduino.cpp:142
0x40209130: wifiSetup() at /Users/hmueller/src/PlatformIO/Projects/rhts-platformio/src/rhts-arduino.cpp:338
0x40223ab0: std::_Function_handler<void (WiFiEventStationModeGotIP const&), wifiSetup()::{lambda(WiFiEventStationModeGotIP const&)#1}>::_M_manager(std::_Any_data&, std::_Function_handler<void (WiFiEventStationModeGotIP const&), wifiSetup()::{lambda(WiFiEventStationModeGotIP const&)#1}> const&, std::_Manager_operation) at /Users/hmueller/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/std_function.h:271
0x3ffeffbc: _Pri_3_Stack at nmi-vector.o:?
0x3ffef690: _Pri_3_Stack at /Users/hmueller/src/PlatformIO/Projects/rhts-platformio/src/rhts-arduino.cpp:142
0x4020de00: loop at /Users/hmueller/src/PlatformIO/Projects/rhts-platformio/src/rhts-arduino.cpp:1427
0x3ffefd4c: _Pri_3_Stack at nmi-vector.o:?
0x3ffefdb0: _Pri_3_Stack at nmi-vector.o:?
0x402188fc: Print::print(char const*) at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/cores/esp8266/Print.cpp:122
0x3ffef690: _Pri_3_Stack at /Users/hmueller/src/PlatformIO/Projects/rhts-platformio/src/rhts-arduino.cpp:142
0x3ffefd4c: _Pri_3_Stack at nmi-vector.o:?
0x3ffefdb0: _Pri_3_Stack at nmi-vector.o:?
0x402092e1: setup at /Users/hmueller/src/PlatformIO/Projects/rhts-platformio/src/rhts-arduino.cpp:1402
0x40100470: ets_post at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/cores/esp8266/core_esp8266_main.cpp:238
0x3ffeff90: _Pri_3_Stack at nmi-vector.o:?
0x40100491: esp_schedule at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/cores/esp8266/core_esp8266_main.cpp:136
0x3ffeff90: _Pri_3_Stack at nmi-vector.o:?
0x4021b608: loop_wrapper() at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/cores/esp8266/core_esp8266_main.cpp:258
0x401003c9: cont_wrapper at ??:?

```

mhightower83, 2023-07-08:
@hmueller01 I see the bug. I should have a PR later today or tomorrow.

For an MCVE the "CaptivePortal" example shows the issue, after adding `Serial.begin`. (`Serial.begin` makes the stack trace visible.)
```cpp
#include <ESP8266WiFi.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>

const byte DNS_PORT = 53;
IPAddress apIP(172, 217, 28, 1);
DNSServer dnsServer;
ESP8266WebServer webServer(80);

String responseHTML = ""
                      "<!DOCTYPE html><html lang='en'><head>"
                      "<meta name='viewport' content='width=device-width'>"
                      "<title>CaptivePortal</title></head><body>"
                      "<h1>Hello World!</h1><p>This is a captive portal example."
                      " All requests will be redirected here.</p></body></html>";

void setup() {
  Serial.begin(115200);
  delay(200);
  Serial.printf("\r\n\r\nDNSServer CaptivePortal example\r\n");
   
  WiFi.mode(WIFI_AP);
  WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
  WiFi.softAP("DNSServer CaptivePortal example");

  // if DNSServer is started with "*" for domain name, it will reply with
  // provided IP to all DNS request
  dnsServer.start(DNS_PORT, "*", apIP);

  // replay to all requests with same HTML
  webServer.onNotFound([]() {
    webServer.send(200, "text/html", responseHTML);
  });
  webServer.begin();
}

void loop() {
  dnsServer.processNextRequest();
  webServer.handleClient();
}
```

hmueller01, 2023-07-10:
I will check your PR. But it might take a few days. Thanks for the quick solution!

hmueller01, 2023-07-12:
Ok, checked your PR. It fixes all my problems (had panics at lots of locations in my code). Thanks a lot!!!
What a simple thing that costs me hours of debugging my code (as I thought I did something wrong or running out of memory). Please bring this to develop asap as it might fix problems for other users too.

closed 2023-07-25 by hmueller01