esp8266/Arduino #8956: esp8266 wifi and dual band 2.4GHz/5GHz router

dmikushin, 2023-07-11:
Dear all, thank you for all your great work!

I'm aware that esp8266 is 2.4GHz band only. I don't understand though what prevents esp8266 from connecting to a dual-band router. I'm seeing my other 2.4GHz-only devices successfully being connected to the same ssid. So why does not it work for esp8266 exactly? Is it a hardware limitation or maybe a logical issue in the esp wifi software published here?

15-ederch, 2023-07-12:
They electricity is maybe low so that the router is connect with the
capacity of electricity

TD-er, 2023-07-18:
Try forcing the ESP to use only 802.11g and not 802.11n
Some router/access points try to steer clients to another band (literally called band steering) by shortly disconnecting the client.
It seems like routers might not do this when a device only claims to support 802.11g as this band steering feature is unknown to devices not supporting 802.11n