esp8266/Arduino #8959: How to fix "ets Jan  8 2013,rst cause:2, boot mode:(1,7)" log

flakka2022, 2023-07-18:
#### Platform

- Hardware:					ESP-12
- Core Version:			[
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			Wemos D1 mini
- Flash Mode:			
- Flash Size: 		4MB
- lwip Variant:   
- Reset Method:		
- Flash Frequency:	
- CPU Frequency:		
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

it cant boot to program i upload . no matter what program is . serial output is always as follow :
```
ets Jan  8 2013,rst cause:2, boot mode:(1,7)
```
or sometimes it show up
```
ets Jan  8 2013,rst cause:2, boot mode:(1,6)
```

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

// Digital pin D2 connected to an LED
int ledPin = 2;

void setup() {
  // Set pin D2 as output 
  pinMode(ledPin, OUTPUT);
}

void loop() {
  // Turn the LED on
  digitalWrite(ledPin, HIGH);  
  delay(1000); // Wait 1 second

  // Turn the LED off  
  digitalWrite(ledPin, LOW);   
  delay(1000); // Wait 1 second
}

```

### Debug Messages

```
ets Jan  8 2013,rst cause:2, boot mode:(1,7)
```
or sometimes it show up
```
ets Jan  8 2013,rst cause:2, boot mode:(1,6)
```

enjoyneering, 2023-07-18:
`int ledPin = 2;`
It is not D2 pin. It is GPOI2. If you want D2 use:
`int ledPin = D2; //GPIO4`

see [pins_arduino.h](https://github.com/esp8266/Arduino/blob/521ae60a89e64bb0d1eb7a0b7addf620ced5cad3/variants/d1_mini/pins_arduino.h#L41)

JAndrassy, 2023-07-18:
boot mode:(1
your esp8266 is still in flashing mode

closed 2023-09-14 by mcspr

flakka2022, 2023-11-03:
> boot mode:(1 your esp8266 is still in flashing mode

yes i can write my program but it doesn't run anything after that .

gordio, 2024-06-26:
Need to pull GPIO0 to VCC (or disconnect, it must be pulled internally to vcc) and reboot esp.

yangxiaokong, 2024-10-25:
I have the same problem, how to solve it