esp8266/Arduino #8962: OTA upload using arduino-cli?

blue-wind-25, 2023-07-20:
Hello,

OTA upload works with Arduino IDE 2.1.1 (it always ask for OTA password even if I do not use OTA password. but it still works well :-)

However it does not work using arduino-cli. I have downloaded the latest arduino-cli from https://github.com/arduino/arduino-cli/releases and then execute (after compile):

`arduino-cli upload --fqbn "esp8266:esp8266:nodemcuv2" --board-options "ip=lm6f" --protocol network --port "192.168.1.113" "OTA_Test"`

It outputs:
`Uploading to specified board using network protocol requires the following info:`
`Error getting user input: user input not supported in non interactive mode`

Note that if I use invalid IP (for example 192.168.1.200), it outputs:
`Error getting port metadata: port not found: 192.168.1.200 network`

So at least it can detect if there is something at '192.168.1.113'.

Is this mean that Arduino ESP8266 core does not support OTA upload using arduino-cli? Or perhaps I missed something in the command line parameter?

mcspr, 2023-07-27:
Code implementing password input seems to require user input through terminal, so that would explain the 'user input not supported in non interactive mode'
https://github.com/search?q=repo%3Aarduino%2Farduino-cli%20user%20provided%20fields&type=code

ref.
https://arduino.github.io/arduino-cli/dev/platform-specification/#user-provided-fields

I can't reproduce with latest 0.33.1, although it *does* sometimes fail to recognise the board and `--discovery-timeout` needs to be longer than the default 1s

blue-wind-25, 2023-07-27:
Hello,
Thank you for your response.

Is it possible that Arduino CLI does not support user input from Windows 7
command prompt? Maybe Win 7 SP 1 is too old?

I am unable to test it properly in Linux because no matter how large the
value I used for
--discovery-timeout, the CLI never found the board, even if MDNS discovery
found it:

~/.arduino15/packages/builtin/tools/mdns-discovery/1.0.9/mdns-discovery
%HOMEDRIVE%%HOMEPATH%\AppData\Local\Arduino15\packages\builtin\tools\mdns-discovery\1.0.8\mdns-discovery
    HELLO 1 "arduino-cli 0.33.0"
    START_SYNC

    {
      "eventType": "add",
      "port": {
        "address": "192.168.1.112",
        "label": "esp8266-7a2798 at 192.168.1.112",
        "protocol": "network",
        "protocolLabel": "Network Port",
        "properties": {
          ".": "nodemcuv2",
          "auth_upload": "no",
          "board": "nodemcuv2",
          "hostname": "esp8266-7a2798.local.",
          "port": "8266",
          "ssh_upload": "no",
          "tcp_check": "no"
        }
      }
    }

/opt/arduino-cli_0.33.1_Linux_64bit/arduino-cli board list
--discovery-timeout 3s
Port          Protocol Type             Board Name
FQBN                      Core
/dev/ttyUSB0  serial   Serial Port(USB) Unknown

Maybe my Linux (Centos 7) is also too old?

Message ID: ***@***.***>
>

blue-wind-25, 2023-07-27:
Sorry, I think I accidentally closed the issue, so I reopened it.

Or you actually have closed it before? If so, you can close it again.

Hmm, is there a more appropriate place to discuss something like this besides using issue?

mcspr, 2023-07-27:
> Linux (Centos 7)

firewalld blocking things? mdns-discovery needs to be allowed to bind on local ports, it sends stuff to 224.0.0.251:5353 and expects response on that local port

> Windows 7

No idea. Maybe golang has invalid code for that case, but that would be arduino-cli issue to fix. Perhaps an issue there would be better suited.

Specific piece of code is this, calling golang.org/x/term
https://github.com/arduino/arduino-cli/blob/62bb4c47cb2bb64f642cba050a958f45902202ee/internal/cli/feedback/terminal.go#L45

> Hmm, is there a more appropriate place to discuss something like this besides using issue?

^ and to the right, 'Convert to discussion'

blue-wind-25, 2023-07-27:
Okay, I'll tested it as soon as possible with firewall disabled and write the result here.

BTW, how if OTA uses other port than 8266, is it possible write `--port "192.168.1.113:5000` for example?

blue-wind-25, 2023-07-27:
Hello again,

Yes it is firewalld blocking things. When I disabled firewalld, it works both when the target is in STA and AP mode. So it is good :-)

However, using hostname such as `nodemcu.local` instead of IP does not work even if in command line `ping nodemcu.local` works and gives a valid IP. It seems that the tools requires the port to be an IP address? If it is, then it is no problem.

Also, specifying non standard network port does not work, `--port 192.168.1.113:8266` prints:
`Error getting port metadata: port not found: 192.168.0.113:8266 network`
So, I guess the tools does not support specifying custom OTA port? If so, it is also OK, because I think using other ports than the default of 8266 is very rare.

The code in https://github.com/arduino/arduino-cli/blob/62bb4c47cb2bb64f642cba050a958f45902202ee/internal/cli/feedback/terminal.go#L45 looks simple enough. So may be the bug is in golang, but I am not sure if I should report this to arduino-cli or to golang because Win 7 SP1 does indeed obsolete.

Thank you very much for your help.

You can close this issue for real now if my assumptions above are correct.

blue-wind-25, 2023-07-27:
Hello,

It seems the problem on Win 7 has been resolved after reboot, now Arduino CLI can upload using network when invoked from command prompt. So it may actually not a Golang/Ardu-CLI bug but something else.

mcspr, 2023-07-28:
> However, using hostname such as nodemcu.local instead of IP does not work even if in command line ping nodemcu.local works and gives a valid IP. It seems that the tools requires the port to be an IP address? If it is, then it is no problem.

> Also, specifying non standard network port does not work, --port 192.168.1.113:8266 prints:

Port means board name as discovered by the tool, see `arduino-cli board list`. `:8266` is not a part of it, we would have to use 'user fields' from the link above to expose it to the user.
i.e. it does not resolve domain names, but allows you to select one of the matching boards that it would find through discovery process

If terminal detection does not work with Win7 and always prints that 'interactive mode' error, I would open an issue at arduino/arduino-cli

blue-wind-25, 2023-07-28:
> Port means board name as discovered by the tool, see arduino-cli board list. :8266 is not a part of it, we would have to use 'user fields' from the link above to expose it to the user.
i.e. it does not resolve domain names, but allows you to select one of the matching boards that it would find through discovery process

OK.

> If terminal detection does not work with Win7 and always prints that 'interactive mode' error, I would open an issue at arduino/arduino-cli

It now works, I do not know what happened before.

However, terminal detection always does not work if arduino-cli is executed under Cygwin/MSys console in Win 7 (I don't know about other Win version), but I'am not sure if an issues should be opened in this case because running arduino-cli under Cygwin/MSys is really not common (and not recommended?).

mcspr, 2023-07-28:
> However, terminal detection always does not work if arduino-cli is executed under Cygwin/MSys console in Win 7 (I don't know about other Win version), but I'am not sure if an issues should be opened [...]

Win11 msys console from git installation works ok with a small test app, so idk
```
$ cat main.go
package main

import (
    "os"
    "fmt"
    term "golang.org/x/term"
)

func main() {
    fmt.Printf("%v\n", term.IsTerminal(int(os.Stdin.Fd())))
}

$ cat go.mod
module isterm

go 1.20

require (
        golang.org/x/sys v0.10.0 // indirect
        golang.org/x/term v0.10.0 // indirect
)

$ go build

$ ./isterm.exe
true
```

Closing here since the issue is not really related to our repo

closed 2023-07-28 by mcspr

GhyslainBruno, 2024-09-04:
> Hello,
> 
> OTA upload works with Arduino IDE 2.1.1 (it always ask for OTA password even if I do not use OTA password. but it still works well :-)
> 
> However it does not work using arduino-cli. I have downloaded the latest arduino-cli from https://github.com/arduino/arduino-cli/releases and then execute (after compile):
> 
> `arduino-cli upload --fqbn "esp8266:esp8266:nodemcuv2" --board-options "ip=lm6f" --protocol network --port "192.168.1.113" "OTA_Test"`
> 
> It outputs: `Uploading to specified board using network protocol requires the following info:` `Error getting user input: user input not supported in non interactive mode`
> 
> Note that if I use invalid IP (for example 192.168.1.200), it outputs: `Error getting port metadata: port not found: 192.168.1.200 network`
> 
> So at least it can detect if there is something at '192.168.1.113'.
> 
> Is this mean that Arduino ESP8266 core does not support OTA upload using arduino-cli? Or perhaps I missed something in the command line parameter?
> 
> The sketch used is a standard OTA sketch test such as https://randomnerdtutorials.com/esp8266-ota-updates-with-arduino-ide-over-the-air.

Hi all, 

Did anyone find a solution to upload code over OTA using arduino-cli without the need to specify password ? 

I'm actually building some sort of deploy pipeline and I'd love to be able to upload code automatically. 

For now, even if I don't use password in my sketch, the cli asks me for that, and just like using the Arduino IDE, if I type anything it works. 

I also tried to "trick" the cli, by doing this : 
`yes | arduino-cli upload mySketch.ino -b esp8266:esp8266:d1_mini -p IP_ADDRESS` 

But I had the same error as you @blue-wind-25 : 

`Error getting user input: user input not supported in non interactive mode`

If anyone knows how to achieve such a thing, he will be blessed (by me, but still 😁). 

Thank you all

blue-wind-25, 2024-09-04:
> Hi all,
>
> Did anyone find a solution to upload code over OTA using arduino-cli without the need to specify password ?
>
> I'm actually building some sort of deploy pipeline and I'd love to be able to [...]

Hello,

Try to turn off your firewall, it works in my Linux. After that you can add IP whitelist maybe? Also, it may also because of incompatible terminal/console, try to use a different one.

If in Windows, rebooting both the PC and targets worked for me, never found what cause the problem.