esp8266/Arduino #8976: Weird Behaviour with BearSSL on GPIO

shrynshjn, 2023-08-22:
I use GPIO 5 to trigger a relay, my code also connects with a secure MQTT connection for which I have to use the BearSSL certificate definitions. 
Without the BearSSL certificates, when the esp restarts it does not re-trigger the relay, and the GPIO maintains its' state between reboots, but when I use the bearSSL certificates the behavior changes, and GPIO 5 goes low on restart. Any explanation why and how to mitigate it? 
I am 100% sure it is because of the import, cause I removed components of my code until I could find the piece of code that causes this, and it was beacuse of the declarations.

`#include <WiFiClientSecure.h>
static const char AWS_CERT_CA[] PROGMEM = R"EOF(
aws root)EOF";
 
 
// Copy contents from XXXXXXXX-certificate.pem.crt here ▼
static const char AWS_CERT_CRT[] PROGMEM = R"KEY(
device cert)KEY";
 
 
// Copy contents from  XXXXXXXX-private.pem.key here ▼
static const char AWS_CERT_KEY[] PROGMEM = R"KEY(
device key )KEY";

// WiFiClientSecure net;
 
BearSSL::X509List cert(AWS_CERT_CA);
BearSSL::X509List client_crt(AWS_CERT_CRT);
BearSSL::PrivateKey key(AWS_CERT_KEY);
 
WiFiClientSecure espClient;
void setup() {
  pinMode(5, OUTPUT);
  delay(2000);
  digitalWrite(5, HIGH);
  delay(2000);
  ESP.restart();
 }

  void loop() {}`

Without the following three lines

` 
BearSSL::X509List cert(AWS_CERT_CA);
BearSSL::X509List client_crt(AWS_CERT_CRT);
BearSSL::PrivateKey key(AWS_CERT_KEY);
`

The LED on GPIO 5 does not flicker on restarts but adding these lines makes them flicker. Any solution or suggestion towards a solution for this problem will be really helpful.

shrynshjn, 2023-08-29:
What worked for me was to move the BearSSL declaration and usage after the pinMode setting of the GPIO. So I had to make the certificate and keys local variables and wrap the entire setting up of wifi, conneting to it, setting up and connecting to the AWS core in a single function. 

```
void setupWiFiAndConnectAWS() {
  BearSSL::X509List cert(AWS_CERT_CA);
  BearSSL::X509List client_crt(AWS_CERT_CRT);
  BearSSL::PrivateKey key(AWS_CERT_KEY);
  WiFi.persistent(false);
  DEBUG_MSG("[setupWiFi] Setting host name\n");
  WiFi.hostname(deviceId);
  DEBUG_MSG("[setupWiFi] Setting station mode\n");
  WiFi.mode(WIFI_STA);
  if (WiFi.getMode() & WIFI_AP) {
    WiFi.softAPdisconnect(true);
  }
  WiFi.persistent(false);
  DEBUG_MSG("[setupWiFi] Loading certificates\n");
  net.setTrustAnchors(&cert);
  net.setClientRSACert(&client_crt, &key);
  loadCredentials();
  connectWiFi();
  attachInputInterrupts(false);
  setupNTP();
  connectAWS();
}
```

after setting up the GPIO pins. 
Although this is a work around, i still don't understand the core problem, and would like to understand it, that is why I am keeping the issue open for now

d-a-v, 2023-11-04:
Nice that you found a workaround.
You can also try and override the default initializer at boot time:
[`extern "C" void resetPins ()`](https://github.com/esp8266/Arduino/blob/1662248b394740d82824f504a021d4dad3d314b6/cores/esp8266/core_esp8266_wiring_digital.cpp#L257)