esp8266/Arduino #8983: 802.11 packet reassembly

razgino11, 2023-09-04:
#### Platform

- Hardware:				ESP-12e
- Core Version:			Latest
- Development Env:		Platformio
- Operating System:	        Windows

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Size: 		        1MB
- lwip Variant:                 Higher Bandwidth
- Upload Using:		SERIAL
- Upload Speed:              115200

### Problem Description

I am attempting to capture a 802.11n packet stream from an iot device,
I was noticing im losing most of the packets, and after analyzing them i figured im losing all of the packets which are fragmented on the 802.11 layer (layer 2)

I have found information online on enabling layer 3 fragmanation and udp fragmanation, but nothing on the 802.11 fragmantation

I would love it if you can point me in the right direction

d-a-v, 2023-09-05:
Radio link layer details are out of reach from the API provided by the nonos-sdk firmware that we are using.
I assume you checked nonos-sdk and rtos-sdk documentation.
As nonos-sdk is now unsupported by espressif, you may try first with the [rtos-sdk](https://github.com/espressif/ESP8266_RTOS_SDK) before asking them. Platformio [can make use of it](https://docs.platformio.org/en/latest/frameworks/esp8266-rtos-sdk.html).

razgino11, 2023-09-05:
> Radio link layer details are out of reach from the API provided by the nonos-sdk firmware that we are using. I assume you checked nonos-sdk and rtos-sdk documentation. As nonos-sdk is now [...]

Thank you for the fast response!
Yes i have tried and wrote the app on both arduino (which uses the no os sdk) and with the rtos in cpp, both version have this problem :(

d-a-v, 2023-09-05:
Did you ask on their repository ?

razgino11, 2023-09-05:
> Did you ask on their repository ?

Hey, not yet, ill open an issue there as well, in the mean time any advice on how i may attempt to fix this myself? also checked on the esp32 and has the same issue

d-a-v, 2023-11-04:
The radio fw is closed-source. Unless being able to disassemble and debug their code, I can't think of anything helpful.