esp8266/Arduino #8986: new board definition

mr-sm1th, 2023-09-09:
new board definition for the boards.txt.py to support ESP-07 - didnt work out how to build it yet - so untested

-----------------------------------------------
    ( 'nodemcu7', {
        'name': 'NodeMCU 0.7 (ESP-07 Module)',
        'opts': {
            '.build.board': 'ESP8266_NODEMCU_ESP07',
            '.build.variant': 'nodemcu7',
            },
        'macro': [
            'resetmethod_nodemcu',
            'flashmode_qio',
            'flashfreq_40',
            '1M', '16M',
            'led216',
            ],
        'desc': [ 'LED is on GPIO2',
                  ],
    }),
----------------------------------------------------

d-a-v, 2023-09-11:
You must read [`tools/boards.txt.py`](https://arduino-esp8266.readthedocs.io/en/latest/faq/a05-board-generator.html#when-do-i-need-to-mess-with-it)

mr-sm1th, 2023-09-22:
thanks - i did,
got it to build with slight changes - still not 100% clear though - it's like something is being cached so changes arent always being mirrored in the boards.txt
very infuriating.

the biggest problem is a lot of people change the flash on the modules and the board definitions all have fixed size.
and the generic option just breaks most projects because something isnt defined - i think i²c.

d-a-v, 2023-11-04:
We tried to do our best with the board definition.
You can try and make a very generic board with
- flashmode as "dout" which is the most compatible.
- option "flashsize" set to experimental "Mapping defined by Hardware and Sketch" as default helps when flash chip size are unknown in advance.