esp8266/Arduino #8997: Dublicate variable in LittleFS.h file

swimm3r, 2023-09-28:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESESP8285 device]
- Core Version:			[7f2deb14a254d0d1b0903e0bb15a87815833f579]
- Development Env:		[Arduino IDE]
- Operating System:		[Ubuntu]

### Settings in IDE

- Module:  			[Wemos D1 mini r2]
- Flash Mode:			[qio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

In libraries/LittleFS/src/LittleFS.h file has a duplicate variable (info.maxOpenFiles) in 'bool info(FSInfo& info) override' structure:

    bool info(FSInfo& info) override {
        if (!_mounted) {
            return false;
        }
        **info.maxOpenFiles = _maxOpenFds;**
        info.blockSize = _blockSize;
        info.pageSize = _pageSize;
        **info.maxOpenFiles = _maxOpenFds;**
        info.maxPathLength = LFS_NAME_MAX;
        info.totalBytes = _size;
        info.usedBytes = _getUsedBlocks() * _blockSize;
        return true;
    }

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

### Debug Messages

earlephilhower, 2023-09-28:
Sometimes you want to be really, really sure a variable is set properly so you assign it twice!  😆 

Seriously, though, this is a straightforward fix. Would you like to make a PR since you found it?

swimm3r, 2023-09-29:
Yes. yes. It's my first found in this project. 🥲

closed 2023-09-29 by swimm3r