esp8266/Arduino #9012: WEP connection issue

Gitoffomalawn, 2023-11-01:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-8266 NodeMCU v3]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[Ubuntu]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[?]
- Flash Size: 		[4MB]
- lwip Variant:   [?]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description

I cannot get the ESP8266 module to connect to a WEP SSID. The sketch below is what I am using. The module will try connecting, and then give the following output:
```
WiFi connected
IP address: 
169.254.215.189
```

All subsequent attempts also yield a 169.154.X.X address. The module is also not visible in the DHCP leases overview of the AP.

Am I missing something here?

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

// Libraries
#include <ESP8266WiFi.h>

// WiFi network
const char* ssid     = "SSID_NAME";
const char* password = "PASSWORD";

void setup() {
  
  // Enable WEP
  WiFi.enableInsecureWEP();
  
  // Start serial
  Serial.begin(115200);
  delay(10);

  // Connecting to a WiFi network
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  
}
```

d-a-v, 2023-11-04:
You can try with other firmware versions:

- select esp8266 generic board in the tools menu
- "hidden menus" will appear, one allows you to select other firmware version.

These firmwares are closed-source and now unmaintained from espressif.

Gitoffomalawn, 2023-11-12:
Thanks for the reply, but I don't think this will solve the problem. I didn't specify this earlier (and probably should have), but it worked before on a different AP. 

It's weird, because the units do reach association with the AP, but seem to fail getting an IP address, instead defaulting to the 169.x.x.x mentioned before.

d-a-v, 2023-11-12:
There are lots of other issues in this repository about some router that are not working but others are.

Solutions usually are:
- try with another nonos firmware
- try avoiding 802.11n, or forcing 802.11b, or 802.11g programmatically on the esp
- fully erase your flash once
- change some router specific parameters

We have sometimes no solutions.