esp8266/Arduino #9013: FQBN: esp8266:esp8266:generic

alyaparan, 2023-11-01:
FQBN: esp8266:esp8266:generic
Using board 'generic' from platform in folder: C:\Users\alikp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2
Using core 'esp8266' from platform in folder: C:\Users\alikp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2

"C:\\Users\\alikp\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\python3\\3.7.2-post1/python3" -I "C:\\Users\\alikp\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.1.2/tools/mkbuildoptglobals.py" "C:\\Program Files\\Arduino IDE\\resources\\app\\lib\\backend\\resources" 10607 "C:\\Users\\alikp\\AppData\\Local\\Temp\\arduino\\sketches\\09D9037E17A77D643A195D118E3D3DA0" "C:\\Users\\alikp\\AppData\\Local\\Temp\\arduino\\sketches\\09D9037E17A77D643A195D118E3D3DA0/core/build.opt" "C:\\Users\\alikp\\OneDrive\\Документы\\duino-coin-master\\ESP8266_Code/ESP8266_Code.ino.globals.h" "C:\\Users\\alikp\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.1.2\\cores\\esp8266/CommonHFile.h"
default_encoding:       cp1252
Assume aggressive 'core.a' caching enabled.
Note: optional global include file 'C:\Users\alikp\OneDrive\\u0414\u043e\u043a\u0443\u043c\u0435\u043d\u0442\u044b\duino-coin-master\ESP8266_Code\ESP8266_Code.ino.globals.h' does not exist.
  Read more at https://arduino-esp8266.readthedocs.io/en/latest/faq/a06-global-build-options.html

*** Traceback (most recent call last):
  File "C:\Users\alikp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/mkbuildoptglobals.py", line 824, in <module>
    rc = main()
  File "C:\Users\alikp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/mkbuildoptglobals.py", line 819, in main
    handle_error(0)   # commit print buffer
  File "C:\Users\alikp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/mkbuildoptglobals.py", line 274, in handle_error
    print(msg_print_buf, file=fd, end='', flush=True)
  File "D:\obj\Windows-Release\37win32_Release\msi_python\zip_win32\cp1252.py", line 19, in encode
UnicodeEncodeError: 'charmap' codec can't encode characters in position 137-145: character maps to <undefined>

exit status 1

Compilation error: exit status 1

d-a-v, 2023-11-04:
If you're able to locate the file `mkbuildoptglobals.py`, can you try and change
https://github.com/esp8266/Arduino/blob/1662248b394740d82824f504a021d4dad3d314b6/tools/mkbuildoptglobals.py#L676

to
```py
    default_encoding = 'utf-8'
```
and report if this fixes your issue ?

mcspr, 2023-11-04:
Note that encoding needs to happen for `sys.stdout`, above only sets it for file I/O. Line prints full path, `basename(fpath)` would help (or just removing the line(s) mentioning user path completely)
https://github.com/esp8266/Arduino/blob/1662248b394740d82824f504a021d4dad3d314b6/tools/mkbuildoptglobals.py#L816

Nuclear opt is to set `-X utf8`, but I still don't think it does the right thing here? Would work-around the above, but pretty missable otherwise

alyaparan, 2023-11-09:
> If you're able to locate the file `mkbuildoptglobals.py`, can you try and change
> 
> https://github.com/esp8266/Arduino/blob/1662248b394740d82824f504a021d4dad3d314b6/tools/mkbuildoptglobals.py#L676
> 
> to
> 
> ```python
>     default_encoding = 'utf-8'
> ```
> 
> and report if this fixes your issue ?

It's not worked, same issue with utf-8

mhightower83, 2023-11-12:
To get more insight, try adding this line to your `platform.local.txt` file and report the build results.
```
mkbuildoptglobals.extra_flags=--debug
```

Which version of the Arduino IDE are you using?