esp8266/Arduino #9019: Cannot Connect to Android Phones with version> 10

EmbeddedAyoub, 2023-11-05:
Hello,

Hardware:
Board: ESP8266-01
Core Installation version: 3.2.0
IDE name: Arduino IDE 
Flash Frequency: 40Mhz
Upload Speed: 115200
Computer OS: Windows 11

Description:
I made an Android Application which tries to send Credentials to the ESP-01 via HTTP Get request. 
I saw 2 different behaviors:
1- Devices with Android version < 10 : The device can connect to the ESP-01 and send the credentials, but it gets disconnected some times.
2- Devices with Android version >= 10 : The device can connect to the ESP-01 but cannot communicate with it. 

I Saw a similar issue in Espressif Community and they confirmed the Ampdu RX/TX to be the issue.

Do you confirm? Otherwise, what are your advices?

Thanks,
Ayoub

Harrisbaig7, 2023-11-05:
Hello Ayoub,

Based on the information you provided, it seems like you are facing communication issues between your Android application and the ESP-01 module, particularly on devices with Android version 10 and above. The problem may be related to Ampdu (A-MPDU) RX/TX handling, as suggested in the Espressif Community.

A-MPDU (Aggregated MAC Protocol Data Unit) is a feature in Wi-Fi technology that allows multiple data frames to be combined into a single transmission, improving efficiency. However, there might be compatibility issues with certain devices or firmware versions, leading to communication problems.

Here are a few suggestions to troubleshoot and potentially resolve the issue:

Firmware Update: Ensure that you are using the latest firmware version for your ESP-01 module. Espressif often releases updates that address compatibility and stability issues. Check the Espressif website or community forums for the latest firmware version and update if necessary.

Check Android App Permissions: Make sure your Android application has the necessary permissions to access the network. Check the AndroidManifest.xml file in your Android project to ensure the necessary permissions, such as internet access, are declared.

Verify HTTP Requests: Double-check the HTTP requests sent from your Android application. Ensure that the request headers and parameters are correctly formatted. You might also consider using a tool like Wireshark to inspect the network traffic and see if the requests are being formed correctly.

Test with a Different Library: If you are using a specific library for handling HTTP requests on the ESP-01, try using a different library or approach. There are various libraries available for ESP8266 that might behave differently in certain scenarios. Experimenting with different libraries can sometimes resolve compatibility issues.

Consider HTTPS: If security permits, consider switching to HTTPS instead of HTTP for communication between the Android app and ESP-01. Some devices and Android versions might have stricter security policies for unencrypted HTTP connections.

Debugging Tools: Use debugging tools and logs on both the Android app and ESP-01 to trace the issue. Print debug messages in your ESP-01 code to understand if the requests are being received correctly and how the module is handling them.

Community Support: Continue monitoring the Espressif Community forums or relevant online communities. Sometimes, community members or developers might find workarounds or solutions to specific issues.

d-a-v, 2023-11-06:
If the issue is A-MPDU (appeared in 802.11n ?), it should disappear when forcing the esp in mode 802.11g.
```
WiFi.setPhyMode(WIFI_PHY_MODE_11G);
```

EmbeddedAyoub, 2023-11-06:
> If the issue is A-MPDU (appeared in 802.11n ?), it should disappear when forcing the esp in mode 802.11g.
> 
> ```
> WiFi.setPhyMode(WIFI_PHY_MODE_11G);
> ```

I tried this solution but didn't work!