esp8266/Arduino #9027: Flash-size agnostic mode, problem with spi_flash_* functions

ACE1046, 2023-11-12:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-07]
- Core Version:			[Core:3.1.2]
- Development Env:		[Platformio]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[dout]
- Flash Size: 		[1MB]
- Upload Using:		[SERIAL/OTA]

### Problem Description

I'm trying to make "Flash-size agnostic build". Firmware still depends on selected ldscript flash size in SDK functions spi_flash_*. FS not working if firmware was build with ldscript flash size less than actual.

Steps to reproduce:
1. Add to sketch: FLASH_MAP_SETUP_CONFIG(FLASH_MAP_OTA)
2. Set -DFLASH_MAP_SUPPORT=1
3. Build firmware for 1M board (i.e. ESP07). Or set board_build.ldscript = eagle.flash.1m128.ld (or anything less than 4M). Or eagle.flash.auto.ld (if boards default flash size is 1M)
4. Upload to any 4M board
5. Everything works (OTA/Arduino updates, except for FS. Can't mount, can't format.

```cpp
DEBUGV("Chip size %d\n", flashchip->chip_size);
DEBUGV("Real Chip size %d\n", ESP.getFlashChipRealSize());
```

```
Chip size 1048576
Real Chip size 4194304
```

Problem: incorrect value in flashchip->chip_size is used by spi_flash_* functions for bounds checking. Flash read above 1M fails.

Possible fix:
flashchip->chip_size = ESP.getFlashChipRealSize();
(somewhere in the code)

d-a-v, 2023-11-12:
> Build firmware for 1M board (i.e. ESP07). Or set board_build.ldscript = eagle.flash.1m128.ld (or anything less than 4M). Or eagle.flash.auto.ld (if boards default flash size is 1M)

When building with `FLASH_MAP_SUPPORT`, only `eagle.flash.auto.ld` is supposed to be used.

The runtime [selected configuration is based on](https://github.com/esp8266/Arduino/blob/7fc2caa72fc2280f3c8a25d2e25dfa85b1dc6c3d/cores/esp8266/flash_hal.h#L50) the same code as in [`ESP.getFlashChipRealSize()`](https://github.com/esp8266/Arduino/blob/7fc2caa72fc2280f3c8a25d2e25dfa85b1dc6c3d/cores/esp8266/Esp.cpp#L292), and an error message should be emitted if the flash size entry is not found in the table. Then without error, showed values should be equal.

> FLASH_MAP_SETUP_CONFIG(FLASH_MAP_OTA)

This symbol [does not exist by default](https://github.com/esp8266/Arduino/blob/7fc2caa72fc2280f3c8a25d2e25dfa85b1dc6c3d/cores/esp8266/FlashMap.h#L28-L56) so you provided one: can you show its content ?
If `FLASH_MAP_SETUP_CONFIG()` is not called, `FLASH_MAP_OTA_FS` is used by default and it seems that it is the one you need. Did you try without this line ?

Can you also tell
- what is your build environment
- the nonos firmware used (2.2.1 or 3.0.5)

ACE1046, 2023-11-12:
It doesn't matter if I use eagle.flash.auto.ld, use FLASH_MAP_OTA_FS or remove FLASH_MAP_SETUP_CONFIG at all - I get FS not working. Checked.
But if I change selected board from esp07 (1M) to esp12e (4M) it works.

System configuration, ESP.getFlashChipRealSize(), ESP.getFlashChipSize() are correct.
But [flashchip->chip_size](https://github.com/esp8266/Arduino/blob/7fc2caa72fc2280f3c8a25d2e25dfa85b1dc6c3d/tools/sdk/include/spi_flash.h#L42) is not.

My build environment is VSCode  with Platformio Core 6.1.11 Home 3.4.4

I used SDK:2.2.1.
With SDK 3.0.5 I get this debug log at start:

```
System partition table registration failed!
boot not set
ota1 not set
ota2 not set
--- The partition table size is larger than flash size 0x100000 ---
please check partition type 6 addr:3fd000 len:3000
Reference info:
  fn(...ex].flash_size_kb) 0x00400000  4194304
  fn(spi_flash_get_id())   0x00400000  4194304
  bin_chip_size            0x00100000  1048576
  PHY_DATA                 0x003FB000
  RF_CAL                   0x003FC000
  SYSTEM_PARAMETER         0x003FD000
  EEPROM_start             0x405FB000
  FS_start                 0x40400000
  FS_end                   0x405FA000
  FS_page                  0x00000100
  FS_block                 0x00002000
```

So, SDK still use incorrect chip size from bin file. And this value is used for bounds checking if flash read/write functions. At least in 2.2.1 SDK. Yes, I can use board with bigger flash size to compile bin, but in this case attempts to read/write flash beyond end will not be catched by SDK on boards with less flash.

So, long story short
1. Value of [flashchip->chip_size](https://github.com/esp8266/Arduino/blob/master/tools/sdk/include/spi_flash.h#42) based on flash size in bin file
2. Same value used for bounds checking in SDK's functions. You can check it. Set flashchip->chip_size = ESP.getFlashChipRealSize()/4; in setup() before initializing FS and it will fail to initialize or format. Checked with SDK 3.0.5.

d-a-v, 2023-11-12:
I need to reproduce.
I will select a 1MB board, build a FS sketch with this "flash size agnostic" mode and check on a 4MB board:
- working FS
- `flashchip->chip_size`
- `ESP.getFlashChipRealSize()`