esp8266/Arduino #9029: Software crashes with an User exception

hmueller01, 2023-11-12:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[framework-arduinoespressif8266 @ 3.2.0-dev+sha.fb8d6d6]
- Development Env:		[Platformio]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

Software crashes with an User exception after connecting to WiFi and MQTT TLS 
```
Reset reason   : External System
connected with @ XXX, channel 1
dhcp client start...
ip:192.168.X.X,mask:255.255.255.0,gw:192.168.X.X
Station connected: 192.168.X.X
otaSetup: Setting up ArduinoOTA
timeSetup: Setting time using SNTP
timeSetup: Current time: Sun Nov 12 09:47:38 2023
mqttSetup: MQTT client id: ESP-XXX
mqttSetup: Attempting connection to server XXX.XXX:8883

No poison after block at: 0x3fff2680, actual data: 0x65 0x0 0xa5 0xa5

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Panic umm_malloc.cpp:751 umm_malloc_core

>>>stack>>>

ctx: sys
sp: 3fffecf0 end: 3fffffb0 offset: 0010
...
```

But it is not always at the same point in software. Sometimes it also starts the HTTP server and transmits several packages before the crash, like this:

```
Reset reason   : Software/System restart
connected with @ XXX, channel 1
dhcp client start...
ip:192.168.X.X,mask:255.255.255.0,gw:192.168.X.X
Station connected: 192.168.X.X
otaSetup: Setting up ArduinoOTA
timeSetup: Setting time using SNTP
timeSetup: Current time: Sun Nov 12 09:47:38 2023
mqttSetup: MQTT client id: ESP-XXX
mqttSetup: Attempting connection to server XXX.XXX:8883
[... 176 lines cut ...]
0x3ffef48c: _Pri_3_Stack at /Users/hmueller/src/PlatformIO/Projects/garage-platformio/src/garage-arduino.cpp:96
0x40212fd1: loop at /Users/hmueller/src/PlatformIO/Projects/garage-platformio/src/garage-arduino.cpp:957
0x40208dd0: std::_Function_base::_Function_base() at /Users/hmueller/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/std_function.h:240
(inlined by) std::function<void ()>::function(std::function<void ()>&&) at /Users/hmueller/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/std_function.h:357
(inlined by) Ticker::attach_ms_scheduled(unsigned int, std::function<void ()>)::{lambda()#1}::function({lambda()#1}&&) at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/libraries/Ticker/src/Ticker.h:81
(inlined by) std::_Function_base::_Base_manager<Ticker::attach_ms_scheduled(unsigned int, std::function<void ()>)::{lambda()#1}>::_M_init_functor(std::_Any_data&, {lambda()#1}&&, std::integral_constant<bool, false>) at /Users/hmueller/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/std_function.h:237
(inlined by) std::_Function_base::_Base_manager<Ticker::attach_ms_scheduled(unsigned int, std::function<void ()>)::{lambda()#1}>::_M_init_functor(std::_Any_data&, {lambda()#1}&&) at /Users/hmueller/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/std_function.h:208
(inlined by) std::function<void ()>::function<Ticker::attach_ms_scheduled(unsigned int, std::function<void ()>)::{lambda()#1}, void, void>(Ticker::attach_ms_scheduled(unsigned int, std::function<void ()>)::{lambda()#1}) at /Users/hmueller/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/std_function.h:609
0x40105f2d: ets_timer_handler_dsr at /home/wchen/gwen/nonos/app/main/ets_timer.c:335
0x3ffef650: _Pri_3_Stack at /Users/hmueller/src/PlatformIO/Projects/garage-platformio/src/garage-arduino.cpp:95
0x402097c5: esp8266webserver::ESP8266WebServerTemplate<WiFiServer>::begin() at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/libraries/ESP8266WebServer/src/ESP8266WebServer-impl.h:82
0x3ffefc78: _Pri_3_Stack at nmi-vector.o:?
0x3ffefc78: _Pri_3_Stack at nmi-vector.o:?
0x3ffef380: _Pri_3_Stack at nmi-vector.o:?
0x40206818: debouncePinTimer_Cb() at /Users/hmueller/src/PlatformIO/Projects/garage-platformio/src/garage-arduino.cpp:388
0x4022e294: std::_Function_handler<void (), void (*)()>::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at /Users/hmueller/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/std_function.h:271
0x4022e280: std::_Function_handler<void (), void (*)()>::_M_invoke(std::_Any_data const&) at /Users/hmueller/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/std_function.h:289
0x4022e28a: std::_Function_handler<void (), void (*)()>::_M_invoke(std::_Any_data const&) at /Users/hmueller/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/std_function.h:293
0x401000e5: std::function<void ()>::operator()() const at /Users/hmueller/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/std_function.h:623
0x40221911: run_scheduled_functions() at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/cores/esp8266/Schedule.cpp:173
0x40100658: ets_post at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/cores/esp8266/core_esp8266_main.cpp:243
0x40100679: esp_schedule at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/cores/esp8266/core_esp8266_main.cpp:136
0x40224e6a: loop_wrapper() at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/cores/esp8266/core_esp8266_main.cpp:269

```

hmueller01, 2024-01-21:
After building another project with similar code and crashes I found out that this was the problem:
```
  configTime(TZ_Europe_Berlin, PSTR("de.pool.ntp.org"), PSTR("europe.pool.ntp.org"), PSTR("time.nist.gov"));
``` 
Changing this to 
```
  configTime(TZ_Europe_Berlin, "de.pool.ntp.org", "europe.pool.ntp.org", "time.nist.gov");
``` 
did the trick.

But it didn't fully help on this project. There I still have a crash after a `zalloc`/ `malloc` right here:

```
void mqttCallback(char *topic, byte *payload, unsigned int length) {
  bool topic_handled = false;
  char topic_pub[100];

  DEBUGF_P("%s" LF, __func__);
  DEBUGF_P("getFreeHeap: %d, getHeapFragmentation: %d, getMaxFreeBlockSize: %d" LF, ESP.getFreeHeap(), ESP.getHeapFragmentation(), ESP.getMaxFreeBlockSize());
  DEBUGF_P("length: %d" LF, length);

  // copy payload to a char string
  //char *data = (char *)zalloc(length + 1);
  char *data = (char *)malloc(length + 1);
  DEBUG_P("after malloc, before memcpy");
  memcpy(data, payload, length);
```
It doesn't reach the `DEBUG_P("after malloc, before memcpy");` any more.
But how is this possible? `length` is only 2 bytes and the getMaxFreeBlockSize is 12080!

```
timeSetup: Setting time using SNTP
timeSetup: Current time: Sun Jan 21 20:08:31 2024
mqttSetup: MQTT client id: ESP-0015AADA
mqttSetup: HomA system id: 123456-garage
mqttSetup: Attempting connection to server mqtt:8883
mqttSetup: Sucsessfully connected.
Stop multiResetDetecting
mqttConnect: connected using PubSubClient
cisternSwitch: switch cistern OFF
mqttConnect: Current time: Sun Jan 21 20:08:32 2024
mqttCallback
getFreeHeap: 13712, getHeapFragmentation: 12, getMaxFreeBlockSize: 12080
length: 2
No poison after block at: 0x3fff2838, actual data: 0x65 0x0 0xa5 0xa5
User exception (panic/abort/assert)
To make this dump useful, DECODE IT - https://tinyurl.com/8266dcdr
Panic umm_malloc.cpp:751 umm_malloc_core
sp: 3ffffcc0 end: 3fffffc0 offset: 0010
To make this dump useful, DECODE IT - https://tinyurl.com/8266dcdr

ctx: cont
0x40101118: umm_malloc_core at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/umm_malloc/umm_malloc.cpp:753
0x3ffefae8: _Pri_3_Stack at nmi-vector.o:?
0x4010145a: umm_malloc at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/umm_malloc/umm_malloc.cpp:912
0x401014c4: umm_poison_malloc at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/umm_malloc/umm_poison.c:173
0x40263737: umm_max_block_size at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/umm_malloc/umm_info.c:203
0x401008f8: malloc at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/heap.cpp:250
0x3ffefae8: _Pri_3_Stack at nmi-vector.o:?
0x40208da5: mqttCallback(char*, unsigned char*, unsigned int) at /Users/hmueller/src/PlatformIO/Projects/garage-pio/src/garage-arduino.cpp:455
0x4021244b: BearSSL::WiFiClientSecureCtx::_run_until(unsigned int, bool) at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp:497
0x40211d28: BearSSL::WiFiClientSecureCtx::_engineConnected() at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp:261 (discriminator 5)
0x40212c38: BearSSL::WiFiClientSecureCtx::read(unsigned char*, unsigned int) at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp:383 (discriminator 4)
0x3ffef48c: _Pri_3_Stack at /Users/hmueller/src/PlatformIO/Projects/garage-pio/src/garage-arduino.cpp:99
0x3ffef48c: _Pri_3_Stack at /Users/hmueller/src/PlatformIO/Projects/garage-pio/src/garage-arduino.cpp:99
0x402295e8: BearSSL::WiFiClientSecureCtx::read() at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp:414
0x3ffef48c: _Pri_3_Stack at /Users/hmueller/src/PlatformIO/Projects/garage-pio/src/garage-arduino.cpp:99
0x40212bbc: BearSSL::WiFiClientSecureCtx::available() at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp:443
0x402289d2: BearSSL::WiFiClientSecure::read() at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.h:279
0x3ffef48c: _Pri_3_Stack at /Users/hmueller/src/PlatformIO/Projects/garage-pio/src/garage-arduino.cpp:99
[... 22 lines cut ...]
(inlined by) void esp_delay<ClientContext::connect(ip4_addr*, unsigned short)::{lambda()#1}>(unsigned int, ClientContext::connect(ip4_addr*, unsigned short)::{lambda()#1}&&) at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/coredecls.h:78
(inlined by) ClientContext::connect(ip4_addr*, unsigned short) at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/include/ClientContext.h:147
(inlined by) WiFiClient::connect(IPAddress, unsigned short) at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiClient.cpp:164
0x40281178: etharp_output at ??:?
0x3ffe88e4: ?? at /Users/hmueller/src/PlatformIO/Projects/garage-pio/include/dynamicParams.h:52
0x3ffefcf8: _Pri_3_Stack at nmi-vector.o:?
0x3ffe88e4: ?? at /Users/hmueller/src/PlatformIO/Projects/garage-pio/include/dynamicParams.h:52
[... cut ...]
0x40209ca8: setup at /Users/hmueller/src/PlatformIO/Projects/garage-pio/src/garage-arduino.cpp:922
0x40100658: ets_post at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_main.cpp:243
0x3ffefcf8: _Pri_3_Stack at nmi-vector.o:?
0x4021efb2: loop_wrapper() at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_main.cpp:263
0x401005a9: cont_wrapper at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/cont.S:81
```

mcspr, 2024-01-21:
> No poison after block at: 0x3fff2838, actual data: 0x65 0x0 0xa5 0xa5
> Panic umm_malloc.cpp:751 umm_malloc_core

Memory is corrupted, allocator is configured to panic() when that happens

hmueller01, 2024-01-21:
Tx for the quick response. Is there a way to figure out the location in the trace? I currently have no clue where to look. I already removed almost everything except the WiFi configuration  and the mqtt stuff (over TLS using BearSSL on top of PubSubClient). This is really a pain ...

d-a-v, 2024-01-21:
> It doesn't reach the `DEBUG_P("after malloc, before memcpy");` any more.

Adding `Serial.flush()` after every `printf` might help to narrow the issue.
You can also try to use [gdb](https://arduino-esp8266.readthedocs.io/en/latest/gdb.html).

mcspr, 2024-01-21:
> 0x3ffefae8: _Pri_3_Stack at nmi-vector.o:?

...or something to do with SDK3? I don't see this one in the default

hmueller01, 2024-01-22:
> > It doesn't reach the `DEBUG_P("after malloc, before memcpy");` any more.
> 
> Adding `Serial.flush()` after every `printf` might help to narrow the issue. You can also try to use [gdb](https://arduino-esp8266.readthedocs.io/en/latest/gdb.html).

Thanks @d-a-v . It crashes exactly at the malloc function (added the flush after the malloc printf / DEBUG_P). Thanks for the gdb hint. Wasn't aware that this is possible with ESP8266 ... but if memory is corrupted I doubt gdb will be of much help. :-(

hmueller01, 2024-01-22:
> nmi-vector.o:?

I use the Ticker, but only once_scheduled() or attach_ms_scheduled() which should execute the callback in the loop context. I will remove all that code and will see if anything changes.

BTW: The crash happens in the PubSubClient MQTT callback message/topic received. But I use this code in other projects in the same way and do not see a crash there.

hmueller01, 2024-01-22:
Removing all the Ticker code didn't help at all. Dump output looks almost the same.
```
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

mqttConnect: connected using PubSubClient
mqttConnect: Current time: Mon Jan 22 20:29:47 2024
mqttCallback
getFreeHeap: 14872, getHeapFragmentation: 2, getMaxFreeBlockSize: 14712
length: 2
No poison after block at: 0x3fff2410, actual data: 0x65 0x0 0xa5 0xa5
User exception (panic/abort/assert)
To make this dump useful, DECODE IT - https://tinyurl.com/8266dcdr
Panic umm_malloc.cpp:751 umm_malloc_core
sp: 3ffffcc0 end: 3fffffc0 offset: 0010
To make this dump useful, DECODE IT - https://tinyurl.com/8266dcdr

ctx: cont
0x40100ea0: umm_malloc_core at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/umm_malloc/umm_malloc.cpp:753
0x401011e2: umm_malloc at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/umm_malloc/umm_malloc.cpp:912
0x3ffef898: _Pri_3_Stack at nmi-vector.o:?
0x4010124c: umm_poison_malloc at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/umm_malloc/umm_poison.c:173
0x3ffef898: _Pri_3_Stack at nmi-vector.o:?
0x40261da3: umm_max_block_size at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/umm_malloc/umm_info.c:203
0x40100680: malloc at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/heap.cpp:250
0x3ffef898: _Pri_3_Stack at nmi-vector.o:?
0x4020c518: mqttCallback(char*, unsigned char*, unsigned int) at /Users/hmueller/src/PlatformIO/Projects/garage-pio/src/garage-arduino.cpp:451
0x40211e6b: BearSSL::WiFiClientSecureCtx::_run_until(unsigned int, bool) at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp:497
0x40211748: BearSSL::WiFiClientSecureCtx::_engineConnected() at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp:261 (discriminator 5)
0x40212658: BearSSL::WiFiClientSecureCtx::read(unsigned char*, unsigned int) at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp:383 (discriminator 4)
0x3ffef244: _Pri_3_Stack at /Users/hmueller/src/PlatformIO/Projects/garage-pio/src/garage-arduino.cpp:90
0x3ffef244: _Pri_3_Stack at /Users/hmueller/src/PlatformIO/Projects/garage-pio/src/garage-arduino.cpp:90
0x40227f20: BearSSL::WiFiClientSecureCtx::read() at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp:414
0x3ffef244: _Pri_3_Stack at /Users/hmueller/src/PlatformIO/Projects/garage-pio/src/garage-arduino.cpp:90
0x402125dc: BearSSL::WiFiClientSecureCtx::available() at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp:443
0x402273ce: BearSSL::WiFiClientSecure::read() at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.h:279
0x3ffef244: _Pri_3_Stack at /Users/hmueller/src/PlatformIO/Projects/garage-pio/src/garage-arduino.cpp:90
0x4020d3e1: PubSubClient::readByte(unsigned char*) at /Users/hmueller/src/PlatformIO/Projects/garage-pio/.pio/libdeps/upstream_develop_flash_serial_debug/PubSubClient/src/PubSubClient.cpp:297
0x3ffef244: _Pri_3_Stack at /Users/hmueller/src/PlatformIO/Projects/garage-pio/src/garage-arduino.cpp:90
0x3ffef244: _Pri_3_Stack at /Users/hmueller/src/PlatformIO/Projects/garage-pio/src/garage-arduino.cpp:90
0x4022784c: std::_Function_handler<void (char*, unsigned char*, unsigned int), void (*)(char*, unsigned char*, unsigned int)>::_M_invoke(std::_Any_data const&, char*&&, unsigned char*&&, unsigned int&&) at /Users/hmueller/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/std_function.h:293
0x40100160: std::function<void (char*, unsigned char*, unsigned int)>::operator()(char*, unsigned char*, unsigned int) const at /Users/hmueller/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/std_function.h:623
0x3ffefa88: _Pri_3_Stack at nmi-vector.o:?
0x3ffef244: _Pri_3_Stack at /Users/hmueller/src/PlatformIO/Projects/garage-pio/src/garage-arduino.cpp:90
0x4020de0c: PubSubClient::loop() at /Users/hmueller/src/PlatformIO/Projects/garage-pio/.pio/libdeps/upstream_develop_flash_serial_debug/PubSubClient/src/PubSubClient.cpp:416
0x3fff0228: _Pri_3_Stack at nmi-vector.o:?
0x3ffef718: _Pri_3_Stack at nmi-vector.o:?
0x402223fc: esp8266::MDNSImplementation::MDNSResponder::_process(bool) at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266mDNS/src/LEAmDNS_Control.cpp:88 (discriminator 1)
0x3ffef718: _Pri_3_Stack at nmi-vector.o:?
0x402208c8: esp8266::MDNSImplementation::MDNSResponder::update() at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266mDNS/src/LEAmDNS.cpp:1248
0x3ffef2c0: _Pri_3_Stack at /Users/hmueller/src/PlatformIO/Projects/garage-pio/src/garage-arduino.cpp:87
0x3ffefa88: _Pri_3_Stack at nmi-vector.o:?
0x3ffef2c0: _Pri_3_Stack at /Users/hmueller/src/PlatformIO/Projects/garage-pio/src/garage-arduino.cpp:87
0x4020cc18: loop at /Users/hmueller/src/PlatformIO/Projects/garage-pio/src/garage-arduino.cpp:944
0x3ffef898: _Pri_3_Stack at nmi-vector.o:?
[... cut ...]
0x4021e0e2: loop_wrapper() at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_main.cpp:263
0x4010171d: cont_wrapper at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/cont.S:81
```
garage-arduino.cpp _Pri_3_Stack lines mentioned above:
```
86: ESP8266WebServer m_webserver(80);
87: ESP_WiFiManager_Lite m_wifi_manager;
88: WiFiEventHandler m_station_got_ip_handler;
89: BearSSL::WiFiClientSecure m_esp_client;
90: PubSubClient m_mqtt_client(m_esp_client);
```

mcspr, 2024-01-22:
> garage-arduino.cpp _Pri_3_Stack lines mentioned above:

...I meant that this shows up b/c of enabled SDK3, apparently. SDK2 190307 does not have this symbol in .a, SDK305 does. platformio.ini options weren't mentioned above

hmueller01, 2024-01-22:
@mcspr You should have access to
https://github.com/arduino-hmueller01/garage-pio/blob/main/platformio.ini

Yes, I use SDK305. I'll see if it compiles with SDK2 and report.

mcspr, 2024-01-26:
> https://github.com/arduino-hmueller01/garage-pio/blob/main/platformio.ini

I must be doing something wrong, it is still 404 for me :shrug:

hmueller01, 2024-01-26:
Oh sorry. I added additional access for you. Hope it works now.

In the meantime I tried SDK 2.x and code crashed even earlier. Maybe I have to remove more code to narrow down the issue. It is really strange that some projects of mine work and others don't. Even that the code is almost the same now.

```
connected with xxx, channel 1
dhcp client start...
wifi evt: 0
ip:192.168.x.x,mask:255.255.255.0,gw:192.168.x.x
wifi evt: 3
Station connected: 192.168.x.x
[WIFIM] Connected:
[WIFIM]   SSID: xxx
[WIFIM]   IP:   192.168.x.x
[WIFIM]   MAC:  48:5D:35:7D:74:41
[WIFIM]   CH:   1
[WIFIM]   RSSI: -66
[WML] WiFi connected after num retries: 0
[WML] SSID=xxx,RSSI=-66
[WML] Channel=1,IP=192.168.x.x
[WML] bg: WiFi OK.
otaSetup: Setting up ArduinoOTA
timeSetup: Setting time using SNTP
timeSetup: Current time: Tue Jan 23 19:22:25 2024
mqttSetup: MQTT client id: ESP-0015AADA
mqttSetup: HomA system id: 123456-garage
mqttSetup: Attempting connection to server y:8883
[hostByName] request IP for: y
No poison after block at: 0x3fff2ef8, actual data: 0x65 0x0 0xa5 0xa5
User exception (panic/abort/assert)
To make this dump useful, DECODE IT - https://tinyurl.com/8266dcdr
Abort called
sp: 3ffffe60 end: 3fffffc0 offset: 0010
To make this dump useful, DECODE IT - https://tinyurl.com/8266dcdr

ctx: cont
0x4020e4ec: make_unique<(anonymous namespace)::_dns_found_result, (anonymous namespace)::_dns_found_result> at /Users/hmueller/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/unique_ptr.h:962
(inlined by) hostByNameImpl at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/libraries/ESP8266WiFi/src/ESP8266WiFiGeneric.cpp:625
0x401006b0: malloc at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/cores/esp8266/heap.cpp:246
0x4021e557: operator new(unsigned int) at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/cores/esp8266/abi.cpp:41
0x3ffe876c: ?? at /Users/hmueller/src/PlatformIO/Projects/garage-pio/include/dynamicParams.h:52
0x4020e4ec: make_unique<(anonymous namespace)::_dns_found_result, (anonymous namespace)::_dns_found_result> at /Users/hmueller/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/unique_ptr.h:962
(inlined by) hostByNameImpl at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/libraries/ESP8266WiFi/src/ESP8266WiFiGeneric.cpp:625
0x4022a0fc: base64_encode_value at ??:?
0x40211ddc: BearSSL::WiFiClientSecureCtx::setFingerprint(char const*) at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp:659
0x3ffe876c: ?? at /Users/hmueller/src/PlatformIO/Projects/garage-pio/include/dynamicParams.h:52
0x4020e7dc: ESP8266WiFiGenericClass::hostByName(char const*, IPAddress&) at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/libraries/ESP8266WiFi/src/ESP8266WiFiGeneric.cpp:685
0x402129c5: BearSSL::WiFiClientSecureCtx::connect(char const*, unsigned short) at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp:220
0x4022a0fc: base64_encode_value at ??:?
0x4027f420: etharp_output at ??:?
0x3ffef010: ?? at /Users/hmueller/src/PlatformIO/Projects/garage-pio/src/garage-arduino.cpp:89
0x3ffe876c: ?? at /Users/hmueller/src/PlatformIO/Projects/garage-pio/include/dynamicParams.h:52
0x40207f05: mqttSetup() at /Users/hmueller/src/PlatformIO/Projects/garage-pio/src/garage-arduino.cpp:423
0x4022841c: std::_Function_handler<void (), otaSetup()::{lambda()#1}>::_M_manager(std::_Any_data&, std::_Function_handler<void (), otaSetup()::{lambda()#1}> const&, std::_Manager_operation) at /Users/hmueller/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/std_function.h:271
0x4020664c: std::_Function_handler<void (), otaSetup()::{lambda()#1}>::_M_invoke(std::_Any_data const&) at /Users/hmueller/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/std_function.h:289
0x40209450: setup at /Users/hmueller/src/PlatformIO/Projects/garage-pio/src/garage-arduino.cpp:927
0x4021e987: loop_wrapper() at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/cores/esp8266/core_esp8266_main.cpp:260
0x40101701: cont_wrapper at /Users/hmueller/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/cores/esp8266/cont.S:81
```

mcspr, 2024-01-27:
Watch out for unchecked (s)printf

```diff
diff --git a/src/garage-arduino.cpp b/src/garage-arduino.cpp
index 85602ae..3088409 100644
--- a/src/garage-arduino.cpp
+++ b/src/garage-arduino.cpp
@@ -419,7 +420,7 @@ void mqttSetup() {
   INFOF_P("%s: MQTT client id: %s" LF, __func__, host_name);

   SAFE_FREE(m_homa_id);
-  m_homa_id = (char *)malloc(strlen_P(PSTR(HOMA_SYSTEM_ID) + 1));
+  m_homa_id = (char *)malloc(strlen_P(PSTR(HOMA_SYSTEM_ID)) + 1);
   sprintf_P(m_homa_id, PSTR(HOMA_SYSTEM_ID));  //, ESP.getChipId());
   INFOF_P("%s: HomA system id: %s" LF, __func__, m_homa_id);
```
`strlen_P` measures -1 byte, then `sprintf_P` breaks the memory block by putting `\0`

For example, `HOMA_SYSTEM_ID` is `123`. umm fails here for me as well when allocating tcp connection memory after noticing that neighbouring memory is broken
```
No poison after block at: 0x3fff2ba6, actual data: 0x33 0x0 0xa5 0xa5
```

`0xa5` is the 'poison' byte which surrounds the allocation block in groups of four, but it is overwritten with `0x33` (char `3`) and `0x0` (aka null terminator), b/c allocation size is 2 and not the expected 4. Here it is pretty simple example, since the offending mem location is just a string.
```gdb
(gdb) x/16xb 0x3fff2ba6 - 8
0x3fff2b9e:     0x00    0x00    0xa5    0xa5    0xa5    0xa5    0x31    0x32
0x3fff2ba6:     0x33    0x00    0xa5    0xa5    0x00    0x00    0x00    0x00
```

Looks like `123\0` above, pointing at the symbol specifically
(gdb) p m_homa_id
$1 = 0x3fff2ba4 "123"
(gdb) x/16xb (uint8_t*)m_homa_id - 4
0x3fff2ba0:     0xa5    0xa5    0xa5    0xa5    0x31    0x32    0x33    0x00
0x3fff2ba8:     0xa5    0xa5    0x00    0x00    0x00    0x00    0x00    0x00
```

hmueller01, 2024-01-28:
@mcspr you are incredible! Thank you sooo much! What a stupid mistake of mine. Shame on me. I owe you something.
From my other project I changed
```
#define _HOMA_SYSTEM_ID "%06x-rhts"  // numerical part (%06x) gets added during runtime
  m_homa_id = (char *)malloc(8 + strlen_P(PSTR(_HOMA_SYSTEM_ID)));
  sprintf_P(m_homa_id, PSTR(_HOMA_SYSTEM_ID), ESP.getChipId());
```
to
```
#define HOMA_SYSTEM_ID	"123456-garage"
  m_homa_id = (char *)malloc(strlen_P(PSTR(HOMA_SYSTEM_ID) + 1));
  sprintf_P(m_homa_id, PSTR(HOMA_SYSTEM_ID));  //, ESP.getChipId());
```
What a bad idea to write the  +1 at the end at the slightly wrong position.

But good to know how to interpret the  'poison' bytes. Is there a documentation somewhere (tried to find something without success)? And good to know how to use the debugger here. Thanks. Thanks. Thanks.

A last question is still there. Is it possible to set another reset reason? This 'panic' resets the ESP and after reboot
`ESP.getResetReason()` outputs `Software/System restart` (`REASON_SOFT_RESTART`) I would expect an `Exception` and not like a 'normal' restart ...

mcspr, 2024-02-04:
> But good to know how to interpret the 'poison' bytes. Is there a documentation somewhere (tried to find something without success)?

Only through reading the umm code and its comments. I think the example above could be a good addition to the gdb (https://github.com/esp8266/Arduino/blob/master/doc/gdb.rst) or faq (https://github.com/esp8266/Arduino/blob/master/doc/faq/a02-my-esp-crashes.rst , where it would corrupt some memory on purpose with an expected result

> A last question is still there. Is it possible to set another reset reason? This 'panic' resets the ESP and after reboot
ESP.getResetReason() outputs Software/System restart (REASON_SOFT_RESTART) I would expect an Exception and not like a 'normal' restart ...

Something overlooked. iirc, reset reason is passed through the 'custom callback' correctly but is not preserved during reboot as we read from SDK and SDK does not know of this special case

'Exception' as the term used here would mean Xtensa hw 'General Exception Causes'
https://github.com/esp8266/Arduino/blob/de1029ffe0f85b40465ecb64218ef7ab3643ffa7/doc/exception_causes.rst

Value is also not preserved during the reboot correctly for this case, as it is only for display and not really an 'Exception' at all