esp8266/Arduino #9046: mDNS Unable to resolve IP normally in IOS17

422234, 2023-12-06:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP8266]
- Core Version:			[2.7.4&3.1.2]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		[4MB]
- lwip Variant:   [vv2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description

I have an ios application which uses mDNS to find specific devices on the network then it resolves an IP address so then the application can connect to it through websocket. It has been working for years now. Recently I updated my iPads to iOS 17 and suddenly this functionality stopped working.
APP uses  **_http._tcp.**  to obtain the IP of ESP8266
Then I used [mDNS_Web_Server.ino](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266mDNS/examples/mDNS_Web_Server/mDNS_Web_Server.ino) from the library file to test, I can resolve the IP normally in IOS16 (tested using discovery&ABC-bonjour APP) and Android devices, but cannot resolve it normally in IOS17.
In IOS17 I can access http://esp8266.local through the browser and it returns the device's IP.
When I thought it might be a problem with the APP, I tried Arduino-ESP32 
I used [ESP32-mDNS_Web_Server.ino](https://github.com/espressif/arduino-esp32/tree/master/libraries/ESPmDNS/examples/mDNS_Web_Server) in the ESP32 library file and found that under IOS17, the APP can resolve the ESP32 IP normally.Both apps discovery&ABC-bonjour can resolve to the IP of ESP32 through MDNS normally.
May I ask what content is missing in the MDNS library of ESP8266 that causes this problem?

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiClient.h>

#ifndef STASSID
#define STASSID "your-ssid"
#define STAPSK "your-password"
#endif

const char* ssid = STASSID;
const char* password = STAPSK;

// TCP server at port 80 will respond to HTTP requests
WiFiServer server(80);

void setup(void) {
  Serial.begin(115200);

  // Connect to WiFi network
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  // Set up mDNS responder:
  // - first argument is the domain name, in this example
  //   the fully-qualified domain name is "esp8266.local"
  // - second argument is the IP address to advertise
  //   we send our IP address on the WiFi network
  if (!MDNS.begin("esp8266")) {
    Serial.println("Error setting up MDNS responder!");
    while (1) { delay(1000); }
  }
  Serial.println("mDNS responder started");

  // Start TCP (HTTP) server
  server.begin();
  Serial.println("TCP server started");

  // Add service to MDNS-SD
  MDNS.addService("http", "tcp", 80);
}

void loop(void) {

  MDNS.update();

  // Check if a client has connected
  WiFiClient client = server.accept();
  if (!client) { return; }
  Serial.println("");
  Serial.println("New client");

  // Wait for data from client to become available
  while (client.connected() && !client.available()) { delay(1); }

  // Read the first line of HTTP request
  String req = client.readStringUntil('\r');

  // First line of HTTP request looks like "GET /path HTTP/1.1"
  // Retrieve the "/path" part by finding the spaces
  int addr_start = req.indexOf(' ');
  int addr_end = req.indexOf(' ', addr_start + 1);
  if (addr_start == -1 || addr_end == -1) {
    Serial.print("Invalid request: ");
    Serial.println(req);
    return;
  }
  req = req.substring(addr_start + 1, addr_end);
  Serial.print("Request: ");
  Serial.println(req);
  client.flush();

  String s;
  if (req == "/") {
    IPAddress ip = WiFi.localIP();
    String ipStr = String(ip[0]) + '.' + String(ip[1]) + '.' + String(ip[2]) + '.' + String(ip[3]);
    s = "HTTP/1.1 200 OK\r\nContent-Type: text/html\r\n\r\n<!DOCTYPE HTML>\r\n<html>Hello from ESP8266 at ";
    s += ipStr;
    s += "</html>\r\n\r\n";
    Serial.println("Sending 200");
  } else {
    s = "HTTP/1.1 404 Not Found\r\n\r\n";
    Serial.println("Sending 404");
  }
  client.print(s);

  Serial.println("Done with client");
}

```

### Debug Messages

```
Debug messages go here
```

morgendagen, 2024-01-28:
I have just experienced the same issue. mDNS worked fine just fine two-three weeks ago but today I noticed it failed. Unfortunately I do not have any iOS og macOS devices running older OS's, so I cannot verify whether it is something on those devices or something in my network that has changed.

But, I did find a partial fix (unfortunately some services are still not being discovered) by adding a TXT record:

    auto http_service = MDNS.addService(NULL, "http", "tcp", 80);
    MDNS.addServiceTxt(http_service, "path", "/");
    MDNS.announce();

`MDNS.announce()` may not be required. By adding the TXT record, `Discovery.app` can now resolve the service on both iOS 17.2.1 and macOS 14.3.

TiElectronic, 2024-02-10:
> 
> But, I did find a partial fix by adding a TXT record:但是，我确实通过添加 TXT 记录找到了部分修复：
> 
> ```
> auto http_service = MDNS.addService(NULL, "http", "tcp", 80);
> MDNS.addServiceTxt(http_service, "path", "/");
> MDNS.announce();
> ```
> 
Thank you very much for your solution! !
Adding TXT records can indeed solve the problem of IOS17 not being able to resolve IP. 
This is a very strange question，I don't quite understand why this happens.

d-a-v, 2024-02-11:
@morgendagen  Thanks

Would you recommend an update in our examples ?

We already have [this one](https://github.com/esp8266/Arduino/blob/9217458353b98bee50ce0d0ef353aca9e164f456/libraries/ESP8266mDNS/examples/LEAmDNS/mDNS_ServiceMonitor/mDNS_ServiceMonitor.ino#L137-L138) but the [legacy example](https://github.com/esp8266/Arduino/blob/9217458353b98bee50ce0d0ef353aca9e164f456/libraries/ESP8266mDNS/examples/mDNS_Web_Server/mDNS_Web_Server.ino#L69) is missing your fix.

ivanharbaruk, 2024-04-15:
> I have just experienced the same issue. mDNS worked fine just fine two-three weeks ago but today I noticed it failed. Unfortunately I do not have any iOS og macOS devices running older OS's, so I cannot verify whether it is something on those devices or something in my network that has changed.
> 
> But, I did find a partial fix by adding a TXT record:
> 
> ```
> auto http_service = MDNS.addService(NULL, "http", "tcp", 80);
> MDNS.addServiceTxt(http_service, "path", "/");
> MDNS.announce();
> ```
> 
> `MDNS.announce()` may not be required. By adding the TXT record, `Discovery.app` can now resolve the service on both iOS 17.2.1 and macOS 14.3.

Man, thank you a lot. Your solution finally helped me to resolve issue with unstable discovering of my esp8266.

P.S. In my case MDNS.announce(); was required to get working solution.