esp8266/Arduino #9051: HTTP download systematically triggers Exception 9 ()

php4fan, 2023-12-10:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP8266]
- Core Version:			[3.1.2]
- Development Env:		[Arduino IDE]
- Operating System:		[Manjaro Linux]

### Settings in IDE

- Module:  			[Sparkfun ESP8266 Thing Dev] https://www.sparkfun.com/products/13711
- Flash Mode:			[qio|dio|other] no idea
- Flash Size: 		[4MB/1MB] 512K I think
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck|nodemcu] no idea
- Flash Frequency:	[40Mhz] no idea
- CPU Frequency:		[80Mhz|160MHz] no idea
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description

I think this is the same issue as https://github.com/esp8266/Arduino/issues/8557, and a few other similar reports over the last few years that basically got ignored.

The sketch below downloads a file with HTTPClient.
After downloading a few bytes, it systematically crashes with
```
Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
```
on the board mentioned above (I tried more than one unit).
See the full decoded stack below.

With some variations of the sketch where nothing significant apparently changes (e.g. I only change what I Serial.print() and when; or the actual downloaded bytes change but not the download size), I sometimes get a **WDT reset instead**, also for  no apparent reason.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#define NO_GLOBAL_EEPROM
#define DEBUG_MY(...) Serial.printf( __VA_ARGS__ )

#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <string>

const int SERIAL_BAUD_RATE = 74880;

const int LED_PIN = 5; // Thing's onboard, green LED

const int DOWNLOAD_MAX_TIME = 180;
const int DOWNLOAD_MAX_TIME_NODATA = 60;

const int DISPLAY_WIDTH=800;
const int DISPLAY_HEIGHT=480;

String client_ssid="Sixty-Seven";
String client_password="8791oetifiw";

uint8_t output_black_buffer[DISPLAY_WIDTH / 8]; 
uint8_t output_red_buffer[DISPLAY_WIDTH / 8]; 

int currentPixelIdx = 0;
uint32_t out_idx = 0;

void setup() 
{
  Serial.begin(SERIAL_BAUD_RATE);
  DEBUG_MY("=============== Setup =====================\n");
  //WiFi.mode(WIFI_OFF);
  WiFi.persistent(false);
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);
  WiFi.mode(WIFI_STA);
  WiFi.begin(client_ssid.c_str(), client_password.c_str());
  
  bool connOk = false;

  for (int i=0; i<20; i++) {
    DEBUG_MY("Trying to connect to Wifi '%s' with password '%s'\n", client_ssid.c_str(), client_password.c_str());
    delay(1000);
    if (WiFi.status()==WL_CONNECTED) {
      connOk = true;
      break;
    }
  }
  if (!connOk) {
    DEBUG_MY("Couldn't connect to WiFi.\n");
    ESP.deepSleep(10000000);
  }

  HTTPClient http;
  WiFiClient wifi;
  http.setTimeout(20000);
  String url="http://opentarget.org/test/esp8266/download/";

  uint8_t httpbuf[128] = {0};

  DEBUG_MY("... Downloading file from url: %s\n", url.c_str());
  int httpcode=200;
  int len = DISPLAY_WIDTH * DISPLAY_HEIGHT / 4;
  int bytesdownloaded=0;
  // NOTE: beware of https://github.com/esp8266/Arduino/issues/9035
  http.begin(wifi, url);
  
  const char* headersToCollect[1] = {"X-Whatever"};
  http.collectHeaders(headersToCollect, 1);
  httpcode=http.GET();
  if (httpcode==200) {
    len=http.getSize();
  }
  
  DEBUG_MY("... got HTTP code %d; size is %d\n", httpcode, len);
  int ticks=0;
  int ticks0bytes=0;

  if (httpcode==304) {
    DEBUG_MY("No new file to download (HTTP 304 response)\n");
    ESP.deepSleep(10000000);
  }
  
  if (httpcode==200) {
    if (len==-1) {
      DEBUG_MY("Assuming size %d\n", DISPLAY_WIDTH*DISPLAY_HEIGHT / 4);
      len = DISPLAY_WIDTH*DISPLAY_HEIGHT / 4;
    }
    DEBUG_MY("Starting actual download.\n");

    currentPixelIdx = 0; // not really the index of a pixel but rather of a group of 8 pixels
    out_idx = 0;
    
    int timestarted = millis();
    int timechecked = millis();
    int timegotbytes = millis();
    
    while (len > 0) {
      if (millis()-timestarted > DOWNLOAD_MAX_TIME*1000) {
        DEBUG_MY("File download is taking too long. Aborting.\n");
        ESP.deepSleep(10000000);
      }
      int gotbytes=0;
        if (!http.connected()) {
          break;
        }
        size_t avl=http.getStream().available();
        if (avl) {
          gotbytes = http.getStream().readBytes(httpbuf, ((avl>sizeof(httpbuf))?sizeof(httpbuf):avl));
          bytesdownloaded+=gotbytes;
          ticks0bytes=0;
        }
        else {
          delay(100);
          ticks0bytes++;
          if (ticks0bytes>=20) {
            DEBUG_MY("... !! been waiting for available bytes for more than 2 seconds!!\n");
            ticks0bytes=0;
            ticks=10000;
          }
        }
      if (gotbytes>0) {
        len=max(0, len-gotbytes);
        ticks+=gotbytes;
        if (millis()-timechecked > 5000) {
          timechecked=millis();
          ticks+=10000;
        }
        if (ticks>=10000) {
          //DEBUG_MY("...((ticks = %d)) %d bytes downloaded; %d remaining\n", ticks, bytesdownloaded, len);
          ticks=0;
        }
        DEBUG_MY("... streaming %d bytes, %d remaining\n", gotbytes, len);
        
        for (int j=0; j<gotbytes; j++) {
          usePixels(httpbuf[j]);
        }
      
        timegotbytes = millis();
      }
      else {
        if (millis()-timegotbytes > DOWNLOAD_MAX_TIME_NODATA*1000) {
          DEBUG_MY("... !! No data for too long. Aborting.\n");
          ESP.deepSleep(10000000);
        }
      }
      
    }
    DEBUG_MY("Done downloading image (%d bytes downloaded).\n", bytesdownloaded);
    ESP.deepSleep(10000000);
  }
  else {
    DEBUG_MY("...HTTP Error %d\n", httpcode);
    ESP.deepSleep(10000000);
  }
}

void loop() {
  
}

void usePixels(uint8_t pixChar) {
   
  if (currentPixelIdx % 2 == 0) {
    output_black_buffer[out_idx] = pixChar;
  }
  else {
    output_red_buffer[out_idx++] = pixChar; 
    if ((currentPixelIdx / 2) == DISPLAY_WIDTH / 8 - 1) {
      // ... this is where I would send one raw of pixels to the e-paper display
      // by giving it the two buffers output_black_buffer and output_red_buffer
      out_idx = 0;
    }
  }

   currentPixelIdx++;
}

```

### Debug Messages

```
Trying to connect to Wifi 'Sixty-Seven' with password '8791oetifiw'
Trying to connect to Wifi 'Sixty-Seven' with password '8791oetifiw'
ip:192.168.0.20,mask:255.255.255.0,gw:192.168.0.1
... Downloading file from url: http://opentarget.org/test/esp8266/download/
... got HTTP code 200; size is -1
Assuming size 96000
Starting actual download.
... streaming 128 bytes, 95872 remaining
... streaming 128 bytes, 95744 remaining
... streaming 45 bytes, 95699 remaining
... streaming 128 bytes, 95571 remaining
... streaming 128 bytes, 95443 remaining
... streaming 128 bytes, 95315 remaining
Fatal exception 9(LoadStoreAlignmentCause):
epc1=0x40203956, epc2=0x00000000, epc3=0x00000000, excvaddr=0x41000003, depc=0x00000000
```

This is a full decoded exception:
```
Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
PC: 0x40203956:  is in Print::printf(char const*, ...) (/home/teo/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/cores/esp8266/Print.cpp:66).
EXCVADDR: 0x41000003

Decoding stack results
0x40203b1c: Stream::timedRead() at /home/teo/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/cores/esp8266/Stream.cpp:35
0x4020148e: setup() at /home/teo/Documents/progetti/iframeesp8266/iframeesp8266-ino/iFrame-firmware/iFrame-firmware.ino:135
0x40202400:  is in WiFiClient::WiFiClient(WiFiClient const&) (/home/teo/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/cores/esp8266/Print.h:46).
0x40205398: loop_wrapper() at /home/teo/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/cores/esp8266/core_esp8266_main.cpp:255
```

php4fan, 2023-12-10:
Turns out it has **nothing to do with HTTPClient**.

This minimal sketch reproduces the issue:

```
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <string>

const int SERIAL_BAUD_RATE = 74880;

const int LED_PIN = 5; // Thing's onboard, green LED

const int DOWNLOAD_MAX_TIME = 180;
const int DOWNLOAD_MAX_TIME_NODATA = 60;

const int DISPLAY_WIDTH=800;
const int DISPLAY_HEIGHT=480;

String client_ssid="Sixty-Seven";
String client_password="8791oetifiw";

uint8_t output_black_buffer[DISPLAY_WIDTH / 8] = {0}; 
uint8_t output_red_buffer[DISPLAY_WIDTH / 8] = {0}; 

int currentIdx = 0;
int out_idx = 0;

void setup() 
{
  Serial.begin(SERIAL_BAUD_RATE);
  Serial.println("=============== Setup =====================");
  WiFi.mode(WIFI_OFF);
  WiFi.persistent(false);
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);

  for (int i = 0; i < 96000; i++) {
    usePixels();
    if (i % 128 == 127) {
      ESP.wdtFeed();
      Serial.println(".");
    }
  }
}

void loop() {
  
}

void usePixels() {
  if (currentIdx % 2 == 0) {
    output_black_buffer[out_idx] = 1;
  }
  else {
    output_red_buffer[out_idx++] = 1; 
    if ((currentIdx / 2) == DISPLAY_WIDTH / 8 - 1) {
      
      out_idx = 0;
    }
  }

   currentIdx++;
}
```

This is the decoded exception:
```
Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
PC: 0x402017ef:  is in Print::write(char const*) (/home/teo/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/cores/esp8266/Print.h:59).
EXCVADDR: 0x01010105

Decoding stack results
0x402019d8:  is in Print::println(char const*) (/home/teo/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/cores/esp8266/Print.cpp:238).
0x402010e5: setup() at /home/teo/Documents/progetti/iframeesp8266/iframeesp8266-ino/iFrame-firmware/iFrame-firmware.ino:33
0x40202568: loop_wrapper() at /home/teo/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/cores/esp8266/core_esp8266_main.cpp:255
```

If I remove the `Serial.println(".")`, then instead of an exception I get a **wdt reset**, which is also unexpected (note the `wdtFeed()`).

php4fan, 2023-12-10:
I'm an idiot. 🤦

I was writing outside of the bounds of the array.
The "alignment" thing in the error message threw me way off.

closed 2023-12-10 by php4fan

d-a-v, 2023-12-13:
For that kind of code you can [run it on host](https://github.com/esp8266/Arduino/blob/master/tests/host/README.txt) and use tools like valgrind or gdb to debug it out.