esp8266/Arduino #9055: Bug in SoftwareSerial.h in version 3.1.2

RobertGnz, 2023-12-17:
In version 3.1.1 of SoftwareSerial.h we have:

enum SoftwareSerialConfig {
SWSERIAL_5N1 = SWSERIAL_PARITY_NONE,
etc...

this was changed in version 3.1.2 with:

enum Config {
SWSERIAL_5N1 = SWSERIAL_PARITY_NONE,
etc...

which lead to a conflicting name problem when Config is used in an application program but with another definition despite the namespace statement added in SoftwareSerial.h version 3.1.2

mcspr, 2023-12-18:
Also see https://github.com/plerup/espsoftwareserial/issues/301

```cpp
using SoftwareSerialConfig = foo::Config;
```
On the topic, this might do it for us
(and since we are kind of stuck with overly-simplified arduino language model for external types that does not play nice with namespaces)

RobertGnz, 2023-12-18:
Can't use 3.2.2 release because it should oblige to change all existing applications.

RobertGnz, 2023-12-18:
> Also see [plerup/espsoftwareserial#301](https://github.com/plerup/espsoftwareserial/issues/301)
> 
> _edit: nvm the code snippet, enum does not work like that (sadly)_

The problem comes from the directive: " using namespace EspSoftwareSerial; " line 433 which is not a good pratice in terms of programming because it may lead to collision problem.

Please have a look at this page https://learn.microsoft.com/en-us/cpp/cpp/namespaces-cpp?view=msvc-170 and pay a special attention to the note in chapter "Using directives"

mcspr, 2023-12-19:
What I mean, the fix should happen upstream first. Please re-read the issue I linked, swserial is an external lib :>

RobertGnz, 2023-12-19:
> What I mean, the fix should happen upstream first. Please re-read the issue I linked, swserial is an external lib :>

You just linked back to this issue and nothing else.

Pablo2048, 2023-12-19:
EspSoftwareSerial is external library, so @mcspr tries to say that you are barking on wrong tree here - please report the error in the original one, because the original repository has to be changed first (it's used as submodule).

RobertGnz, 2023-12-19:
> EspSoftwareSerial is external library, so @mcspr tries to say that you are barking on wrong tree here - please report the error in the original one, because the original repository has to be changed first (it's used as submodule).

Am I not in master branch ?

RobertGnz, 2023-12-19:
> EspSoftwareSerial is external library, so @mcspr tries to say that you are barking on wrong tree here - please report the error in the original one, because the original repository has to be changed first (it's used as submodule).

Where is the original repository ?

Pablo2048, 2023-12-19:
Did you try to click https://github.com/esp8266/Arduino/tree/master/libraries at the library? It leads to https://github.com/plerup/espsoftwareserial/tree/bcfd6d10e6a45a0d07705d08728f293defe9cc1d ...

RobertGnz, 2023-12-22:
> Did you try to click https://github.com/esp8266/Arduino/tree/master/libraries at the library? It leads to https://github.com/plerup/espsoftwareserial/tree/bcfd6d10e6a45a0d07705d08728f293defe9cc1d ...

The problem comes because of the directive using namespace EspSoftwareSerial wich leads to a global pollution.

Pablo2048, 2023-12-24:
Yes, exactly described here https://github.com/plerup/espsoftwareserial/issues/301 as @mcspr told you... This is wrong repository as it has nothing to do with the issue.