esp8266/Arduino #9061: Wifi does not reconnect, memory leaks when reconnecting

Laxilef, 2023-12-28:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					ESP-12
- Core Version:			SDK:3.0.5(b29dcd3)/Core:3.1.2=30102000/lwIP:STABLE-2_1_3_RELEASE/glue:1.2-65-g06164fb/BearSSL:b024386
- Development Env:		Platformio
- Operating System:		Windows

### Settings in IDE

- Module:  			Wemos D1 mini
- Flash Mode:			default on platformio
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		default on platformio
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		OTA and SERIAL
- Upload Speed:   115200

### Problem Description

On some ESP8266 I am seeing a problem with reconnecting to Wifi. At a random moment in time, the ESP loses its connection to the Wifi and I try to reconnect, but the ESP does not connect. I also tried adding the following code before calling ``WiFi.begin()``:
```c++
        if (wifi_station_dhcpc_status() == DHCP_STARTED) {
          wifi_station_dhcpc_stop();
        }

        if (wifi_softap_dhcps_status() == DHCP_STARTED) {
          wifi_softap_dhcps_stop();
        }

        WiFi.setSleepMode(WIFI_NONE_SLEEP);
        WiFi.persistent(false);
        WiFi.softAPdisconnect();
        WiFi.disconnect(true, false);
        delay(200);
        WiFi.forceSleepWake();
        delay(100);
        WiFi.mode(WIFI_STA);
        delay(100);
        WiFi.setAutoConnect(true);
        WiFi.begin();
``` 
But this does not give any result. Eventually the memory runs out and the ESP reboots with an exception. Decoding the stacktrace does not give anything, because all the exceptions are related to the inability to allocate memory.
The strange thing is that this does not happen with all ESPs.

I compiled with the flags ``-D DEBUG_ESP_CORE -D DEBUG_ESP_WIFI -D DEBUG_ESP_PORT=Serial`` and saw something strange:
```
state: 5 -> 2 (25a0)
rm 0
pm close 7
:er -13 0x00000000
wifi evt: 1
STA disconnect: 37
forcibly stopping the station connection manager
state: 2 -> 0 (0)
wifi evt: 1
STA disconnect: 202
```

After this, the ESP refuses to connect to Wifi. But after a restart (cold start), the connection works correctly.
[... 186 lines cut ...]
[01:41:21][MAIN][VERB] Free heap size: 3144 of 81920 bytes (min: 3144, diff: 232), max free block: 3048 (min: 3048, diff: 232, frag: 3%)
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
wifi evt: 1
STA disconnect: 2
[01:41:25][WIFI][INFO] Reconnecting...
[APdisconnect] set_config failed!
del if0
usl
mode : null
wifi evt: 8
mode : sta(fc:f5:c4:96:39:dc)
add if0
wifi evt: 8
[01:41:25][WIFI][TRACE] Successfully set mode '1'
[01:41:25][WIFI][TRACE] Hostname 'opentherm' is already set
[01:41:25][MAIN][VERB] Free heap size: 2912 of 81920 bytes (min: 2912, diff: 232), max free block: 2816 (min: 2816, diff: 232, frag: 3%)
[01:41:25][MAIN][VERB] Free heap size: 2680 of 81920 bytes (min: 2680, diff: 232), max free block: 2584 (min: 2584, diff: 232, frag: 3%)
[01:41:25][MAIN][VERB] Free heap size: 2448 of 81920 bytes (min: 2448, diff: 232), max free block: 2352 (min: 2352, diff: 232, frag: 3%)
[01:41:25][MAIN][VERB] Free heap size: 2216 of 81920 bytes (min: 2216, diff: 232), max free block: 2120 (min: 2120, diff: 232, frag: 4%)
[01:41:25][MAIN][VERB] Free heap size: 1984 of 81920 bytes (min: 1984, diff: 232), max free block: 1888 (min: 1888, diff: 232, frag: 4%)
[01:41:25][MAIN][VERB] Free heap size: 1520 of 81920 bytes (min: 1520, diff: 464), max free block: 1424 (min: 1424, diff: 464, frag: 6%)
[01:41:26][MAIN][VERB] Free heap size: 1288 of 81920 bytes (min: 1288, diff: 232), max free block: 1192 (min: 1192, diff: 232, frag: 7%)
```

mcspr, 2023-12-29:
Code above does not look like the linked WiFiManager .ino file though. You sure it is Core and / or SDK leaking something and not your app / WiFiManager / some other lib?

> STA disconnect: 202

https://github.com/esp8266/Arduino/blob/d5eb265f78bff9deb7063d10030a02d021c8c66c/tools/sdk/include/user_interface.h#L523

Name would suggest it could not authenticate with the given passphrase and / or auth settings

> WiFi.begin();

begin() without parameters usually means SSID & passphrase were already set, either in persistent mode when those are read from flash or without one where those are managed in RAM. Invalid credentials are the most likely explanation for the error above.

mcspr, 2023-12-30:
> Hmm. SSID and password are stored in flash memory. However, after a reboot or cold start, the ESP successfully connects to the router. For example, calling WiFi.reconnect() gave the same result: the ESP did not connect and the memory decreased with each call to this method.

Depends on two SDK settings, WiFi has slightly different names for them (see ESP8266WiFiSTA.cpp for the underlying stuff)
- WiFi.setAutoConnect(true / false), WiFi.getAutoConnect for current setting - try to connect on boot. assuming this works ok, and already enabled
- WiFi.setAutoReconnect, WiFi.getAutoReconnect - it would keep re-trying when connection attempt fails or after disconnecting from an already connected AP. No user interaction required, it is supposed to work without any extra calls like begin(), reconnect(), disconnect(), etc.

The minimal setup would look something like this, without WiFiManager for right now. Loop then never calls anything related to connection management
```cpp
void setup() {
    // otherwise, Core turns WiFi off. this reverts to 2.x.x and SDK default STATION opmode
    enableWiFiAtBootTime();

    // so WiFi.begin(..., ...) puts stuff into flash
    WiFi.persistent(true);

    // note that persistent() does not affect these two, both values always persist between reboots
    if (!WiFi.getAutoConnect()) {
        WiFi.setAutoConnect(true);
    }

    if (!WiFi.getAutoReconnect()) {
        WiFi.setAutoReconnect(true);
    }

    // assume connection already happening. call begin to reset credentials
    if (WiFi.waitForConnectResult(10000) != WL_CONNECTED) {
        WiFi.begin(MY_SSID, MY_PASS);
    }
}
```

I don't see anything suspicious with a simple ping app. Also no difference between SDK versions. 2.2.x and then upgraded to 3.0.5, both seem to work just fine.

Laxilef, 2024-01-02:
Hi, @mcspr! Happy New Year and thanks for your reply.
I also tried this option, but previously abandoned it because there were complaints that when disconnected from Wifi, the ESP did not reconnect. But I did not call begin when there was no connection, since it is assumed that the connection data is taken from flash.

I am now confused about the situation I described in the first msg. For tests, I’ll try to make the code without all the extra stuff and come back with the results. If you have any ideas why this might be happening, please let me know. Thanks!

timarnoldev, 2024-10-22:
Same problem here, but in softAP mode. Randomly disconnects the client and leaks and bunch of memory when trying to reconnect, which ultimately fails.