esp8266/Arduino #9065: OTA update from GitHub release

tkbstudios, 2024-01-04:
Hello, I would like to download any new release and do an OTA update by downloading from GitHub releases.

But for some reason it doesn't work (I think it's because of HTTPS redirects)

Is there any way to make it work?

https://github.com/tkbstudios/tinet-bridge-esp8266/blob/8b1ccbaddeb60b04542cddbe77f53c2d8d1728e5/tinet-bridge-esp8266.ino#L197

It just tells me HTTP_UPDATE_FAILED.

mcspr, 2024-01-05:
I don't think github supports anything but https:// ? Trying to fetch our release via http:// just redirects to the corresponding https:// url via `HTTP/1.1 301 Moved Permanently`

Swapping client to WiFiClientSecure and changing port to 443 should fix the download
https://github.com/esp8266/Arduino/blob/d5eb265f78bff9deb7063d10030a02d021c8c66c/libraries/ESP8266httpUpdate/examples/httpUpdateSecure/httpUpdateSecure.ino#L87
https://github.com/esp8266/Arduino/blob/d5eb265f78bff9deb7063d10030a02d021c8c66c/libraries/ESP8266httpUpdate/examples/httpUpdateSecure/httpUpdateSecure.ino#L101

closed 2024-01-05 by mcspr

tkbstudios, 2024-01-06:
Thanks will try it out now 👍

tkbstudios, 2024-01-06:
It doesn't work for me, it says HTTP connection failed when I print out the last error.

https://github.com/tkbstudios/tinet-bridge-esp8266/blob/d7ea3c8478d50953f401695c1cf6e3885b0614b9/tinet-bridge-esp8266.ino#L265C12-L265C12