esp8266/Arduino #9068: IPv6 mDNS disabled?

wuast94, 2024-01-10:
Is there a reason why mDNS for IPv6 is disabled by default? 

I want to use it in a project but dont want to modify existing libarys 

https://github.com/esp8266/Arduino/blob/d5eb265f78bff9deb7063d10030a02d021c8c66c/libraries/ESP8266mDNS/src/LEAmDNS.h#L137

d-a-v, 2024-01-10:
To the best of my knowledge, this implementation of mDNS on IPv6 is not actually tested. Its implementation may be incomplete.
Did you try to enable it ?

HeMan, 2024-04-11:
Datatype `IP6Address` seem to be missing.

HeMan, 2024-04-11:
Heh, seem to be a lot more, started fixing in https://github.com/HeMan/Arduino/tree/fix_mdns_ipv6

HeMan, 2024-05-07:
After some nights of pulling my hair I'd say it's broken. If anyone are interested in continue my work I could share my findings. I'll leave my branch as is.

randybb, 2026-10-13:
#9330 enables the IPv6 side of the mDNS responder: it answers AAAA questions and announces the IPv6 addresses (and services) over ff02::fb, so `<host>.local` resolves to the ESP's IPv6 address. The query side (`MDNS_IP6_SUPPORT`) is still off, so this is a partial fix, but it covers the common case of finding the device over IPv6. Tested on several ESP8266 devices with ESPHome.