esp8266/Arduino #9069: ESP8266 Crashing During Prints randomly

vgahlaut, 2024-01-11:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:				ESP8266
- Core Version:			Rel 3.1.2
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[dio]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

We have implemented IOT device. It works stable without any issue if we disable debug prints that we are printing using `Serial.printf()` . However When we enable printing It it crashes randmonly once out of around 5 -7 messages. With printing disabled it did not crash even once with 500 messages sent in loop.

We have checked the print statement, arguments but do not see anything wrong. For example we have below print statement

`uint16_t *rawlen, uint32_t *freq, uint32_t mapid, uint32_t codeid;

Serial.printf("%s:Raw data len : %d Freq: %d retrieved for mapid:codeid %d:%d\r\n", __FUNCTION__, *rawlen, *freq, mapid, codeid);`

Not sure if it is crashing on this print, but we tried to decode exception using plugin [https://github.com/dankeboy36/esp-exception-decoder/actions](url) and here are the exception and decoded details that seems to be pointing to DevNVStore.cpp:1115. Provided above is the print statement on this line. Entire code is too big to be shared.

Below is exception and its decoding
`
Exception (2):
epc1=0x3ffef420 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffef420 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffecdc end: 3fffffb0 offset: 0150
3fffee2c:  3fffdab0 00000000 3fffdad0 3fff187c
3fffee3c:  4021de2e 3ffeb410 40000f49 ffffff00
3fffee4c:  40000f49 40000e19 0000003f 0000027c
3fffee5c:  3ffffed0 000005e0 00000000 40104685
3fffee6c:  3ffef450 0000003c 00000000 3ffffed0
3fffee7c:  00000100 0000001c 00000001 401057cd
3fffee8c:  3ffef450 3ffef420 3fffee2c 402438f1
3fffee9c:  3ffff268 3ffeeb80 00000000 00000000
3fffeeac:  feefeffe 401053b2 3ffef450 40104685
3fffeebc:  3ffef450 0000003c 40104d01 3ffef450
3fffeecc:  000000ff 0000001c 00000001 401057cd
3fffeedc:  3ffef450 401052b5 3ffef450 3ffeeb80
3fffeeec:  00040000 53000000 feefeffe 00000000
3fffeefc:  401025e8 4010558d 00080000 00000000
3fffef0c:  feefeffe 4010385a 40104a42 3ffef450
3fffef1c:  feefeffe 40104eb3 00000000 3ffef618
3fffef2c:  40103350 3ffebb5c 2c9f0300 4000050c
3fffef3c:  3fffc278 4010352c 3fffc200 00000022
3fffef4c:  00040000 00000000 00000000 00000020
3fffef5c:  401005d0 4000d529 00000032 00000000
3fffef6c:  401025e8 3ffeb2d0 40105867 3ffeeb80
[... 302 lines cut ...]
0x4021ec51:  is in uart_do_write_char(int, char) (C:\Users\vipin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266\uart.cpp:513).
0x4022760c: __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio\nano-vfprintf.c:232
0x4022760c: __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio\nano-vfprintf.c:232
0x40222dbd: _printf_i at /workdir/repo/newlib/newlib/libc/stdio\nano-vfprintf_i.c:194
0x4022760c: __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio\nano-vfprintf.c:232
0x40227548: __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio\nano-vfprintf.c:182
0x40222eec: _printf_i at /workdir/repo/newlib/newlib/libc/stdio\nano-vfprintf_i.c:246
0x4022760c: __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio\nano-vfprintf.c:232
0x40222dbd: _printf_i at /workdir/repo/newlib/newlib/libc/stdio\nano-vfprintf_i.c:194
0x40227809: _svfprintf_r at /workdir/repo/newlib/newlib/libc/stdio\nano-vfprintf.c:528
0x40222eec: _printf_i at /workdir/repo/newlib/newlib/libc/stdio\nano-vfprintf_i.c:246
0x4022760c: __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio\nano-vfprintf.c:232
0x40224a9d: _vsnprintf_r at /workdir/repo/newlib/newlib/libc/stdio\vsnprintf.c:71`

Can someone please help? Is it stack smashing? If yes how could such simple print can cause stack to be corrupted?

Do you see any issue in Serial.printf implementation in core?

```

### Debug Messages

```
Debug messages go here
```

vgahlaut, 2024-01-12:
Hi Experts,

Now we have commented particular printf as reported in previous message and decoded stack and run it again. Now every time we are seeing stack smashing every time as below but with exception decoder unable to figure out problem. Can you please help us where to look in to code. Decoded exception does not show any code written by us, its all SDK code

`Stack smashing detected.

Exception (5):
epc1=0x40100553 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3ffffe80 end: 3fffffb0 offset: 0000
3ffffe80:  3fffdad0 00000000 3fff1810 40100553
3ffffe90:  3fffdad0 00000000 3fff1810 4021e09b
3ffffea0:  feefeffe feefeffe 3fffdab0 40100529
3ffffeb0:  feefeffe 3fffee90 00000278 3ffffed0
3ffffec0:  4010fda5 00000004 00000008 4010feae
3ffffed0:  00000000 00000000 00000000 9a000000
3ffffee0:  00000000 400042db 3ffeada0 00000001
3ffffef0:  00000000 401000c0 00000001 401000b8
3fffff00:  00001000 60000200 3fffffb0 3ffffe30
3fffff10:  00000278 00000000 00064320 40100000
3fffff20:  00002780 5ffffe00 00000000 00ff0000
3fffff30:  401006a9 000003fc 00000000 3ffffec2
3fffff40:  400203e9 40100438 ffffff00 ffffe6fc
3fffff50:  ffffffff 3fffff70 00000004 4000177a
3fffff60:  00ff0000 383d373d 343e363e 00000000
3fffff70:  00000000 003c1d0d 003c003c 003c007c
3fffff80:  0f03090f 070f0307 03070f03 fe85130c
3fffff90:  4a4e800e 383c4046 0702fdfa 02fe0e0b
3fffffa0:  12100c07 100a0502 05021613 1613100a
<<<stack<<<

Exception 5: Alloca: MOVSP instruction, if caller's registers are not in the register file
PC: 0x40100553: cont_check(cont_t*) at C:\Users\vipin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266\cont_util.cpp:56
EXCVADDR: 0x00000000

Decoding stack results
0x40100553: cont_check(cont_t*) at C:\Users\vipin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266\cont_util.cpp:56
0x4021e09b: loop_wrapper() at C:\Users\vipin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266\core_esp8266_main.cpp:260
0x401000c0: app_entry() at C:\Users\vipin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266\core_esp8266_main.cpp:392
0x401000b8: app_entry_redefinable() at C:\Users\vipin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266\core_esp8266_main.cpp:386
0x401006a9: uart_write_char_d(char) at C:\Users\vipin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266\core_esp8266_postmortem.cpp:310`

mcspr, 2024-01-19:
^ dump above just means crash had happened, same as 'Exception' line is used to fit into existing crash format. At that point stack is already rolled back and we are calling `cont_check` that causes the panic
(although, I wonder if we should use a different format here)

One way to guard against stack usage is to enable 'stack protection' option from the IDE menu, which would force the stack check after every function call instead of just after user task suspends (yield(), delay(), esp\_suspend(), esp\_yield()). This way crash would happen while inside of some function, not within our

While that is enabled, you could also amend the option build flags for your board with `-Wstack-usage=NUM` where NUM can be any number of bytes we are worried about. e.g. for `generic` board, like you mentioned in the template above
```ini
generic.menu.stacksmash.enabled.build.stacksmash_flags=-fstack-protector
```
```ini
generic.menu.stacksmash.enabled.build.stacksmash_flags=-fstack-protector -Wstack-usage=512
```
I would also suggest going lower even to possibly catch some nested calls causing stack to grow too much. Note that build log size would greatly increase.
See `C:\Users\vipin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\boards.txt` and find your specific board, modify the .txt and reload the IDE. Make sure build warnings are enabled in the preferences menu, otherwise it wouldn't print anything.

Maybe related to #8666