esp8266/Arduino #9086: Callbacks of WiFi events (onX) are not called

srvmgr, 2024-02-09:
https://github.com/esp8266/Arduino/blob/de1029ffe0f85b40465ecb64218ef7ab3643ffa7/libraries/ESP8266WiFi/src/ESP8266WiFiGeneric.cpp#L253

I figured out that callbacks of events registered with methods onX... (like onStationModeGotIP) are not executed.
Looking in code, in file ESP8266WiFiGeneric.cpp, method _eventCallback:

For events registered using predefined methods (which have mCanExpire = true), are removed and not executed, because of the next line:

`if (handler->canExpire() && handler.unique()) {`

But when they are registered using the old way (onEvent(....)), because mCanExpire = false, they are executed.

```
for(auto it = std::begin(sCbEventList); it != std::end(sCbEventList); ) {
        WiFiEventHandler &handler = *it;
        if (handler->canExpire() && handler.unique()) {
            it = sCbEventList.erase(it);
        }
        else { /** ELSE SHOULD EXISTS? or  (*handler)(event); should be called allways regardless of if (handler->canExpire()... ***/
            (*handler)(event);
            ++it;
        }
    }
```

mcspr, 2024-02-09:
Are you adding event callback like this?
```cpp
void setup() {
  WiFi.onStationModeGotIP([](auto&& event) {
    ...
  });
}
```
*or*
```cpp
void somefunc() {
  auto foo = WiFi.onStationModeGotIP([](auto&& event) {
    ...
  });
}
```
If so, you cause the 'unique()' check above to succeed. Return value of `onStationModeGotIP` is a shared pointer object, which counts amount of references to it. If there is only one reference i.e. reference of the object in the sCbEventList *and* it was not registered through onEvent(...), it erases the callback as it deems it unused. This was done with the intent of allowing de-registering callbacks

srvmgr, 2024-02-09:
I register it one time only, like the first example

PR #9087 (2024-03-17): ESP8266WiFi - document event handler lifetime, add [[nodiscard]]
See https://github.com/esp8266/Arduino/issues/9086#issuecomment-1936372500 / @srvmgr
resolve #9086

Mention that the object should either be global or static, and that the callback must be called from either setup() or after it is done.
Mark event methods as `[[nodiscard]]` to add a warning when building without assigning them anywhere
closed 2024-03-17 by mcspr