esp8266/Arduino #9101: HTTPClient redirect fails with http code -7 (guessed root cause included)

1plaintext, 2024-03-14:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:		  ESP8266 NodeMCU
- Core Version:      3.1.2
- Development Env:		 Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  		NodeMCU 1.0 (ESP-12E Module)
- Flash Mode:		other
- Flash Size: 		2MB/1MB
- lwip Variant:    v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:    115200

### Problem Description

When using HTTPClient setFollowRedirects(HTTPC_FORCE_FOLLOW_REDIRECTS) (follow redirect) on api.weather.gov (very popular weather source), redirect fails with http code -7
This is likely because when sending out 301, the server sends out not only _headers_, but _http content_ as well (content length not 0). The guess is: the HTTPClient code itself, when making the second (redirected) request reuses connection, seemingly did not clear out the pipe enough (reason unknown), and caused subsequent reply to not look like an HTTP reply

From debug output:
Reply has content, not just headers
02:44:38.199 -> [HTTP-Client][handleHeaderResponse] RX: 'Content-Length: 461'
subsequent reply has gibberish in front, presumably left over from previous reply
02:44:38.333 -> [HTTP-Client][handleHeaderResponse] RX: 'est with the redundancy removed. If your client supports it, you will be redirected.",'
02:44:38.333 -> [HTTP-Client][handleHeaderResponse] RX: '    "instance": "https://api.weather.gov/requests/1aa0d75b"'
02:44:38.333 -> [HTTP-Client][handleHeaderResponse] RX: '}HTTP/1.1 200 OK'

Setting _reuse to false (new connection for redirect) solves the problem.

I am not sure api.weather.gov is violating standard by doing this, but "force follow" is supposed to throw standard out of the window, and consider api.weather.gov is very popular, this is worth fixing IMHO.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <ESP8266HTTPClient.h>
#include <WiFiClientSecureBearSSL.h>

#ifndef STASSID
#define STASSID "myapssid"
#define STAPSK "password"
#endif

ESP8266WiFiMulti WiFiMulti;

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println();
  Serial.println();
  WiFi.mode(WIFI_STA);
  WiFiMulti.addAP(STASSID, STAPSK);
  Serial.println("setup() done connecting to ssid '" STASSID "'");
}
[... 113 lines cut ...]
'
02:44:38.399 -> [HTTP-Client][handleHeaderResponse] RX: 'Cache-Control: public, max-age=83281, s-maxage=120
'
02:44:38.399 -> [HTTP-Client][handleHeaderResponse] RX: 'Expires: Fri, 15 Mar 2024 08:52:40 GMT
'
02:44:38.433 -> [HTTP-Client][handleHeaderResponse] RX: 'Date: Thu, 14 Mar 2024 09:44:39 GMT
'
02:44:38.433 -> [HTTP-Client][handleHeaderResponse] RX: 'Content-Length: 3098
'
02:44:38.433 -> [HTTP-Client][handleHeaderResponse] RX: 'Connection: keep-alive
'
02:44:38.433 -> [HTTP-Client][handleHeaderResponse] RX: 'X-Edge-Request-ID: 293e96aa
'
02:44:38.433 -> [HTTP-Client][handleHeaderResponse] RX: 'Vary: Accept,Feature-Flags,Accept-Language
'
02:44:38.433 -> [HTTP-Client][handleHeaderResponse] RX: 'Strict-Transport-Security: max-age=31536000 ; includeSubDomains ; preload
'
02:44:38.465 -> [HTTP-Client][handleHeaderResponse] RX: '
'
02:44:38.465 -> [HTTP-Client][handleHeaderResponse] code: 0
02:44:38.465 -> [HTTP-Client][handleHeaderResponse] size: 3098
02:44:38.465 -> [HTTP-Client][handleHeaderResponse] Remote host is not an HTTP Server![HTTP-Client][returnError] error(-7): no HTTP server
02:44:38.465 -> [HTTP-Client][returnError] tcp stop
02:44:38.503 -> [HTTPS] GET... failed, error: no HTTP server
02:44:38.503 -> [HTTP-Client][end] tcp is closed

1plaintext, 2024-03-15:
Jumped the gun too soon when opening the issue, it appears with the latest master I indeed no longer experience this issue. It appears to be same cause as #8871 - only part of the content body was flushed before it timed out. PR #8874 appears to resolve it.

closed 2024-03-15 by 1plaintext

d-a-v, 2024-03-16:
Thanks for investigating. We need to make a new release.