esp8266/Arduino #9113: Network port no aparece para usar OTA

Edwi2026ch, 2024-03-28:
Buenos dia, soy nuevo en este mundo del esp8266. Por favor requiero ayuda.
Mi problema es que cuando cargo el programa de un ejemplo de OTA en el esp 8266 se carga bien pero no aparece los puertos network ports. Estoy usando IDE 1.8.19
Gracias

Juniwell, 2024-03-29:
If you've uploaded your OTA example sketch to your ESP8266 but it doesn't appear in your network ports, try this:
- Check that your PC is connected to the same network as your ESP.
- Select your network in the Network and Sharing Center if you're on Windows.
- Uncheck the "Internet Protocol version 6 (TCP/IPv6)" option.
-If you return to the Arduino IDE, in "Tool > Port" you should see the network port option.
- You can check IPv6 again once your network port is recognized.

Juniwell, 2024-03-29:
If you've uploaded your OTA example sketch to your ESP8266 but it doesn't appear in your network ports, try this:
- Check that your PC is connected to the same network as your ESP.
- Select your network in the Network and Sharing Center if you're on Windows.
- Uncheck the "Internet Protocol version 6 (TCP/IPv6)" option.
- If you return to the Arduino IDE, in "Tool > Port" you should see the network port option.
- You can check IPv6 again once your network port is recognized.

Edwi2026ch, 2024-03-30:
Buen dia.
Muchas gracias por responder. Seguí los pasos que me indico y nada, no aparece el puerto de red. 
Si mepodrian ayudar seria genial muchas gracias,
Saludos
Edwin