esp8266/Arduino #9137: ESP8266 MAC Custom  After Power Oscillations

fpalarminiscenario, 2024-05-13:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:				ESP-12
- Core Version:			2.7.4
- Development Env:		Platformio
- Operating System:		Linux Mint

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			qio
- Flash Size: 		4MB
- lwip Variant:   v2 Higher Bandwidth
- Reset Method:		nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		OTA
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

I have been setting a custom MAC address on the ESP8266 using the wifi_set_macaddr and wifi_get_macaddr functions. So far, it has been working as expected. I confirmed this by turning the device off and on again, and the new MAC address remained in effect.

Recently, I conducted a new test to simulate an oscillation in the power source by quickly unplugging and plugging it back in several times. During this test, I noticed an unusual behavior. After the power oscillations, the MAC address read by wifi_get_macaddr was reverted to its original value

To resolve this issue, I read the MAC address from the EEPROM and compared it to the value returned by wifi_get_macaddr. If they didn't match, I executed ESP.restart(), and the correct custom MAC address was restored.

I would like to understand why this issue occurs. Is there any code within user_init that could fail due to power oscillations? I appreciate any insights or suggestions to prevent this issue in the future.

Thank you

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
#include <EEPROM.h>

void loadMAC() {
    uint8_t mac[6];
    for(int i = 0; i < 6; i++)
    {
        mac[i] = EEPROM.read(i);
    }
    wifi_set_macaddr(STATION_IF, mac);
}

void verifyMAC() {
    uint8_t macESP[6];
    uint8_t macEEPROM[6];
    WiFi.macAddress(macESP);
    for(int i = 0;i < 6; i++)
    {
        macEEPROM[i] = EEPROM.read(i);
        if (macEEPROM[i] != macESP[i])
        {
            ESP.restart();
        }
    }
}

void setup() {
EEPROM.begin(100);
loadMAC();
verifyMAC();
}

void loop() {

}

```

mcspr, 2024-05-15:
https://github.com/esp8266/Arduino/issues/8554#issuecomment-1162376481 ?

> STA interface needs to be up before changing its mac address

I don't see any mode change in your code, so you rely entirely on a chance SDK decides to bring IF up before setup(). This may not always be the case.

Unless the issue you are observing is with EEPROM erasing its contents?

mcspr, 2024-05-15:
> 2.7.4

Also, please update to 3.1.2 w/ WiFi mode() call and retry

fpalarminiscenario, 2024-05-27:
> [#8554 (comment)](https://github.com/esp8266/Arduino/issues/8554#issuecomment-1162376481) ?
> 
> > STA interface needs to be up before changing its mac address
> 
> I don't see any mode change in your code, so you rely entirely on a chance SDK decides to bring IF up before setup(). This may not always be the case.

You are right, it resolved my issue. I was setting mode after changing MAC address. Thanks

closed 2024-05-27 by fpalarminiscenario