esp8266/Arduino #9138: Device is unable to retrieve domain IP after DNS change on router without interrupting the WiFi network

cziter15, 2024-05-16:
#### Platform

- Hardware:				ESP8266 generic / own PCB
- Core Version:			Release 3.1.2
- Development Env:		Platformio, but it doesn't matter
- Operating System:		Windows / Linux (I have automatic pipeline)

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			qio
- Flash Size: 		        4MB
- lwip Variant: 		        default
- Reset Method:		n/a
- Flash Frequency:		40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		OTA

### Problem Description - short
After pulling out SIM card from my LTE router and plugging it back again, the esp8266 based device is unable to reconnect to the server via domain name - **getHostByName** fails with timeout (-3).

The workaround is to set fixed DNS just after WiFi connection establishment.

```cpp
	inline void overrideDnsWithGoogleDns()
	{
		/* Define override DNS addresses. */
		static const ip_addr primaryDns{LWIP_MAKEU32(8, 8, 8, 8)};
		static const ip_addr secondaryDns{LWIP_MAKEU32(8, 8, 4, 4)};
		
		/* Set DNS servers. */
		dns_setserver(0, &primaryDns);
		dns_setserver(1, &secondaryDns);
	}
```

### Problem Description - long

I'm working on home automation system since 2017 or so. My hardware platform is based on various ESP versions. I've also built a library that integrates shared device functionality, like MQTT and so on. The library is open source: https://github.com/cziter15/ksIotFrameworkLib.

The issue appear only on arduino for esp8266 and is not observed on esp32 arduino core.

I have ksMqttConnector component in the library that manages MQTT connection. The user can pass domain name, which will be used to retry connection after disconnect. It's very flexible, because it also handles the case when domain's IP changes (dyn DNS configurations or public domains).

For refrence, please check:
https://github.com/cziter15/ksIotFrameworkLib/blob/master/src/ksf/comp/ksMqttConnector.cpp

Reproduction steps:
- Start the device and wait until it connects to the MQTT.
- Pull out the SIM from the LTE router and wait few seconds.
- Plug in the card and wait until the network is back again on the PC.
- Notice that arduino-esp8266 based device is unable to reconnect to the target.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch
N/A

d-a-v, 2024-05-25:
It would be interesting to have a network dump from esp8266's pov of what's happening between the moments when the sim is removed and replugged. If that's a bug, we shall fix it.

You can enable it with the included [netdump](https://github.com/esp8266/Arduino/tree/master/libraries/Netdump/examples/Netdump) library.

Did you try with git-master version version of the core ?

cziter15, 2024-05-25:
I didn't try master, will try to do so. Just need some time (few days) to get one of my devices as I don't have any 8266 devkit around. I'll also add the dump. Repro is simple, not random at all, so it shouldn't take much time.

I'll come back with the details.

cziter15, 2024-06-01:
```
13:18:22.100 ->     4499 0 out IGMP type(0x16)
13:18:22.240 ->     4641 0 out MDNS 192.168.1.118>224.0.0.251 ID=0x0000 F=0x0000 Q=1 TR=1 
13:18:22.240 ->            H 14 e9 14 e9 00 4c 92 6d                                                 .....L.m
13:18:22.240 ->            D 00 00 00 00 00 01 00 00 00 01 00 00 0b 6e 65 74 64 75 6d 70 68 6f 73 74 .............netdumphost
13:18:22.240 ->            D 05 6c 6f 63 61 6c 00 00 ff 80 01 0b 6e 65 74 64 75 6d 70 68 6f 73 74 05 .local......netdumphost.
13:18:22.240 ->            D 6c 6f 63 61 6c 00 00 01 00 01 00 00 00 78 00 04 c0 a8 01 76 00 00 00 00 local........x.....v....
13:18:22.286 ->            D a4 02 4c 02                                                             ..L.
13:18:22.520 ->     4938 0 out MDNS 192.168.1.118>224.0.0.251 ID=0x0000 F=0x0000 Q=1 TR=1 
13:18:22.520 ->            H 14 e9 14 e9 00 4c 92 6d                                                 .....L.m
13:18:22.520 ->            D 00 00 00 00 00 01 00 00 00 01 00 00 0b 6e 65 74 64 75 6d 70 68 6f 73 74 .............netdumphost
13:18:22.567 ->            D 05 6c 6f 63 61 6c 00 00 ff 80 01 0b 6e 65 74 64 75 6d 70 68 6f 73 74 05 .local......netdumphost.
13:18:22.567 ->            D 6c 6f 63 61 6c 00 00 01 00 01 00 00 00 78 00 04 c0 a8 01 76 00 00 00 00 local........x.....v....
13:18:22.567 ->            D a4 02 4c 02                                                             ..L.
13:18:22.800 ->     5235 0 out MDNS 192.168.1.118>224.0.0.251 ID=0x0000 F=0x0000 Q=1 TR=1 
13:18:22.847 ->            H 14 e9 14 e9 00 4c 92 6d                                                 .....L.m
13:18:22.847 ->            D 00 00 00 00 00 01 00 00 00 01 00 00 0b 6e 65 74 64 75 6d 70 68 6f 73 74 .............netdumphost
13:18:22.847 ->            D 05 6c 6f 63 61 6c 00 00 ff 80 01 0b 6e 65 74 64 75 6d 70 68 6f 73 74 05 .local......netdumphost.
13:18:22.847 ->            D 6c 6f 63 61 6c 00 00 01 00 01 00 00 00 78 00 04 c0 a8 01 76 00 00 00 00 local........x.....v....
13:18:22.847 ->            D a4 02 4c 02                                                             ..L.
13:18:24.113 ->     6533 0 out MDNS 192.168.1.118>224.0.0.251 ID=0x0000 F=0x8400 R=2 
13:18:24.113 ->            H 14 e9 14 e9 00 5d e2 1b                                                 .....]..
13:18:24.113 ->            D 00 00 84 00 00 00 00 02 00 00 00 00 0b 6e 65 74 64 75 6d 70 68 6f 73 74 .............netdumphost
13:18:24.159 ->            D 05 6c 6f 63 61 6c 00 00 01 80 01 00 00 00 78 00 04 c0 a8 01 76 03 31 31 .local........x.....v.11
13:18:24.159 ->            D 38 01 31 03 31 36 38 03 31 39 32 07 69 6e 2d 61 64 64 72 04 61 72 70 61 8.1.168.192.in-addr.arpa
13:18:24.159 ->            D 00 00 0c 80 01 00 00 00 78 00 02 c0 0c 00 00 01 8a 02 2a 02 30          ........x.........*.0
13:18:24.439 ->     6859 0 in  IGMP IGMPv3 Membership Report
13:18:24.439 ->     6860 0 in  IGMP IGMPv3 Membership Report
13:18:25.047 ->     7469 0 in  MDNS 192.168.1.105>224.0.0.251 ID=0x0000 F=0x8400 R=1 
13:18:25.047 ->            H 14 e9 14 e9 00 2e 87 24                                                 .......$
13:18:25.047 ->            D 00 00 84 00 00 00 00 01 00 00 00 00 04 4b 52 49 53 05 6c 6f 63 61 6c 00 .............KRIS.local.
13:18:25.092 ->            D 00 01 80 01 00 00 00 0a 00 04 c0 a8 01 69 00 00 ff ff 00 00 01 00       .............i........
13:18:25.092 ->     7496 0 in  MDNS 192.168.1.105>224.0.0.251 ID=0x0000 F=0x8400 R=1 DR=1 
13:18:25.092 ->            H 14 e9 14 e9 00 42 45 b7                                                 .....BE.
13:18:25.092 ->            D 00 00 84 00 00 00 00 01 00 00 00 01 04 4b 52 49 53 05 6c 6f 63 61 6c 00 .............KRIS.local.
13:18:25.092 ->            D 00 01 80 01 00 00 00 78 00 04 c0 a8 01 69 c0 0c 00 2f 80 01 00 00 00 78 .......x.....i.../.....x
13:18:25.139 ->            D 00 08 c0 0c 00 04 40 00 00 08 00 00 00 00 00 3d 16 04                   ......@........=..
13:18:25.186 ->     7582 0 out MDNS 192.168.1.118>224.0.0.251 ID=0x0000 F=0x8400 R=2 
13:18:25.186 ->            H 14 e9 14 e9 00 5d e2 1b                                                 .....]..
13:18:25.186 ->            D 00 00 84 00 00 00 00 02 00 00 00 00 0b 6e 65 74 64 75 6d 70 68 6f 73 74 .............netdumphost
13:18:25.186 ->            D 05 6c 6f 63 61 6c 00 00 01 80 01 00 00 00 78 00 04 c0 a8 01 76 03 31 31 .local........x.....v.11
13:18:25.186 ->            D 38 01 31 03 31 36 38 03 31 39 32 07 69 6e 2d 61 64 64 72 04 61 72 70 61 8.1.168.192.in-addr.arpa
13:18:25.186 ->            D 00 00 0c 80 01 00 00 00 78 00 02 c0 0c 30 38 20 8a 02 2a 02 30          ........x....08 ..*.0
13:18:25.373 ->     7776 0 in  ARP  who has 192.168.1.102 tell 192.168.1.102
13:18:25.373 ->            D 00 01 08 00 06 04 00 01 fa f1 b8 70 f6 86 c0 a8 01 66 00 00 00 00 00 00 ...........p.....f......
13:18:25.373 ->            D c0 a8 01 66                                                             ...f
13:18:26.211 ->     8631 0 out MDNS 192.168.1.118>224.0.0.251 ID=0x0000 F=0x8400 R=2 
13:18:26.211 ->            H 14 e9 14 e9 00 5d e2 1b                                                 .....]..
13:18:26.211 ->            D 00 00 84 00 00 00 00 02 00 00 00 00 0b 6e 65 74 64 75 6d 70 68 6f 73 74 .............netdumphost
13:18:26.258 ->            D 05 6c 6f 63 61 6c 00 00 01 80 01 00 00 00 78 00 04 c0 a8 01 76 03 31 31 .local........x.....v.11
[... cut ...]
13:19:16.022 ->            H 60 00 00 00 00 24 00 01 fe 80 00 00 00 00 00 00 c6 66 3d 13 c6 7e 2b db `....$...........f=..~+.
13:19:16.022 ->            H ff 02 00 00 00 00 00 00 00 00 00 00 00 00 00 16                         ................
13:19:16.069 ->            D 3a 00 05 02 00 00 01 00 8f 00 7b 3c 00 00 00 01 02 00 00 00 ff 02 00 00 :.........{<............
13:19:16.069 ->            D 00 00 00 00 00 00 00 00 00 00 00 fb                                     ............
13:19:16.069 ->    58477 0 in  IGMP IGMPv3 Membership Report
```
~ 13:18:35 ->  sim ejected
~ 13:18:55 -> sim inserted

mcspr, 2024-06-01:
Are we looking for DNS or MDNS requests though? `KRIS.local` is the one failing to resolve?

cziter15, 2024-06-01:
> Are we looking for DNS or MDNS requests though? `KRIS.local` is the one failing to resolve?

No, just trying to resolve external domain like google.com. Maybe I should put some logic from my app into the sketch.